RoboStack/ros-humble #113: rosdep failing on Windows

danzimmerman, 2023-11-04:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

When I try to use `rosdep` on a new Humble installation on Windows, including a fresh `miniforge3` installation, I get the following error:

```
rosdep version: 0.22.2

Bad installer [conda]: None of micromamba.exe, mamba.exe, conda.exe was found.
```

It seems that in `rosdep2\platforms\conda.py` the candidate installation commands are explicitly anticipating an `.exe` on Windows:

```python
def get_conda_mamba_cmd():
    candidate_list = ['micromamba', 'mamba', 'conda'] if sys.platform != 'win32' else ['micromamba.exe', 'mamba.exe', 'conda.exe']
    for candidate in candidate_list:
        try:
            subprocess.Popen([candidate], stdout=subprocess.PIPE, stderr=subprocess.PIPE).communicate()
            return candidate
        except OSError as e:
            continue
    raise Exception('None of ' + ', '.join(candidate_list) + ' was found.')
```

but the actual commands point to batch files:

```
(humble) C:\code\ros\moveit_ws>where mamba
C:\Users\dan\miniforge3\condabin\mamba.bat

(humble) C:\code\ros\moveit_ws>where conda
C:\Users\dan\miniforge3\condabin\conda.bat
```

`mamba.exe` and `conda.exe` aren't on the path:

```
(humble) C:\code\ros\moveit_ws>mamba.exe
'mamba.exe' is not recognized as an internal or external command,
operable program or batch file.
```

It does seem to make it further if I modify `conda_mamba_cmd()`, changing `.exe` to `.bat`:

```python
candidate_list = ['micromamba', 'mamba', 'conda'] if sys.platform != 'win32' else ['micromamba.bat', 'mamba.bat', 'conda.bat']
```

Some other info: 

When I installed `miniforge3` I did not check the option to add its executables to the system path as described [here](https://github.com/conda-forge/miniforge#windows). 

I'm using Windows Terminal and using the following command to activate the Humble Conda environment and source my workspace:

```cmd
C:\Windows\System32\cmd.exe /k "%USERPROFILE%\miniforge3\Scripts\activate.bat %USERPROFILE%\miniforge3\envs\humble && C:\code\ros\moveit_ws\install\setup.bat"
```

Wanted to open this to discuss if this should be considered a config issue or if `rosdep` logic possibly should look for `.bat` files in certain circumstances.

### Installed packages

```shell
# packages in environment at C:\Users\dan\miniforge3\envs\humble:
#
# Name                    Version                   Build  Channel
[... 634 lines cut ...]
                          C:\Users\dan\miniforge3\envs\humble\.condarc
          conda version : 23.9.0
    conda-build version : not installed
         python version : 3.10.12.final.0
       virtual packages : __archspec=1=x86_64
                          __win=0=0
       base environment : C:\Users\dan\miniforge3  (writable)
      conda av data dir : C:\Users\dan\miniforge3\etc\conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack-staging/win-64
                          https://conda.anaconda.org/robostack-staging/noarch
                          https://conda.anaconda.org/conda-forge/win-64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : C:\Users\dan\miniforge3\pkgs
                          C:\Users\dan\.conda\pkgs
                          C:\Users\dan\AppData\Local\conda\conda\pkgs
       envs directories : C:\Users\dan\miniforge3\envs
                          C:\Users\dan\.conda\envs
                          C:\Users\dan\AppData\Local\conda\conda\envs
               platform : win-64
             user-agent : conda/23.9.0 requests/2.31.0 CPython/3.10.12 Windows/10 Windows/10.0.22621
          administrator : False
             netrc file : None
           offline mode : False
```

Tobias-Fischer, 2023-11-04:
Hi @danzimmerman, many thanks for reporting this! Could you please describe what you mean by it “makes it further” when you change .exe to .bat? Is it working as expected, or not? I’d be happy to add another check for the .bat scripts, but want to make sure first that it actually works.

danzimmerman, 2023-11-05:
> Is it working as expected, or not? I’d be happy to add another check for the .bat scripts, but want to make sure first that it actually works.

I can confirm that I successfully rosdep-installed a package for which there was an installation candidate.

It was appearing to hang after attempting to install a number of packages for which there is no candidate, but now I think that Mamba may just be running very slowly on this laptop for some reason. Will try again for the failure case and make sure it actually completes.

I'll try again

danzimmerman, 2023-11-05:
Yes, it seems to be working as expected.

I modified `conda.py` with:

```python
candidate_list = ['micromamba', 'mamba', 'conda'] if sys.platform != 'win32' else ['micromamba.exe', 'mamba.exe', 'conda.exe', 'mamba.bat', 'conda.bat']
```

Adding a package I don't have installed, but that can be installed successfully:

```
(humble) C:\code\ros\moveit_ws>rosdep install --from-paths src\dz_launch_examples --ignore-src -r -y
executing command [mamba.bat install -c robostack-staging -c conda-forge -y ros-humble-camera-calibration]

Looking for: ['ros-humble-camera-calibration']

conda-forge/win-64                                          Using cache
conda-forge/noarch                                          Using cache
robostack-staging/win-64                                      No change
robostack-staging/noarch                                      No change

Pinned packages:
  - python 3.10.*

Transaction

  Prefix: C:\Users\dan\miniforge3\envs\humble

  Updating specs:

   - ros-humble-camera-calibration
   - ca-certificates
   - certifi
   - openssl

  Package                          Version  Build            Channel                 Size
-------------------------------------------------------------------------------------------
  Install:
-------------------------------------------------------------------------------------------

  + ros-humble-camera-calibration    3.0.0  py310ha45506e_3  robostack-staging     Cached

  Summary:

  Install: 1 packages

  Total download: 0 B

-------------------------------------------------------------------------------------------

Downloading and Extracting Packages:

Preparing transaction: done
Verifying transaction: done
Executing transaction: done

```

Adding a fake single package with no resolvable `rosdep` key:

```
(humble) C:\code\ros\moveit_ws>rosdep install --from-paths src\dz_launch_examples --ignore-src -r -y
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
dz_launch_examples: Cannot locate rosdep definition for [fake_ros_package_mamba_test]
Continuing to install resolvable dependencies...
#All required rosdeps installed successfully
```

Adding a real package with a `rosdep` key but no installation candidate on Windows:
[... 41 lines cut ...]
robostack-staging/noarch                                      No change
conda-forge/noarch                                  12.4MB @  15.1MB/s  0.8s
conda-forge/win-64                                  19.5MB @  14.3MB/s  1.4s

Pinned packages:
  - python 3.10.*

Could not solve for environment specs
The following packages are incompatible
├─ ros-humble-ament-clang-tidy does not exist (perhaps a typo or a missing channel);
├─ ros-humble-gazebo-ros2-control does not exist (perhaps a typo or a missing channel);
├─ ros-humble-gripper-controllers does not exist (perhaps a typo or a missing channel);
├─ ros-humble-joint-state-broadcaster does not exist (perhaps a typo or a missing channel);
├─ ros-humble-joint-trajectory-controller does not exist (perhaps a typo or a missing channel);
├─ ros-humble-moveit-hybrid-planning does not exist (perhaps a typo or a missing channel);
├─ ros-humble-moveit-ros-perception does not exist (perhaps a typo or a missing channel);
├─ ros-humble-moveit-servo does not exist (perhaps a typo or a missing channel);
├─ ros-humble-pick-ik does not exist (perhaps a typo or a missing channel);
├─ ros-humble-picknik-reset-fault-controller does not exist (perhaps a typo or a missing channel);
├─ ros-humble-picknik-twist-controller does not exist (perhaps a typo or a missing channel);
└─ ros-humble-ros2-controllers does not exist (perhaps a typo or a missing channel).

... still waiting after 15 min ...
```
</details>

Tobias-Fischer, 2023-11-06:
The .bat issue will be resolved with https://github.com/conda-forge/rosdep-feedstock/pull/26

Regarding the hanging issue: That's not a RoboStack specific issue, and should be reported upstream at https://github.com/mamba-org/mamba

Have you tried micromamba? Often works much better than conda/mamba :)

closed 2023-11-06 by Tobias-Fischer