RoboStack/ros-humble #115: Isolating Micro-ROS topics

Muthukumar4796, 2023-12-16:
When I add the line 'export ROS_DOMAIN_ID=47' to the bashrc file and then load Micro-ROS, my custom topics are not visible. However, when I comment out the ROS domain ID line, restart the agent, and reload Micro-ROS, my custom topics become visible. I'm stuck and unsure how to properly isolate my Micro-ROS topics without interference from the ROS domain ID Can anyone provide guidance on how to properly isolate Micro-ROS topics in this setup?

Note: For Isolating the ROS2 topics using ROS_DOMAIN_ID is working fine.

Tobias-Fischer, 2023-12-16:
Hi, is that a RoboStack related question? It sounds like this issues would be better placed in a Micro ROS repository.

Best, Tobi

Muthukumar4796, 2023-12-16:
Thanks for your response, Tobi. I did submit my query to the Micro ROS repository, but I also wanted to reach out to this community for any additional insights or support. Your positive feedback is greatly appreciated.

Tobias-Fischer, 2023-12-16:
Unfortunately I’ll have to close this issue if it is not directly related with RoboStack - the https://discourse.ros.org/ is the place to go for additional non-RoboStack related support.

Best wishes, Tobi
PS: feel free to reopen if your issue is indeed related to RoboStack and not ROS in general.

closed 2023-12-16 by Tobias-Fischer