RoboStack/ros-humble #116: cctools-port: generating fake signature

xiao-wang007, 2023-12-16:
Hi 

Just tried to build the ros turtorial turtlesim package, which gives me the following error:

❯ colcon build
Starting >>> turtlesim
--- stderr: turtlesim
[cctools-port]: generating fake signature for '/Users/xiao/ros2_ws/install/turtlesim/lib/libturtlesim__rosidl_generator_c.dylib'
[cctools-port]: generating fake signature for '/Users/xiao/ros2_ws/install/turtlesim/lib/libturtlesim__rosidl_typesupport_fastrtps_c.dylib'
[cctools-port]: generating fake signature for '/Users/xiao/ros2_ws/install/turtlesim/lib/libturtlesim__rosidl_typesupport_fastrtps_cpp.dylib'
[cctools-port]: generating fake signature for '/Users/xiao/ros2_ws/install/turtlesim/lib/libturtlesim__rosidl_typesupport_introspection_c.dylib'
[cctools-port]: generating fake signature for '/Users/xiao/ros2_ws/install/turtlesim/lib/libturtlesim__rosidl_typesupport_c.dylib'
[cctools-port]: generating fake signature for '/Users/xiao/ros2_ws/install/turtlesim/lib/libturtlesim__rosidl_typesupport_introspection_cpp.dylib'
[cctools-port]: generating fake signature for '/Users/xiao/ros2_ws/install/turtlesim/lib/libturtlesim__rosidl_typesupport_cpp.dylib'
[cctools-port]: generating fake signature for '/Users/xiao/ros2_ws/install/turtlesim/lib/libturtlesim__rosidl_generator_py.dylib'
---
Finished <<< turtlesim [5.90s]

Summary: 1 package finished [6.39s]
  1 package had stderr output: turtlesim
(ros_env) 

Search for it but does not have much info about this.

Cheers

traversaro, 2023-12-17:
That print seems to be printed on stderr, but it does not seem a fatal error. Did you check if the executables you want to create were actually created?

traversaro, 2023-12-17:
Anyhow, for reproducibility can you please follow the template and provide your environment?

Tobias-Fischer, 2023-12-19:
Closing for now - feel free to reopen one additional information is provided.

closed 2023-12-19 by Tobias-Fischer