RoboStack/ros-humble #118: rviz2 Win11 missing entrypoint symbol

jkordani, 2023-12-18:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

rviz2 fails to start with 
[image]
[image]

I wish I could copy that text out and run it through c++filt, but alas.

I ran though the getting started guide, and chose to install ros-humble.  I elected not to install the vs command prompt.

### Installed packages

```shell
(ros2_env) C:\Users\joshu>mamba list
# packages in environment at C:\Users\joshu\miniforge3\envs\ros2_env:
#
# Name                    Version                   Build  Channel
aiohttp                   3.9.1           py310h8d17308_0    conda-forge
aiosignal                 1.3.1              pyhd8ed1ab_0    conda-forge
aom                       3.5.0                h63175ca_0    conda-forge
argcomplete               3.2.1              pyhd8ed1ab_0    conda-forge
assimp                    5.2.5                h4dcb625_0    conda-forge
async-timeout             4.0.3              pyhd8ed1ab_0    conda-forge
attrs                     23.1.0             pyh71513ae_1    conda-forge
blosc                     1.21.5               hdccc3a2_0    conda-forge
boost                     1.78.0          py310h4856b71_5    conda-forge
boost-cpp                 1.78.0               h9f4b32c_4    conda-forge
brotli                    1.1.0                hcfcfb64_1    conda-forge
brotli-bin                1.1.0                hcfcfb64_1    conda-forge
bullet                    3.21                 hbbfc1a7_4    conda-forge
bullet-cpp                3.21                 h1c4a608_4    conda-forge
bzip2                     1.0.8                hcfcfb64_5    conda-forge
c-compiler                1.7.0                hcfcfb64_0    conda-forge
ca-certificates           2023.11.17           h56e8100_0    conda-forge
cairo                     1.16.0            hd694305_1014    conda-forge
catkin_pkg                1.0.0              pyhd8ed1ab_0    conda-forge
catkin_tools              0.8.2              pyhd8ed1ab_0    conda-forge
certifi                   2023.11.17         pyhd8ed1ab_0    conda-forge
cffi                      1.16.0          py310h8d17308_0    conda-forge
cfitsio                   4.1.0                h5a969a9_0    conda-forge
clangdev                  5.0.0                   flang_3    conda-forge
cli11                     2.3.2                h63175ca_0    conda-forge
cmake                     3.27.4               h1537add_0    conda-forge
colcon-cmake              0.2.28             pyhd8ed1ab_0    conda-forge
colcon-common-extensions  0.3.0           py310h5588dad_1    conda-forge
colcon-core               0.15.0             pyhd8ed1ab_0    conda-forge
colcon-defaults           0.2.8              pyhd8ed1ab_0    conda-forge
colcon-devtools           0.2.5              pyhd8ed1ab_0    conda-forge
colcon-library-path       0.2.1                      py_0    conda-forge
colcon-metadata           0.2.5                      py_0    conda-forge
colcon-notification       0.2.15          py310h5588dad_3    conda-forge
colcon-output             0.2.13             pyhd8ed1ab_0    conda-forge
colcon-package-information 0.3.3                      py_0    conda-forge
colcon-package-selection  0.2.10                     py_0    conda-forge
colcon-parallel-executor  0.2.4                      py_0    conda-forge
colcon-pkg-config         0.1.0                      py_0    conda-forge
colcon-powershell         0.4.0              pyhd8ed1ab_0    conda-forge
colcon-python-setup-py    0.2.7                      py_0    conda-forge
colcon-recursive-crawl    0.2.3              pyhd8ed1ab_0    conda-forge
colcon-ros                0.4.1              pyhd8ed1ab_0    conda-forge
colcon-test-result        0.3.8                      py_0    conda-forge
colorama                  0.4.6              pyhd8ed1ab_0    conda-forge
coloredlogs               15.0.1             pyhd8ed1ab_3    conda-forge
compilers                 1.7.0                h57928b3_0    conda-forge
console_bridge            1.0.2                h5362a0b_1    conda-forge
contourpy                 1.2.0           py310h232114e_0    conda-forge
[... 632 lines cut ...]
         python version : 3.10.13.final.0
                 solver : libmamba (default)
       virtual packages : __archspec=1=x86_64
                          __conda=23.11.0=0
                          __cuda=12.2=0
                          __win=0=0
       base environment : C:\Users\joshu\miniforge3  (writable)
      conda av data dir : C:\Users\joshu\miniforge3\etc\conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/conda-forge/win-64
                          https://conda.anaconda.org/conda-forge/noarch
                          https://conda.anaconda.org/robostack-staging/win-64
                          https://conda.anaconda.org/robostack-staging/noarch
          package cache : C:\Users\joshu\miniforge3\pkgs
                          C:\Users\joshu\.conda\pkgs
                          C:\Users\joshu\AppData\Local\conda\conda\pkgs
       envs directories : C:\Users\joshu\miniforge3\envs
                          C:\Users\joshu\.conda\envs
                          C:\Users\joshu\AppData\Local\conda\conda\envs
               platform : win-64
             user-agent : conda/23.11.0 requests/2.31.0 CPython/3.10.13 Windows/10 Windows/10.0.22621 solver/libmamba conda-libmamba-solver/23.12.0 libmambapy/1.5.5
          administrator : False
             netrc file : None
           offline mode : False
```

Tobias-Fischer, 2024-02-16:
Hi @jkordani @traversaro - could you please check whether this is still an issue with the recently built packages?

Tobias-Fischer, 2024-02-17:
Awesome :)

closed 2024-02-17 by Tobias-Fischer