RoboStack/ros-humble #122: Package request: gazebo_ros

hoffnung8493, 2024-01-16:
### Package name

gazebo_ros

### Operating System(s)

- [ ] linux-64
- [ ] linux-aarch64
- [ ] osx-64
- [X] osx-arm64
- [ ] win-64

### Additional context (optional)

I want to setup ROS2 humble with Gazebo on my macbook pro M2 chip.
I have tried ubuntu 22.04 with VM using UTM. I have also tried dual boot with Asahi Ubuntu.
Both of them do not seem to support Gazebo GUI (only server mode).

Finally I tried Robostack with native Gazebo installation using homebrew. 
Both worked very well independently.
Typing `gazebo` on terminal, did open the GUI and it seemed to work well.
Also ROS2 worked well with the Robostack created environment. (I just followed the docs)

But when I run `ros2 launch gazebo_ros gazebo.launch.py`,

it returns this error: `Package 'gazebo_ros' not found: "package 'gazebo_ros' not found, searching: ['/Users/sihoonkim/miniconda3/envs/ros_env']"`

I tried to install it with `mamba install gazebo_ros`, but got:
```
(ros_env) sihoonkim@gimsihun-ui-MacBookPro-2 ~ % mamba install gazebo_ros_pkgs

Looking for: ['gazebo_ros_pkgs']

pkgs/main/osx-arm64                                           No change
conda-forge/noarch                                            No change
pkgs/r/noarch                                                 No change
pkgs/r/osx-arm64                                              No change
pkgs/main/noarch                                              No change
conda-forge/osx-arm64                                8.1MB @  29.5MB/s  0.3s
robostack-staging/osx-arm64                                   No change
robostack-staging/noarch                                      No change

Pinned packages:
  - python 3.10.*

Could not solve for environment specs
The following package could not be installed
└─ gazebo_ros_pkgs does not exist (perhaps a typo or a missing channel).
```

Tobias-Fischer, 2024-01-16:
Please `mamba install ros-humble-gazebo-ros`.

closed 2024-01-16 by Tobias-Fischer

hoffnung8493, 2024-01-17:
@Tobias-Fischer Thanks for the fast response. It does seem to work on OSX M2 quite nicely! No need to run VM or dual boot ubuntu. Is there any documentation on this? and maybe a list of unsupported (or not yet) features for OSX ARM64?