RoboStack/ros-humble #130: Package request: gazebo_ros

Nir0n, 2024-02-03:
### Package name

gazebo_ros

### Operating System(s)

- [ ] linux-64
- [ ] linux-aarch64
- [ ] osx-64
- [ ] osx-arm64
- [ ] win-64

### Additional context (optional)

required for running ros with gazebo (as far as I know)
I don't test it on linux or windows but I think it platform agnostic

### Links and references (optional)

https://stackoverflow.com/questions/77933813/cannot-find-gazebo-ros-package-in-robostack-how-run-gazebo-with-ros2-humble-in

Tobias-Fischer, 2024-02-06:
In ROS2, one should use ros-ign-bridge (https://docs.ros.org/en/humble/Tutorials/Advanced/Simulators/Gazebo/Gazebo.html). This is available: https://anaconda.org/robostack-staging/ros-humble-ros-ign-bridge

Nir0n, 2024-02-06:
Got you, sorry for the incorrect query, sir.

Tobias-Fischer, 2024-02-06:
@traversaro - you are the gazebo expert. Do you have any ideas?

traversaro, 2024-02-06:
Can you provide mamba list of the environment you are using, and ideally also the command through which you created it?

Nir0n, 2024-02-06:
Greetings, gentlemen.
if you mean list of environments:
```
(ros_env) vaallakhverdyants@MSK-C02DW5J1MD6R ros_ws % conda info --envs
# conda environments:
#
                         /Users/vaallakhverdyants/miniconda3
base                     /usr/local/Caskroom/miniforge/base
jupyros                  /usr/local/Caskroom/miniforge/base/envs/jupyros
ros_env               *  /usr/local/Caskroom/miniforge/base/envs/ros_env
```
I use ros_env.
If you mean list of packages in environment - provided file:
[ros_env_pkgs.txt](https://github.com/RoboStack/ros-humble/files/14179685/ros_env_pkgs.txt)

about creation of environment: I use this tutorial https://robostack.github.io/GettingStarted.html#__tabbed_1_1
so the command was `mamba create -n ros_env`
Thank you again, gentlemen, for your support!

traversaro, 2024-02-06:
I was referring to how to installed the packages that ended in the environment, anyhow I guess that the command was `mamba install ros-humble-desktop` and later `mamba install ros-humble-ros-ign-bridge` . However, this actually fails for some reason, at least on linux-64 . Then I tried to install:
~~~
mamba create -n ros_env -c conda-forge -c robostack-staging ros-humble-desktop ros-humble-ros-ign-bridge
~~~
and that seems to be working fine, even if it  install the latest Python 3.11 packages instead of the Python 3.10 packages you have. However, also in this case `ign gazebo` fails. Indeed, `ros-humble-ros-ign-bridge` actually depends only on ign-transport, not on Ignition Gazebo itself. So to actually run Ignition Gazebo, run:
~~~
mamba install libignition-gazebo6
~~~

Feel free to report any problem!

traversaro, 2024-02-06:
> I was referring to how to installed the packages that ended in the environment, anyhow I guess that the command was `mamba install ros-humble-desktop` and later `mamba install [...]

I think I got the problem. The actual problem that users should install (and that depends on libignition-gazebo6) is `ros-humble-ros-gz` (or `ros-humble-ros-ign`). However, the package at the moment is available only on linux-64, not sure what is the blocker on other platforms (perhaps we should just try to build it, not sure).

traversaro, 2024-02-06:
To reply to the original question (that was asking about ROS2 integration with "Classic" Gazebo, not ROS2 integration with "Modern" ign/gz), the correct package to install is `ros-humble-gazebo-ros`, and it seems that the latest version of it is available for `linux-64`, `osx-64` and `osx-arm64`, see https://anaconda.org/robostack-staging/ros-humble-gazebo-ros/files .

Nir0n, 2024-02-06:
thank you, sir!
some clarification: i DONT want to install integration with gazebo classic, I was just poorly informed how create integration between ros and gazebo in new versions, and I incorrectly assumed that for it I require "gazebo_ros".

Just to understand that I still with you on this topic,
in order to run ros-humble with gazebo I need:
1) I need to install `libignition-gazebo6` package with command that you provided
2) I need to install `ros-humble-ros-gz` package, but for now its not available on osx, so instead I will install older version  `ros-humble-gazebo-ros`
Everything correct?
And , please if you have opportunity, can you provide launch.py file for running my model with gazebo?
Thank you!

traversaro, 2024-02-06:
Just to avoid confusion, please refer either to "Classic" Gazebo or "Modern" ign/gz, as otherwise any reference to "gazebo" is ambiguous, thanks!

Nir0n, 2024-02-07:
Summarize: gazebo_ros can be installed for ros humble like this: `mamba install ros-humble-gazebo-ros`
regarding question for launch.py file for "modern" (ign/gz) version lets discuss [here](https://stackoverflow.com/questions/77933813/cannot-find-gazebo-ros-package-in-robostack-how-run-gazebo-with-ros2-humble-in) if you have time, gentelmen.
Thank you for your time!