RoboStack/ros-humble #133: Package request: ros-humble-slam-toolbox

aptxlee, 2024-02-06:
### Package name

ros-humble-slam-toolbox

### Operating System(s)

- [ ] linux-64
- [ ] linux-aarch64
- [ ] osx-64
- [ ] osx-arm64
- [X] win-64

### Additional context (optional)

It is a package that can run in Windows binaries, and must be supported to perform full nav2 functions.

Tobias-Fischer, 2024-02-16:
This currently fails as csparse cannot be found. Note that the last version of suitesparse on Windows is very old, see https://github.com/conda-forge/suitesparse-feedstock/issues/51 (but it seems feasible to build as that's what they do in cvxopt)

/cc @traversaro

edvart-ros, 2024-03-25:
Bump! @traversaro is this doable?

traversaro, 2024-05-04:
> As reported in [conda-forge/suitesparse-feedstock#51](https://github.com/conda-forge/suitesparse-feedstock/issues/51) updating suitesparse on windows seems doable, however I do not have direct interest on doing this at the moment, so I can't work on this myself. If anyone wants to volunteer to work on this I would be happy to advise and help.

Ok, actually there was no need to update suitesparse, the fix necessary to find suitesparse was just https://github.com/SteveMacenski/slam_toolbox/pull/699 .

I looked a bit on the remaining failure, but it seems that the package requires several changes to build on Windows, see https://github.com/SteveMacenski/slam_toolbox/pull/440/files . However, to correctly support `stack_size_to_use` parameter, it is necessary to do some further changes, described in https://github.com/SteveMacenski/slam_toolbox/pull/439#discussion_r728201871 . At the moment I do not plan to work on this, but if anyone wants to work on it I will be happy to advice and/or help. 

It also necessary to have compatibility with the latest ceres, but this is given by https://github.com/SteveMacenski/slam_toolbox/pull/687 .

Tobias-Fischer, 2025-01-11:
See #238

closed 2025-01-11 by Tobias-Fischer