RoboStack/ros-humble #140: Cannot install newer nav2 packages cause of conflicts

jakub-kozlowicz, 2024-02-12:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

When trying to install newer nav2 packages (like 1.1.13) there is a conflict and packages cannot be installed.

```bash
❯ mamba install ros-humble-nav2-msgs=1.1.13

Looking for: ['ros-humble-nav2-msgs=1.1.13']

conda-forge/osx-arm64                                       Using cache
conda-forge/noarch                                          Using cache
robostack-staging/noarch                                      No change
robostack-staging/osx-arm64                                   No change

Pinned packages:
  - python 3.10.*

warning  libmamba Added empty dependency for problem type SOLVER_RULE_UPDATE
Could not solve for environment specs
The following packages are incompatible
├─ jpeg is requested and can be installed;
├─ ros-humble-cv-bridge is installable with the potential options
│  ├─ ros-humble-cv-bridge 3.2.1 would require
│  │  └─ libopencv >=4.9.0,<4.9.1.0a0  with the potential options
│  │     ├─ libopencv 4.9.0 would require
│  │     │  └─ jasper [>=4.1.1,<5.0a0 |>=4.1.2,<5.0a0 ], which requires
│  │     │     └─ libjpeg-turbo >=3.0.0,<4.0a0 , which requires
│  │     │        └─ jpeg <0.0.0a , which conflicts with any installable versions previously reported;
│  │     ├─ libopencv 4.9.0 would require
│  │     │  └─ python >=3.11,<3.12.0a0 *_cpython, which can be installed;
│  │     ├─ libopencv 4.9.0 would require
│  │     │  └─ python >=3.12,<3.13.0a0 *_cpython, which can be installed;
│  │     ├─ libopencv 4.9.0 would require
│  │     │  └─ python >=3.8,<3.9.0a0 *_cpython, which can be installed;
│  │     └─ libopencv 4.9.0 would require
│  │        └─ python >=3.9,<3.10.0a0 *_cpython, which can be installed;
│  ├─ ros-humble-cv-bridge 3.2.1 would require
│  │  └─ ros2-distro-mutex 0.3.* , which can be installed;
│  └─ ros-humble-cv-bridge 3.2.1 would require
│     └─ ros2-distro-mutex 0.2.* humble, which can be installed;
└─ ros-humble-nav2-msgs 1.1.13**  is not installable because it requires
   └─ ros2-distro-mutex 0.5.* , which conflicts with any installable versions previously reported.
```

```bash
❯ mamba install ros-humble-nav2-simple-commander=1.1.13

Looking for: ['ros-humble-nav2-simple-commander=1.1.13']

conda-forge/osx-arm64                                       Using cache
conda-forge/noarch                                          Using cache
robostack-staging/osx-arm64                                   No change
robostack-staging/noarch                                      No change

Pinned packages:
  - python 3.10.*

warning  libmamba Added empty dependency for problem type SOLVER_RULE_UPDATE
Could not solve for environment specs
The following packages are incompatible
├─ jpeg is requested and can be installed;
├─ ros-humble-cv-bridge is installable with the potential options
│  ├─ ros-humble-cv-bridge 3.2.1 would require
│  │  └─ libopencv >=4.9.0,<4.9.1.0a0  with the potential options
│  │     ├─ libopencv 4.9.0 would require
│  │     │  └─ jasper [>=4.1.1,<5.0a0 |>=4.1.2,<5.0a0 ], which requires
[... 646 lines cut ...]
          conda version : 23.11.0
    conda-build version : not installed
         python version : 3.10.13.final.0
                 solver : libmamba (default)
       virtual packages : __archspec=1=m1
                          __conda=23.11.0=0
                          __osx=14.3.1=0
                          __unix=0=0
       base environment : /opt/homebrew/Caskroom/miniforge/base  (writable)
      conda av data dir : /opt/homebrew/Caskroom/miniforge/base/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack-staging/osx-arm64
                          https://conda.anaconda.org/robostack-staging/noarch
                          https://conda.anaconda.org/conda-forge/osx-arm64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : /opt/homebrew/Caskroom/miniforge/base/pkgs
                          /Users/jakub/.conda/pkgs
       envs directories : /opt/homebrew/Caskroom/miniforge/base/envs
                          /Users/jakub/.conda/envs
               platform : osx-arm64
             user-agent : conda/23.11.0 requests/2.31.0 CPython/3.10.13 Darwin/23.3.0 OSX/14.3.1 solver/libmamba conda-libmamba-solver/23.12.0 libmambapy/1.5.6
                UID:GID : 502:20
             netrc file : None
           offline mode : False
```

Tobias-Fischer, 2024-02-12:
Hi @jakub-kozlowicz  - you’ll need to update to Python 3.11 :). Just add python=3.11 to your mamba install.

jakub-kozlowicz, 2024-02-12:
With Python 3.11 the packages cannot be build due to deprecated build system. Python 3.11 dead not allow direct `setup.py` call and require to use the standard build system.

Tobias-Fischer, 2024-02-12:
Which packages? We’re using direct setup.py calls.

jakub-kozlowicz, 2024-02-12:
I have my custom ros packages (https://git.kcir.pwr.edu.pl/turtlebot2/mmrs-controller) and when I invoked `colcon build` I've received:

```
❯ colcon build
Starting >>> mmrs_msgs
--- stderr: mmrs_msgs
CMake Warning (dev) at /opt/homebrew/Caskroom/miniforge/base/envs/mmrs-controller/share/rosidl_generator_py/cmake/rosidl_generator_py_generate_interfaces.cmake:20 (find_package):
  Policy CMP0148 is not set: The FindPythonInterp and FindPythonLibs modules
  are removed.  Run "cmake --help-policy CMP0148" for policy details.  Use
  the cmake_policy command to set the policy and suppress this warning.

Call Stack (most recent call first):
  /opt/homebrew/Caskroom/miniforge/base/envs/mmrs-controller/share/ament_cmake_core/cmake/core/ament_execute_extensions.cmake:48 (include)
  /opt/homebrew/Caskroom/miniforge/base/envs/mmrs-controller/share/rosidl_cmake/cmake/rosidl_generate_interfaces.cmake:286 (ament_execute_extensions)
  CMakeLists.txt:17 (rosidl_generate_interfaces)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at /opt/homebrew/Caskroom/miniforge/base/envs/mmrs-controller/share/python_cmake_module/cmake/Modules/FindPythonExtra.cmake:52 (find_package):
  Policy CMP0148 is not set: The FindPythonInterp and FindPythonLibs modules
  are removed.  Run "cmake --help-policy CMP0148" for policy details.  Use
  the cmake_policy command to set the policy and suppress this warning.

Call Stack (most recent call first):
  /opt/homebrew/Caskroom/miniforge/base/envs/mmrs-controller/share/rosidl_generator_py/cmake/rosidl_generator_py_generate_interfaces.cmake:23 (find_package)
  /opt/homebrew/Caskroom/miniforge/base/envs/mmrs-controller/share/ament_cmake_core/cmake/core/ament_execute_extensions.cmake:48 (include)
  /opt/homebrew/Caskroom/miniforge/base/envs/mmrs-controller/share/rosidl_cmake/cmake/rosidl_generate_interfaces.cmake:286 (ament_execute_extensions)
  CMakeLists.txt:17 (rosidl_generate_interfaces)
This warning is for project developers.  Use -Wno-dev to suppress it.

[cctools-port]: generating fake signature for '/Users/jakub/Projects/mmrs-controller/install/mmrs_msgs/lib/libmmrs_msgs__rosidl_typesupport_fastrtps_c.dylib'
[cctools-port]: generating fake signature for '/Users/jakub/Projects/mmrs-controller/install/mmrs_msgs/lib/libmmrs_msgs__rosidl_typesupport_introspection_c.dylib'
[cctools-port]: generating fake signature for '/Users/jakub/Projects/mmrs-controller/install/mmrs_msgs/lib/libmmrs_msgs__rosidl_typesupport_c.dylib'
[cctools-port]: generating fake signature for '/Users/jakub/Projects/mmrs-controller/install/mmrs_msgs/lib/python3.11/site-packages/mmrs_msgs/mmrs_msgs_s__rosidl_typesupport_fastrtps_c.cpython-311-darwin.so'
[cctools-port]: generating fake signature for '/Users/jakub/Projects/mmrs-controller/install/mmrs_msgs/lib/python3.11/site-packages/mmrs_msgs/mmrs_msgs_s__rosidl_typesupport_introspection_c.cpython-311-darwin.so'
[cctools-port]: generating fake signature for '/Users/jakub/Projects/mmrs-controller/install/mmrs_msgs/lib/python3.11/site-packages/mmrs_msgs/mmrs_msgs_s__rosidl_typesupport_c.cpython-311-darwin.so'
[cctools-port]: generating fake signature for '/Users/jakub/Projects/mmrs-controller/install/mmrs_msgs/lib/libmmrs_msgs__rosidl_generator_py.dylib'
---
Finished <<< mmrs_msgs [14.0s]
Starting >>> mmrs_controller
--- stderr: mmrs_controller
/opt/homebrew/Caskroom/miniforge/base/envs/mmrs-controller/lib/python3.11/site-packages/setuptools/_distutils/cmd.py:66: SetuptoolsDeprecationWarning: setup.py install is deprecated.
!!

        ********************************************************************************
        Please avoid running ``setup.py`` directly.
        Instead, use pypa/build, pypa/installer or other
        standards-based tools.

        See https://blog.ganssle.io/articles/2021/10/setup-py-deprecated.html for details.
        ********************************************************************************

!!
  self.initialize_options()
---
Finished <<< mmrs_controller [1.16s]

Summary: 2 packages finished [15.8s]
  2 packages had stderr output: mmrs_controller mmrs_msgs
```

Somehow earlier I thought that this is error and the build did not complete.

Thanks for help :)

Tobias-Fischer, 2024-02-12:
Yes, there are a lot of warnings that look like errors, but it should all be fine. Soon ROS2 upstream will need to avoid using setup.py as in Python 3.12 it is indeed not an option anymore, and Ubuntu 24.04 will use Python 3.12 (just repeating @traversaro here from another conversation that I can’t find at the moment).

Feel free to reopen if you have more issues.

closed 2024-02-12 by Tobias-Fischer

traversaro, 2024-02-12:
> Soon ROS2 upstream will need to avoid using setup.py as in Python 3.12 it is indeed not an option anymore, and Ubuntu 24.04 will use Python 3.12 (just repeating @traversaro here from another conversation that I can’t find at the moment).

For reference: https://matrix.to/#/!rdzftyfSnEcdOEyuoI:gitter.im/$tQzsI8lr8hHGOfZoTJBu00uyd8_5QsCINgZ3tDU0C2o?via=gitter.im&via=matrix.org .