RoboStack/ros-humble #149: Custom Msgs package won't build on Mac M1; error with ros_idl_generators dependency

apr600, 2024-02-25:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

Build fails for any package that requires building custom interfaces (including the custom_interfaces tutorial [here](https://docs.ros.org/en/humble/Tutorials/Beginner-Client-Libraries/Custom-ROS2-Interfaces.html)), with error: 
```
CMake Error at /Users/ahalyaprabhakar/miniconda3/envs/ros2_humble_base_env/share/cmake-3.28/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find Python3 (missing: Python3_NumPy_INCLUDE_DIRS NumPy) (found
  version "3.12.1")
Call Stack (most recent call first):
  /Users/ahalyaprabhakar/miniconda3/envs/ros2_humble_base_env/share/cmake-3.28/Modules/FindPackageHandleStandardArgs.cmake:600 (_FPHSA_FAILURE_MESSAGE)
  /Users/ahalyaprabhakar/miniconda3/envs/ros2_humble_base_env/share/cmake-3.28/Modules/FindPython/Support.cmake:3867 (find_package_handle_standard_args)
  /Users/ahalyaprabhakar/miniconda3/envs/ros2_humble_base_env/share/cmake-3.28/Modules/FindPython3.cmake:545 (include)
  /Users/ahalyaprabhakar/miniconda3/envs/ros2_humble_base_env/share/rosidl_generator_py/cmake/rosidl_generator_py-extras.cmake:2 (find_package)
  /Users/ahalyaprabhakar/miniconda3/envs/ros2_humble_base_env/share/rosidl_generator_py/cmake/rosidl_generator_pyConfig.cmake:41 (include)
  /Users/ahalyaprabhakar/miniconda3/envs/ros2_humble_base_env/share/rosidl_default_generators/cmake/rosidl_default_generators-extras.cmake:21 (find_package)
  /Users/ahalyaprabhakar/miniconda3/envs/ros2_humble_base_env/share/rosidl_default_generators/cmake/rosidl_default_generatorsConfig.cmake:41 (include)
  CMakeLists.txt:15 (find_package)
```

It seems to be an error with the rosidl_generators packages for building the custom interfaces and only seems to be an error on Mac computers- I have tested it on multiple Linux, Windows WSL and Mac computers. This was only an issue on Mac computers, and it happened on multiple Mac computers (possibly all Mac M* computers). 

### Installed packages

```shell
# packages in environment at /Users/ahalyaprabhakar/miniconda3/envs/ros2_humble_base_env:
#
# Name                    Version                   Build  Channel
aiohttp                   3.9.3           py311h05b510d_0    conda-forge
aiosignal                 1.3.1              pyhd8ed1ab_0    conda-forge
aom                       3.8.1                h078ce10_0    conda-forge
argcomplete               3.2.2              pyhd8ed1ab_0    conda-forge
assimp                    5.3.1                he63ff86_2    conda-forge
atk-1.0                   2.38.0               hcb7b3dd_1    conda-forge
attrs                     23.2.0             pyh71513ae_0    conda-forge
blosc                     1.21.5               hc338f07_0    conda-forge
boost                     1.82.0               hf3b2ce4_6    conda-forge
brotli                    1.1.0                hb547adb_1    conda-forge
brotli-bin                1.1.0                hb547adb_1    conda-forge
bullet                    3.25                 hf3b2ce4_2    conda-forge
bullet-cpp                3.25            py311h6e08293_2    conda-forge
bzip2                     1.0.8                h93a5062_5    conda-forge
c-ares                    1.27.0               h93a5062_0    conda-forge
c-compiler                1.7.0                h6aa9301_0    conda-forge
ca-certificates           2024.2.2             hf0a4a13_0    conda-forge
cairo                     1.18.0               hd1e100b_0    conda-forge
catkin_pkg                1.0.0              pyhd8ed1ab_0    conda-forge
catkin_tools              0.9.4              pyhd8ed1ab_0    conda-forge
cctools                   973.0.1             h4faf515_16    conda-forge
cctools_osx-arm64         973.0.1             h62378fb_16    conda-forge
certifi                   2024.2.2           pyhd8ed1ab_0    conda-forge
cffi                      1.16.0          py311h4a08483_0    conda-forge
clang                     16.0.6               h30cc82d_5    conda-forge
clang-16                  16.0.6          default_he012953_5    conda-forge
clang_impl_osx-arm64      16.0.6               hc421ffc_9    conda-forge
clang_osx-arm64           16.0.6               h54d7cd3_9    conda-forge
clangxx                   16.0.6          default_h4cf2255_5    conda-forge
clangxx_impl_osx-arm64    16.0.6               hcd7bac0_9    conda-forge
clangxx_osx-arm64         16.0.6               h54d7cd3_9    conda-forge
cmake                     3.28.3               h50fd54c_0    conda-forge
colcon-argcomplete        0.3.3              pyhd8ed1ab_0    conda-forge
colcon-bash               0.5.0              pyhd8ed1ab_0    conda-forge
colcon-cd                 0.1.1              pyhd8ed1ab_0    conda-forge
colcon-cmake              0.2.28             pyhd8ed1ab_0    conda-forge
colcon-common-extensions  0.3.0           py311h267d04e_1    conda-forge
colcon-core               0.15.0             pyhd8ed1ab_0    conda-forge
colcon-defaults           0.2.8              pyhd8ed1ab_0    conda-forge
colcon-devtools           0.2.5              pyhd8ed1ab_0    conda-forge
[... 595 lines cut ...]
          conda version : 23.11.0
    conda-build version : not installed
         python version : 3.11.7.final.0
                 solver : libmamba (default)
       virtual packages : __archspec=1=m2
                          __conda=23.11.0=0
                          __osx=14.3.1=0
                          __unix=0=0
       base environment : /Users/ahalyaprabhakar/miniconda3  (writable)
      conda av data dir : /Users/ahalyaprabhakar/miniconda3/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack-staging/osx-arm64
                          https://conda.anaconda.org/robostack-staging/noarch
                          https://conda.anaconda.org/conda-forge/osx-arm64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : /Users/ahalyaprabhakar/miniconda3/pkgs
                          /Users/ahalyaprabhakar/.conda/pkgs
       envs directories : /Users/ahalyaprabhakar/miniconda3/envs
                          /Users/ahalyaprabhakar/.conda/envs
               platform : osx-arm64
             user-agent : conda/23.11.0 requests/2.31.0 CPython/3.11.7 Darwin/23.3.0 OSX/14.3.1 solver/libmamba conda-libmamba-solver/23.12.0 libmambapy/1.5.6
                UID:GID : 501:20
             netrc file : None
           offline mode : False
```

Tobias-Fischer, 2024-02-25:
Can you please check the latest FAQ item at https://robostack.github.io/FAQ.html and in addition pass `DPython3_FIND_STRATEGY=LOCATION`?

apr600, 2024-02-28:
That worked, thanks!

closed 2024-02-28 by apr600

Tobias-Fischer, 2024-02-28:
Thanks, I've updated the FAQs: https://github.com/RoboStack/robostack.github.io/commit/4b0343aa520791831d41275214650866c94c98fa