RoboStack/ros-humble #150: Demo Python scripts not running in conda - "No executable found"

TSoli, 2024-02-25:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

Hi, I followed the getting started guide to install ros-humble with mamba however it seems that the demo python packages are not working (I have not tried with building my own yet so can't confirm if it is all python packages). The cpp example packages do work though. For the below, the I made a mamba environment called test which I then activated. Now when I try the example from the ros2 humble docs:

```bash
# terminal 1
ros2 run demo_nodes_cpp talker
# terminal 2
ros2 run demo_nodes_py listener
```

The output from terminal 2 is "No executable found".

I think this is likely related to #41 but I can't tell if this was supposed to be fixed or there was a good workaround.

### Installed packages

```shell
# packages in environment at /home/tariq/miniconda3/envs/test:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
aiohttp                   3.9.3           py311h459d7ec_0    conda-forge
aiosignal                 1.3.1              pyhd8ed1ab_0    conda-forge
alsa-lib                  1.2.10               hd590300_0    conda-forge
aom                       3.8.1                h59595ed_0    conda-forge
argcomplete               3.2.2              pyhd8ed1ab_0    conda-forge
assimp                    5.3.1                hfb0e8fe_2    conda-forge
atk-1.0                   2.38.0               hd4edc92_1    conda-forge
attr                      2.5.1                h166bdaf_1    conda-forge
attrs                     23.2.0             pyh71513ae_0    conda-forge
blosc                     1.21.5               h0f2a231_0    conda-forge
boost                     1.82.0               h781c19f_6    conda-forge
brotli                    1.1.0                hd590300_1    conda-forge
brotli-bin                1.1.0                hd590300_1    conda-forge
bullet                    3.25                 h781c19f_2    conda-forge
bullet-cpp                3.25                 h320fe9a_2    conda-forge
bzip2                     1.0.8                hd590300_5    conda-forge
c-ares                    1.27.0               hd590300_0    conda-forge
ca-certificates           2024.2.2             hbcca054_0    conda-forge
cairo                     1.18.0               h3faef2a_0    conda-forge
catkin_pkg                1.0.0              pyhd8ed1ab_0    conda-forge
certifi                   2024.2.2           pyhd8ed1ab_0    conda-forge
cffi                      1.16.0          py311hb3a22ac_0    conda-forge
cmake                     3.28.3               hcfe8598_0    conda-forge
colorama                  0.4.6              pyhd8ed1ab_0    conda-forge
console_bridge            1.0.2                h924138e_1    conda-forge
contourpy                 1.2.0           py311h9547e67_0    conda-forge
cppcheck                  2.13.1          py311hc954e77_0    conda-forge
cryptography              42.0.5          py311h63ff55d_0    conda-forge
cycler                    0.12.1             pyhd8ed1ab_0    conda-forge
dav1d                     1.2.1                hd590300_0    conda-forge
dbus                      1.13.6               h5008d03_3    conda-forge
distro                    1.9.0              pyhd8ed1ab_0    conda-forge
docutils                  0.20.1          py311h38be061_3    conda-forge
double-conversion         3.3.0                h59595ed_0    conda-forge
eigen                     3.4.0                h00ab1b0_0    conda-forge
empy                      3.3.4              pyh9f0ad1d_1    conda-forge
exceptiongroup            1.2.0              pyhd8ed1ab_2    conda-forge
expat                     2.5.0                hcb278e6_1    conda-forge
ffmpeg                    6.1.1           gpl_h8007c5b_104    conda-forge
flake8                    7.0.0              pyhd8ed1ab_0    conda-forge
flann                     1.9.2                h2b5ea80_0    conda-forge
fmt                       10.2.1               h00ab1b0_0    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
[... 575 lines cut ...]
    conda-build version : not installed
         python version : 3.11.5.final.0
                 solver : libmamba (default)
       virtual packages : __archspec=1=skylake
                          __conda=23.11.0=0
                          __glibc=2.35=0
                          __linux=5.15.0=0
                          __unix=0=0
       base environment : /home/tariq/miniconda3  (writable)
      conda av data dir : /home/tariq/miniconda3/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack-staging/linux-64
                          https://conda.anaconda.org/robostack-staging/noarch
                          https://conda.anaconda.org/conda-forge/linux-64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : /home/tariq/miniconda3/pkgs
                          /home/tariq/.conda/pkgs
       envs directories : /home/tariq/miniconda3/envs
                          /home/tariq/.conda/envs
               platform : linux-64
             user-agent : conda/23.11.0 requests/2.31.0 CPython/3.11.5 Linux/5.15.0-97-generic linuxmint/21.3 glibc/2.35 solver/libmamba conda-libmamba-solver/23.12.0 libmambapy/1.5.6
                UID:GID : 1000:1000
             netrc file : None
           offline mode : False
```

Tobias-Fischer, 2024-02-25:
Ah gosh, it seems like we got it wrong - again :(. Any ideas on how to best proceed @traversaro? It's such a pain that we can use `pip` for everything.

traversaro, 2024-02-26:
I think this is another problem I noticed while reviewing patches. There are a bunch of patches that just rename executables/scrips, probably to avoid clobbering when executables/scripts were not installed in package-specific location. However, now that they are, I guess we can just drop this patches, for example https://github.com/RoboStack/ros-humble/blob/main/patch/ros-humble-demo-nodes-py.patch . As a workaround one can call 

~~~
ros2 run demo_nodes_py demo_listener
~~~

Until the issue is fixed.

traversaro, 2024-02-26:
Other similar patches that we can just drop:

* https://github.com/RoboStack/ros-humble/blob/502ae3e160271b8eb240adf244b96ecca1c80f2a/patch/ros-humble-examples-rclpy-minimal-client.patch#L8
* https://github.com/RoboStack/ros-humble/blob/502ae3e160271b8eb240adf244b96ecca1c80f2a/patch/ros-humble-examples-rclpy-minimal-action-client.patch#L8

TSoli, 2024-02-26:
> I think this is another problem I noticed while reviewing patches. There are a bunch of patches that just rename executables/scrips, probably to avoid clobbering when executables/scripts were not [...]

Amazing! Thanks so much for the extremely quick replies. This seems to work. To confirm, this won't likely cause problems with building my own python packages?

traversaro, 2024-02-26:
> Amazing! Thanks so much for the extremely quick replies. This seems to work. To confirm, this won't likely cause problems with building my own python packages?

It should not.

Tobias-Fischer, 2024-02-26:
Ah awesome, that's good news, thanks a lot for clarifying @traversaro! Have removed the patches; will be fixed after the next rebuild :)

closed 2025-01-06 by Tobias-Fischer