RoboStack/ros-humble #153: Package request: ros-humble-g2o

TSoli, 2024-02-28:
### Package name

ros-humble-g2o

### Operating System(s)

- [X] linux-64
- [ ] linux-aarch64
- [ ] osx-64
- [ ] osx-arm64
- [ ] win-64

### Additional context (optional)

Hi, I would like to use [g2o](https://github.com/RainerKuemmerle/g2o) as it is a dependency for [stella_vslam](https://github.com/stella-cv/stella_vslam) which I would like to use for a project involving visual SLAM. It seems to be a fairly popular package so I assume it could be broadly useful for other projects too.

### Links and references (optional)

https://github.com/RainerKuemmerle/g2o

traversaro, 2024-02-28:
Are you sure that the package has been released in Humble? I can't find it in index.ros.org . Note that there is a g2o package in conda-forge, did you tried if it fits your needs?

TSoli, 2024-02-28:
> Are you sure that the package has been released in Humble? I can't find it in index.ros.org . Note that there is a g2o package in conda-forge, did you tried if it fits your needs?

I can see it when I run `apt-cache search ros-humble-libg2o`.  I get
```
ros-humble-libg2o - The libg2o library from http://openslam.org/g2o.html
ros-humble-libg2o-dbgsym - debug symbols for ros-humble-libg2o
```

I'm assuming the conda-forge one you're talking about is `conda-forge::g2o`. I tried installing this but when I use `rosdep` for `stella_vslam` it seems it is looking for `ros-humble-libg2o`

```
executing command [mamba install -c robostack-staging -c conda-forge -y ros-humble-libg2o sqlite=3.*]

Looking for: ['ros-humble-libg2o', 'sqlite=3']

conda-forge/linux-64                                        Using cache
conda-forge/noarch                                          Using cache
robostack-staging/linux-64                                    No change
robostack-staging/noarch                                      No change

Pinned packages:
  - python 3.11.*

Could not solve for environment specs
The following package could not be installed
└─ ros-humble-libg2o does not exist (perhaps a typo or a missing channel).
```

Tobias-Fischer, 2024-02-28:
Done in #154

closed 2024-02-28 by Tobias-Fischer