RoboStack/ros-humble #160: While building a ros2 package encountering python3 not found error

mavneo, 2024-03-19:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

- I have installed the ROS2 mamba env as give the installation guide. While building a package, I find this error.

```
CMake Error at /Users/manojbin2/miniforge3/envs/ros_humble/share/cmake-3.25/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find Python3 (missing: Python3_NumPy_INCLUDE_DIRS NumPy) (found
  version "3.12.0")
```

### Installed packages

```shell
# packages in environment at /Users/manojbin2/miniforge3/envs/ros_humble:
#
# Name                    Version                   Build  Channel
aom                       3.5.0                hf0c8a7f_0    conda-forge
argcomplete               3.2.3              pyhd8ed1ab_0    conda-forge
assimp                    5.2.4                hfcdac6d_0    conda-forge
boost                     1.74.0           py39ha1f3e3e_5    conda-forge
boost-cpp                 1.74.0               h8b082ac_8    conda-forge
brotli                    1.1.0                h0dc2134_1    conda-forge
brotli-bin                1.1.0                h0dc2134_1    conda-forge
bullet                    3.21                 h2a79d86_4    conda-forge
bullet-cpp                3.21                 hc6b5a04_4    conda-forge
bzip2                     1.0.8                h10d778d_5    conda-forge
c-ares                    1.27.0               h10d778d_0    conda-forge
c-compiler                1.7.0                h282daa2_0    conda-forge
ca-certificates           2024.2.2             h8857fd0_0    conda-forge
cairo                     1.16.0            h904041c_1014    conda-forge
catkin_pkg                1.0.0              pyhd8ed1ab_0    conda-forge
cctools                   973.0.1             h40f6528_13    conda-forge
cctools_osx-64            973.0.1             h5676edf_13    conda-forge
certifi                   2024.2.2           pyhd8ed1ab_0    conda-forge
cffi                      1.16.0           py39h18ef598_0    conda-forge
clang                     16.0.1               h49458c0_0    conda-forge
clang-16                  16.0.1          default_hd25f202_0    conda-forge
clang_osx-64              16.0.1               hac0a0c4_1    conda-forge
clangxx                   16.0.1          default_hd25f202_0    conda-forge
clangxx_osx-64            16.0.1               h950475c_1    conda-forge
cli11                     2.4.1                h73e2aa4_0    conda-forge
cmake                     3.25.2               h4032537_0    conda-forge
colcon-argcomplete        0.3.3              pyhd8ed1ab_0    conda-forge
colcon-bash               0.5.0              pyhd8ed1ab_0    conda-forge
colcon-cd                 0.1.1              pyhd8ed1ab_0    conda-forge
colcon-cmake              0.2.28             pyhd8ed1ab_0    conda-forge
colcon-common-extensions  0.3.0            py39h6e9494a_1    conda-forge
colcon-core               0.16.0             pyhd8ed1ab_0    conda-forge
colcon-defaults           0.2.8              pyhd8ed1ab_0    conda-forge
colcon-devtools           0.3.0              pyhd8ed1ab_0    conda-forge
colcon-library-path       0.2.1                      py_0    conda-forge
colcon-metadata           0.2.5                      py_0    conda-forge
colcon-output             0.2.13             pyhd8ed1ab_0    conda-forge
colcon-package-information 0.4.0              pyhd8ed1ab_0    conda-forge
colcon-package-selection  0.2.10                     py_0    conda-forge
colcon-parallel-executor  0.2.4                      py_0    conda-forge
colcon-pkg-config         0.1.0                      py_0    conda-forge
colcon-powershell         0.4.0              pyhd8ed1ab_0    conda-forge
colcon-python-setup-py    0.2.7                      py_0    conda-forge
colcon-recursive-crawl    0.2.3              pyhd8ed1ab_0    conda-forge
colcon-ros                0.4.1              pyhd8ed1ab_0    conda-forge
colcon-test-result        0.3.8                      py_0    conda-forge
colcon-zsh                0.5.0              pyhd8ed1ab_0    conda-forge
colorama                  0.4.6              pyhd8ed1ab_0    conda-forge
coloredlogs               15.0.1             pyhd8ed1ab_3    conda-forge
compiler-rt               16.0.1               h613da45_0    conda-forge
[... 612 lines cut ...]
       virtual packages : __archspec=1=skylake
                          __conda=24.1.2=0
                          __osx=14.2.1=0
                          __unix=0=0
       base environment : /Users/manojbin2/miniforge3  (writable)
      conda av data dir : /Users/manojbin2/miniforge3/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack-experimental/osx-64
                          https://conda.anaconda.org/robostack-experimental/noarch
                          https://conda.anaconda.org/robostack-humble/osx-64
                          https://conda.anaconda.org/robostack-humble/noarch
                          https://conda.anaconda.org/robostack/osx-64
                          https://conda.anaconda.org/robostack/noarch
                          https://conda.anaconda.org/conda-forge/osx-64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : /Users/manojbin2/miniforge3/pkgs
                          /Users/manojbin2/.conda/pkgs
       envs directories : /Users/manojbin2/miniforge3/envs
                          /Users/manojbin2/.conda/envs
               platform : osx-64
             user-agent : conda/24.1.2 requests/2.31.0 CPython/3.10.13 Darwin/23.2.0 OSX/14.2.1 solver/libmamba conda-libmamba-solver/23.12.0 libmambapy/1.5.7
                UID:GID : 501:20
             netrc file : None
           offline mode : False
```

traversaro, 2024-03-19:
Can you try with “What to do if I have Python-related problems during the build of custom messages?" question in https://robostack.github.io/FAQ.html ?

Tobias-Fischer, 2024-03-19:
Feel free to reopen if it’s still an issue

mavneo, 2024-03-19:
Thanks for the quick update, but still encountering the same error.
zsh: no such file or directory: -DPython_EXECUTABLE=/Users/manojbin2/miniforge3/envs/ros_env/bin/python

traversaro, 2024-03-19:
> Thanks for the quick update, but still encountering the same error.
> zsh: no such file or directory: -DPython_EXECUTABLE=/Users/manojbin2/miniforge3/envs/ros_env/bin/python

Are you encountering the same error of the original post or the new error you are reporting?

mavneo, 2024-03-19:
@traversaro I'm encountering the same error

traversaro, 2024-03-19:
You are installing packages from robostack-humble, can you follow the instructions on the website and use the robostack-staging channel?

mavneo, 2024-03-19:
@traversaro I have used this https://robostack.github.io/GettingStarted.html as ref for installing.

traversaro, 2024-03-19:
> @traversaro I have used this https://robostack.github.io/GettingStarted.html as ref for installing.

Those instructions refer to robostack-staging channel, that is not the one reported in mamba list.

Tobias-Fischer, 2024-03-21:
I’m quite certain that the description in the FAQ fixes this - please follow the instructions carefully.

etherswangel, 2024-03-22:
I'm having the same problem. Followed the instructions in the FAQ, it works for `cmake` but not for `colcon build`. Using `robostack-staging` channel on an aarch64 Mac, just updated all packages. 

```zsh
$ colcon build --cmake-args="-DPython_EXECUTABLE=$CONDA_PREFIX/bin/python -DPython3_EXECUTABLE=$CONDA_PREFIX/bin/python -DPYTHON_EXECUTABLE=$CONDA_PREFIX/bin/python -DPython3_FIND_STRATEGY=LOCATION -DPython_FIND_STRATEGY=LOCATION" 

...
CMake Error at /opt/homebrew/Caskroom/miniconda/base/envs/ros2/share/cmake-3.28/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find Python3 (missing: Python3_NumPy_INCLUDE_DIRS NumPy) (found
  version "3.12.2")
...
```

```zsh
$ cmake -DCMAKE_EXPORT_COMPILE_COMMANDS=ON -DPython_EXECUTABLE=/opt/homebrew/Caskroom/miniconda/base/envs/ros2/bin/python -DPython3_EXECUTABLE=/opt/homebrew/Caskroom/miniconda/base/envs/ros2/bin/python -DPYTHON_EXECUTABLE=/opt/homebrew/Caskroom/miniconda/base/envs/ros2/bin/python -DPython3_FIND_STRATEGY=LOCATION -DPython_FIND_STRATEGY=LOCATION

...
success
```

Please help, thank you!

Tobias-Fischer, 2024-03-22:
Hi @etherswangel - please open a new issue and provide all the requested information :)