RoboStack/ros-humble #161: Package request: ros-humble-control-toolbox v3.2.0

bjin-bdai, 2024-03-21:
### Package name

ros-humble-control-toolbox v3.2.0

### Operating System(s)

- [X] linux-64
- [ ] linux-aarch64
- [ ] osx-64
- [ ] osx-arm64
- [ ] win-64

### Additional context (optional)

[Documentation](https://robostack.github.io/humble.html) states that v3.2.0 is available, but a `mamba search ros-humble-control-toolbox` yields the following:

```
Loading channels: done                                                      
# Name                       Version           Build  Channel               
ros-humble-control-toolbox           2.0.2  py39h6fdeb60_0  robostack-humble
ros-humble-control-toolbox           2.1.0  py39h63710d1_1  robostack-humble
ros-humble-control-toolbox           2.1.0  py39h6fdeb60_0  robostack-humble
```

Tobias-Fischer, 2024-03-24:
Looks like you’re using the wrong channel - please use robostack-staging

Tobias-Fischer, 2024-03-24:
https://anaconda.org/robostack-staging/ros-humble-control-toolbox

closed 2024-03-24 by Tobias-Fischer