RoboStack/ros-humble #162: `colcon build` could not find Python3

etherswangel, 2024-03-22:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

I'm having the same problem as mentioned in https://github.com/RoboStack/ros-humble/issues/160. Followed the instructions in the FAQ, it works for `cmake` but not for `colcon build`. Using `robostack-staging` channel on an aarch64 Mac (M3 Max), just updated all packages. 

```zsh
$ colcon build --cmake-args="-DPython_EXECUTABLE=$CONDA_PREFIX/bin/python -DPython3_EXECUTABLE=$CONDA_PREFIX/bin/python -DPYTHON_EXECUTABLE=$CONDA_PREFIX/bin/python -DPython3_FIND_STRATEGY=LOCATION -DPython_FIND_STRATEGY=LOCATION" 

...
CMake Error at /opt/homebrew/Caskroom/miniconda/base/envs/ros2/share/cmake-3.28/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find Python3 (missing: Python3_NumPy_INCLUDE_DIRS NumPy) (found
  version "3.12.2")
...
```

```zsh
$ cmake -DCMAKE_EXPORT_COMPILE_COMMANDS=ON -DPython_EXECUTABLE=/opt/homebrew/Caskroom/miniconda/base/envs/ros2/bin/python -DPython3_EXECUTABLE=/opt/homebrew/Caskroom/miniconda/base/envs/ros2/bin/python -DPYTHON_EXECUTABLE=/opt/homebrew/Caskroom/miniconda/base/envs/ros2/bin/python -DPython3_FIND_STRATEGY=LOCATION -DPython_FIND_STRATEGY=LOCATION

...
success
```

To note that I'm not using `mamba` since it's a drop-in replacement for `conda`, I'd like to stick with it (if only that's not the issue).

### Installed packages

```shell
# packages in environment at /opt/homebrew/Caskroom/miniconda/base/envs/ros2:
#
# Name                    Version                   Build  Channel
aiohttp                   3.9.3           py311h05b510d_1    conda-forge
aiosignal                 1.3.1              pyhd8ed1ab_0    conda-forge
aom                       3.8.2                h078ce10_0    conda-forge
argcomplete               3.2.3              pyhd8ed1ab_0    conda-forge
assimp                    5.3.1                he63ff86_2    conda-forge
atk-1.0                   2.38.0               hcb7b3dd_1    conda-forge
attrs                     23.2.0             pyh71513ae_0    conda-forge
blosc                     1.21.5               hc338f07_0    conda-forge
boost                     1.82.0               hf3b2ce4_6    conda-forge
brotli                    1.1.0                hb547adb_1    conda-forge
brotli-bin                1.1.0                hb547adb_1    conda-forge
bullet                    3.25                 hf3b2ce4_2    conda-forge
bullet-cpp                3.25            py311h6e08293_2    conda-forge
bzip2                     1.0.8                h93a5062_5    conda-forge
c-ares                    1.27.0               h93a5062_0    conda-forge
ca-certificates           2024.2.2             hf0a4a13_0    conda-forge
cairo                     1.18.0               hd1e100b_0    conda-forge
catkin_pkg                1.0.0              pyhd8ed1ab_0    conda-forge
certifi                   2024.2.2           pyhd8ed1ab_0    conda-forge
cffi                      1.16.0          py311h4a08483_0    conda-forge
cmake                     3.28.4               h50fd54c_0    conda-forge
colcon-argcomplete        0.3.3              pyhd8ed1ab_0    conda-forge
colcon-bash               0.5.0              pyhd8ed1ab_0    conda-forge
colcon-cd                 0.1.1              pyhd8ed1ab_0    conda-forge
colcon-cmake              0.2.28             pyhd8ed1ab_0    conda-forge
colcon-common-extensions  0.3.0           py311h267d04e_1    conda-forge
colcon-core               0.16.0             pyhd8ed1ab_0    conda-forge
colcon-defaults           0.2.8              pyhd8ed1ab_0    conda-forge
colcon-devtools           0.3.0              pyhd8ed1ab_0    conda-forge
colcon-library-path       0.2.1                      py_0    conda-forge
colcon-metadata           0.2.5                      py_0    conda-forge
colcon-output             0.2.13             pyhd8ed1ab_0    conda-forge
colcon-package-information 0.4.0              pyhd8ed1ab_0    conda-forge
colcon-package-selection  0.2.10                     py_0    conda-forge
colcon-parallel-executor  0.2.4                      py_0    conda-forge
colcon-pkg-config         0.1.0                      py_0    conda-forge
colcon-powershell         0.4.0              pyhd8ed1ab_0    conda-forge
[... 564 lines cut ...]
          conda version : 24.1.2
    conda-build version : not installed
         python version : 3.11.5.final.0
                 solver : libmamba (default)
       virtual packages : __archspec=1=m1
                          __conda=24.1.2=0
                          __osx=14.4=0
                          __unix=0=0
       base environment : /opt/homebrew/Caskroom/miniconda/base  (writable)
      conda av data dir : /opt/homebrew/Caskroom/miniconda/base/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack-staging/osx-arm64
                          https://conda.anaconda.org/robostack-staging/noarch
                          https://conda.anaconda.org/conda-forge/osx-arm64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : /opt/homebrew/Caskroom/miniconda/base/pkgs
                          /Users/walker/.conda/pkgs
       envs directories : /opt/homebrew/Caskroom/miniconda/base/envs
                          /Users/walker/.conda/envs
               platform : osx-arm64
             user-agent : conda/24.1.2 requests/2.31.0 CPython/3.11.5 Darwin/23.4.0 OSX/14.4 solver/libmamba conda-libmamba-solver/24.1.0 libmambapy/1.5.6
                UID:GID : 501:20
             netrc file : None
           offline mode : False
```

traversaro, 2024-03-22:
In plain cmake you expanded manually the `$CONDA_PREFIX`, while in colcon you used `$CONDA_PREFIX`, do you think this can be related? Can you point us to the specific package whose compilation is failing?

etherswangel, 2024-03-22:
> In plain cmake you expanded manually the $CONDA_PREFIX, while in colcon you used $CONDA_PREFIX, do you think this can be related? 

Sorry, I copied the wrong command. I was debugging that I replaced `$CONDA_PREFIX` manually😄and that doesn't help

> Can you point us to the specific package whose compilation is failing?

I'm testing some random packages, currently seems all packages with custom message (interface) have this issue, since rosidl is using python

etherswangel, 2024-03-23:
> Can you report at least one that is failing? Thanks.

`jackal_msgs`

The following is a minimum example to reproduce on my machine.

CMakeLists.txt
```
cmake_minimum_required(VERSION 3.5)
project(cpp_pubsub)

find_package(rosidl_default_generators REQUIRED)
```

package.xml
```xml
<package format="2">
  <name>cpp_pubsub</name>

  <buildtool_depend>rosidl_default_generators</buildtool_depend>
  <exec_depend>rosidl_default_runtime</exec_depend>
  <member_of_group>rosidl_interface_packages</member_of_group>

</package>
```

Tobias-Fischer, 2024-03-23:
Could you please try and increase the cmake_minimum_version to 3.20 or something like that?

etherswangel, 2024-03-23:
OMG that solves the problem, didn't thought a higher cmake version is needed. Thank you so much 😄

closed 2024-03-23 by etherswangel

Tobias-Fischer, 2024-03-23:
Could you do us a favour and test what the lowest working minimum version is? I’d like to add it to the FAQ

traversaro, 2024-03-23:
For reference, if you need to support older CMake versions (for example for apt compatibility) you can also just specify a version that should be used only for setting policy values, via `cmake_minimum_required(VERSION 3.5...3.20)`, this should behave as `cmake_minimum_required(VERSION 3.20)`.

etherswangel, 2024-03-23:
Sure, in my case I need at least cmake 3.15 to have it work

traversaro, 2024-03-23:
Thanks! Then I guess it is something related to [`CMP0094`](https://cmake.org/cmake/help/latest/policy/CMP0094.html).

Tobias-Fischer, 2024-03-24:
Have added it to the docs: https://github.com/RoboStack/robostack.github.io/commit/54755f314a6fe1c01e2efb8e6d95bf22c173ef92

masterccioli, 2024-05-01:
> Sure, in my case I need at least cmake 3.15 to have it work

I had the same issue -- 3.15 is the minimum working version for me.