RoboStack/ros-humble #164: symbol not found in flat namespace '_PyExc_RuntimeError' On macOS

etherswangel, 2024-03-26:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

I'm running the `cpp_pubsub` tutorial but encountered 

```
~> ros2 run cpp_pubsub talker                                                                                                         dyld[24732]: symbol not found in flat namespace '_PyExc_RuntimeError'
```

for some packages it might be 

```
dyld[20988]: symbol not found in flat namespace '_PyBool_Type'
```

My build command is

```
~> colcon build --symlink-install --cmake-args ' -DPython_EXECUTABLE=$CONDA_PREFIX/bin/python -DPython3_EXECUTABLE=$CONDA_PREFIX/bin/python -DPYTHON_EXECUTABLE=$CONDA_PREFIX/bin/python -DPython3_FIND_STRATEGY=LOCATION -DPython_FIND_STRATEGY=LOCATION' 
```

but some warnings were raised

```
CMake Warning:
  Manually-specified variables were not used by the project:

    Python_EXECUTABLE
```

I added the following to CMakeLists.txt but with no help

```cmake
set(CMAKE_BUILD_WITH_INSTALL_RPATH TRUE)
set(CMAKE_INSTALL_RPATH_USE_LINK_PATH TRUE)
set(CMAKE_INSTALL_RPATH "${CMAKE_INSTALL_PREFIX}/lib/${PROJECT_NAME}")
```

### Installed packages

```shell
# packages in environment at /opt/homebrew/Caskroom/miniconda/base/envs/ros2:
#
# Name                    Version                   Build  Channel
aiohttp                   3.9.3           py311h05b510d_1    conda-forge
aiosignal                 1.3.1              pyhd8ed1ab_0    conda-forge
aom                       3.8.2                h078ce10_0    conda-forge
argcomplete               3.2.3              pyhd8ed1ab_0    conda-forge
assimp                    5.3.1                he63ff86_2    conda-forge
atk-1.0                   2.38.0               hcb7b3dd_1    conda-forge
attrs                     23.2.0             pyh71513ae_0    conda-forge
blosc                     1.21.5               hc338f07_0    conda-forge
boost                     1.82.0               hf3b2ce4_6    conda-forge
brotli                    1.1.0                hb547adb_1    conda-forge
brotli-bin                1.1.0                hb547adb_1    conda-forge
bullet                    3.25                 hf3b2ce4_2    conda-forge
bullet-cpp                3.25            py311h6e08293_2    conda-forge
bzip2                     1.0.8                h93a5062_5    conda-forge
c-ares                    1.27.0               h93a5062_0    conda-forge
ca-certificates           2024.2.2             hf0a4a13_0    conda-forge
cairo                     1.18.0               hd1e100b_0    conda-forge
catkin_pkg                1.0.0              pyhd8ed1ab_0    conda-forge
certifi                   2024.2.2           pyhd8ed1ab_0    conda-forge
cffi                      1.16.0          py311h4a08483_0    conda-forge
cmake                     3.28.3               h50fd54c_0    conda-forge
colcon-argcomplete        0.3.3              pyhd8ed1ab_0    conda-forge
colcon-bash               0.5.0              pyhd8ed1ab_0    conda-forge
[... 581 lines cut ...]
          conda version : 24.1.2
    conda-build version : not installed
         python version : 3.11.5.final.0
                 solver : libmamba (default)
       virtual packages : __archspec=1=m1
                          __conda=24.1.2=0
                          __osx=14.4=0
                          __unix=0=0
       base environment : /opt/homebrew/Caskroom/miniconda/base  (writable)
      conda av data dir : /opt/homebrew/Caskroom/miniconda/base/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack-staging/osx-arm64
                          https://conda.anaconda.org/robostack-staging/noarch
                          https://conda.anaconda.org/conda-forge/osx-arm64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : /opt/homebrew/Caskroom/miniconda/base/pkgs
                          /Users/walker/.conda/pkgs
       envs directories : /opt/homebrew/Caskroom/miniconda/base/envs
                          /Users/walker/.conda/envs
               platform : osx-arm64
             user-agent : conda/24.1.2 requests/2.31.0 CPython/3.11.5 Darwin/23.4.0 OSX/14.4 solver/libmamba conda-libmamba-solver/24.1.0 libmambapy/1.5.6
                UID:GID : 501:20
             netrc file : None
           offline mode : False
```

Tobias-Fischer, 2024-03-28:
I think you need:

```
if(APPLE)
set_target_properties(CHANGE_TARGET_HERE PROPERTIES LINK_FLAGS "-undefined dynamic_lookup")
endif()
```

etherswangel, 2024-03-28:
This doesn't help. Here's my CMakeLists.txt

```
cmake_minimum_required(VERSION 3.20)
project(cpp_pubsub)

if(CMAKE_COMPILER_IS_GNUCXX OR CMAKE_CXX_COMPILER_ID MATCHES "Clang")
  add_compile_options(-Wall -Wextra -Wpedantic)
endif()

set(CMAKE_BUILD_WITH_INSTALL_RPATH TRUE)
set(CMAKE_INSTALL_RPATH_USE_LINK_PATH TRUE)
set(CMAKE_INSTALL_RPATH "${CMAKE_INSTALL_PREFIX}/lib/${PROJECT_NAME}")
set(LINK_FLAGS "${LINK_FLAGS} -undefined dynamic_lookup")

# find dependencies
find_package(ament_cmake REQUIRED)
find_package(rclcpp REQUIRED)
find_package(std_msgs REQUIRED)

add_executable(talker src/publisher_member_function.cpp)
ament_target_dependencies(talker rclcpp std_msgs)

add_executable(listener src/subscriber_member_function.cpp)
ament_target_dependencies(listener rclcpp std_msgs)

set_target_properties(talker listener PROPERTIES LINK_FLAGS "-undefined dynamic_lookup")

install(TARGETS
  talker
  listener
  DESTINATION lib/${PROJECT_NAME})

if(BUILD_TESTING)
  find_package(ament_lint_auto REQUIRED)
  set(ament_cmake_copyright_FOUND TRUE)
  set(ament_cmake_cpplint_FOUND TRUE)
  ament_lint_auto_find_test_dependencies()
endif()

ament_package()
```

Tobias-Fischer, 2024-04-22:
Can you please try changing the LINK_FLAGS to `-flto -Wl,-export_dynamic`?

Tobias-Fischer, 2024-04-22:
I've tried to replicate this locally, and failed. I noticed that you do not have the `compilers` package and other recommended build tools (see Installation tools for local development under https://robostack.github.io/GettingStarted.html) installed. Please install those and try again.

Tobias-Fischer, 2024-04-22:
Feel free to reopen if the problem persists.

closed 2024-04-22 by Tobias-Fischer