RoboStack/ros-humble #165: Package request: ros-humble-rosidl-generator-dds-idl

pengzhenghao, 2024-03-31:
### Package name

ros-humble-rosidl-generator-dds-idl

### Operating System(s)

- [X] linux-64
- [ ] linux-aarch64
- [ ] osx-64
- [ ] osx-arm64
- [ ] win-64

### Additional context (optional)

Thanks for the great work! I can't install ros-humble-rosidl-generator-dds-idl in my ubuntu 18.04..

pengzhenghao, 2024-03-31:
```

Looking for: ['ros-humble-rosidl-generator-dds-idl']

robostack-staging/linux-64                                  Using cache
robostack-staging/noarch                                    Using cache
conda-forge/linux-64                                        Using cache
conda-forge/noarch                                          Using cache
pkgs/main/linux-64                                            No change
pkgs/r/linux-64                                               No change
pkgs/r/noarch                                                 No change
pkgs/main/noarch                                              No change

Pinned packages:
  - python 3.11.*

Could not solve for environment specs
The following package could not be installed
└─ ros-humble-rosidl-generator-dds-idl does not exist (perhaps a typo or a missing channel).

```

Tobias-Fischer, 2024-04-02:
All done

closed 2024-04-02 by Tobias-Fischer