RoboStack/ros-humble #166: ModuleNotFoundError: No module named 'rclpy._rclpy_pybind11'

ashwingur, 2024-04-01:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

I'm having issues with using the `rclpy` library in my python script. When I run `python3` and `import rclpy` I get the following error message
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/__init__.py", line 49, in <module>
    from rclpy.signals import install_signal_handlers
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/signals.py", line 15, in <module>
    from rclpy.exceptions import InvalidHandle
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/exceptions.py", line 15, in <module>
    from rclpy.impl.implementation_singleton import rclpy_implementation as _rclpy
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/impl/implementation_singleton.py", line 32, in <module>
    rclpy_implementation = import_c_library('._rclpy_pybind11', package)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/ros/humble/lib/python3.10/site-packages/rpyutils/import_c_library.py", line 39, in import_c_library
    return importlib.import_module(name, package=package)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ashwin/miniforge3/envs/ros_env/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ModuleNotFoundError: No module named 'rclpy._rclpy_pybind11'
The C extension '/opt/ros/humble/lib/python3.10/site-packages/_rclpy_pybind11.cpython-311-x86_64-linux-gnu.so' isn't present on the system. Please refer to 'https://docs.ros.org/en/humble/Guides/Installation-Troubleshooting.html#import-failing-without-library-present-on-the-system' for possible solutions
```
I have setup my environment as instructed in https://robostack.github.io/GettingStarted.html. I also had a look on https://docs.ros.org/en/humble/Guides/Installation-Troubleshooting.html#import-failing-without-library-present-on-the-system but was unable to find enough information to resolve my issue. I did already have ros humble outside of conda before I setup the ros_env. Is there some possible conflict happening here?

### Installed packages

```shell
# packages in environment at /home/ashwin/miniforge3/envs/ros_env:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
aiohttp                   3.9.3           py311h459d7ec_1    conda-forge
aiosignal                 1.3.1              pyhd8ed1ab_0    conda-forge
alsa-lib                  1.2.11               hd590300_1    conda-forge
aom                       3.8.2                h59595ed_0    conda-forge
argcomplete               3.2.3              pyhd8ed1ab_0    conda-forge
assimp                    5.3.1                hfb0e8fe_2    conda-forge
atk-1.0                   2.38.0               hd4edc92_1    conda-forge
attr                      2.5.1                h166bdaf_1    conda-forge
attrs                     23.2.0             pyh71513ae_0    conda-forge
binutils                  2.40                 hdd6e379_0    conda-forge
binutils_impl_linux-64    2.40                 hf600244_0    conda-forge
binutils_linux-64         2.40                 hdade7a5_3    conda-forge
blosc                     1.21.5               h0f2a231_0    conda-forge
boost                     1.82.0               h781c19f_6    conda-forge
brotli                    1.1.0                hd590300_1    conda-forge
brotli-bin                1.1.0                hd590300_1    conda-forge
bullet                    3.25                 h781c19f_2    conda-forge
bullet-cpp                3.25                 h320fe9a_2    conda-forge
bzip2                     1.0.8                hd590300_5    conda-forge
c-ares                    1.28.1               hd590300_0    conda-forge
c-compiler                1.7.0                hd590300_0    conda-forge
ca-certificates           2024.2.2             hbcca054_0    conda-forge
cairo                     1.18.0               h3faef2a_0    conda-forge
catkin_pkg                1.0.0              pyhd8ed1ab_0    conda-forge
catkin_tools              0.9.4              pyhd8ed1ab_0    conda-forge
certifi                   2024.2.2           pyhd8ed1ab_0    conda-forge
cffi                      1.16.0          py311hb3a22ac_0    conda-forge
cmake                     3.28.3               hcfe8598_0    conda-forge
colcon-argcomplete        0.3.3              pyhd8ed1ab_0    conda-forge
colcon-bash               0.5.0              pyhd8ed1ab_0    conda-forge
colcon-cd                 0.1.1              pyhd8ed1ab_0    conda-forge
colcon-cmake              0.2.28             pyhd8ed1ab_0    conda-forge
[... 642 lines cut ...]
         python version : 3.10.14.final.0
                 solver : libmamba (default)
       virtual packages : __archspec=1=zen3
                          __conda=24.1.2=0
                          __cuda=12.2=0
                          __glibc=2.35=0
                          __linux=6.5.0=0
                          __unix=0=0
       base environment : /home/ashwin/miniforge3  (writable)
      conda av data dir : /home/ashwin/miniforge3/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack-staging/linux-64
                          https://conda.anaconda.org/robostack-staging/noarch
                          https://conda.anaconda.org/conda-forge/linux-64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : /home/ashwin/miniforge3/pkgs
                          /home/ashwin/.conda/pkgs
       envs directories : /home/ashwin/miniforge3/envs
                          /home/ashwin/.conda/envs
               platform : linux-64
             user-agent : conda/24.1.2 requests/2.31.0 CPython/3.10.14 Linux/6.5.0-26-generic ubuntu/22.04.4 glibc/2.35 solver/libmamba conda-libmamba-solver/24.1.0 libmambapy/1.5.7
                UID:GID : 1000:1000
             netrc file : None
           offline mode : False
```

Tobias-Fischer, 2024-04-01:
It looks like your system installation of ROS conflicts with the RoboStack one - avoid activating them both in the same terminal.

ashwingur, 2024-04-01:
Forgot I had `source /opt/ros/humble/setup.bash` in my bashrc which was causing conflict. Removing that resolved the issue thanks

closed 2024-04-01 by ashwingur