RoboStack/ros-humble #170: Package request: groot

SebastianBitsch, 2024-04-09:
### Package name

groot

### Operating System(s)

- [ ] linux-64
- [ ] linux-aarch64
- [ ] osx-64
- [X] osx-arm64
- [ ] win-64

### Additional context (optional)

Hi, 
[groot](https://github.com/BehaviorTree/Groot) is a nice tool when working with behavior trees. 
It's main dependency ```ros-humble-behaviortree-cpp-v3``` is already built and available on RoboStack, but when installing groot by git cloning the repo into a ```~/ws/src/```-dir and running ```colcon build```, it fails to find the dependency. A groot package would alleviate this. 

### Links and references (optional)

https://github.com/BehaviorTree/Groot

wolfv, 2024-04-09:
@SebastianBitsch do you know if `groot` is packaged as a ROS package? e.g. is there a `ros-humble-groot` package in Ubuntu etc.?

SebastianBitsch, 2024-04-09:
@wolfv I don't think so - or at least I haven't been able to find one

Tobias-Fischer, 2024-04-09:
It might be easier to try and fix your local build. Have you installed all the tools required for a local build (https://robostack.github.io/GettingStarted.html)? Can you please provide the output of `conda list`, confirm that you have activated your robostack-env when trying to build, and post the build log?

SebastianBitsch, 2024-04-10:
Hi @Tobias-Fischer, thanks for helping out. I have followed the getting started and have been using ROS w gazebo, rviz2, etc. in RoboStack for some time without issue now - I should have all the tools, and i am sure I have the environment activated when trying to build. 

Here is the output of conda list:
```bash
(ros2_env) sebastianbitsch@MBPtilhebastian ros2_ws % conda list
# packages in environment at /Users/sebastianbitsch/micromamba/envs/ros2_env:
#
# Name                    Version                   Build  Channel
aiohttp                   3.9.3           py311he705e18_1    conda-forge
aiosignal                 1.3.1              pyhd8ed1ab_0    conda-forge
aom                       3.8.2                h73e2aa4_0    conda-forge
argcomplete               3.2.3              pyhd8ed1ab_0    conda-forge
assimp                    5.3.1                h460e769_2    conda-forge
atk-1.0                   2.38.0               h1d18e73_1    conda-forge
attrs                     23.2.0             pyh71513ae_0    conda-forge
aws-c-auth                0.7.16               h79cb451_6    conda-forge
aws-c-cal                 0.6.10               h7beb4c2_1    conda-forge
aws-c-common              0.9.13               h10d778d_0    conda-forge
aws-c-compression         0.2.18               hf5538d0_1    conda-forge
aws-c-event-stream        0.4.2                hb970d5a_4    conda-forge
aws-c-http                0.8.1                h45f72bc_5    conda-forge
aws-c-io                  0.14.5               h59ded20_1    conda-forge
aws-c-mqtt                0.10.2               ha7d5480_4    conda-forge
aws-c-s3                  0.5.2                hb71b441_0    conda-forge
aws-c-sdkutils            0.1.15               hf5538d0_1    conda-forge
aws-checksums             0.1.18               hf5538d0_1    conda-forge
aws-crt-cpp               0.26.2               h56a5195_7    conda-forge
aws-sdk-cpp               1.11.267             h4da54b2_1    conda-forge
azure-core-cpp            1.10.3               hbb1e571_1    conda-forge
azure-storage-blobs-cpp   12.10.0              he51d815_0    conda-forge
azure-storage-common-cpp  12.5.0               hf4badfb_2    conda-forge
benchmark                 1.8.3                he965462_0    conda-forge
blosc                     1.21.5               heccf04b_0    conda-forge
boost                     1.82.0               h9bdd199_6    conda-forge
boost-cpp                 1.82.0               h07eb623_6    conda-forge
brotli                    1.1.0                h0dc2134_1    conda-forge
brotli-bin                1.1.0                h0dc2134_1    conda-forge
bullet                    3.25                 h9bdd199_2    conda-forge
bullet-cpp                3.25                 h7be5568_2    conda-forge
bzip2                     1.0.8                h10d778d_5    conda-forge
c-ares                    1.27.0               h10d778d_0    conda-forge
c-compiler                1.7.0                h282daa2_0    conda-forge
ca-certificates           2024.2.2             h8857fd0_0    conda-forge
cairo                     1.18.0               h99e66fa_0    conda-forge
cartographer              2.0.0           lua54h9c039b3_20    conda-forge
catkin_pkg                1.0.0              pyhd8ed1ab_0    conda-forge
catkin_tools              0.9.4              pyhd8ed1ab_0    conda-forge
cctools                   986                  h40f6528_0    conda-forge
cctools_osx-64            986                  ha1c5b94_0    conda-forge
ceres-solver              2.2.0                h5cbf65c_1    conda-forge
certifi                   2024.2.2           pyhd8ed1ab_0    conda-forge
cffi                      1.16.0          py311hc0b63fd_0    conda-forge
cfitsio                   4.4.0                h60fb419_0    conda-forge
clang                     16.0.6               hdae98eb_5    conda-forge
clang-16                  16.0.6          default_h7151d67_5    conda-forge
clang_impl_osx-64         16.0.6              h8787910_10    conda-forge
clang_osx-64              16.0.6              hb91bd55_10    conda-forge
clangxx                   16.0.6          default_h7151d67_5    conda-forge
clangxx_impl_osx-64       16.0.6              h6d92fbe_10    conda-forge
clangxx_osx-64            16.0.6              hb91bd55_10    conda-forge
cmake                     3.28.3               h7c85d92_0    conda-forge
colcon-argcomplete        0.3.3              pyhd8ed1ab_0    conda-forge
colcon-bash               0.5.0              pyhd8ed1ab_0    conda-forge
colcon-cd                 0.1.1              pyhd8ed1ab_0    conda-forge
colcon-cmake              0.2.28             pyhd8ed1ab_0    conda-forge
colcon-common-extensions  0.3.0           py311h6eed73b_1    conda-forge
colcon-core               0.16.0             pyhd8ed1ab_0    conda-forge
colcon-defaults           0.2.8              pyhd8ed1ab_0    conda-forge
colcon-devtools           0.3.0              pyhd8ed1ab_0    conda-forge
colcon-library-path       0.2.1                      py_0    conda-forge
[... 768 lines cut ...]
make[2]: *** [CMakeFiles/behavior_tree_editor.dir/build.make:185: CMakeFiles/behavior_tree_editor.dir/bt_editor/utils.cpp.o] Error 1
make[2]: *** [CMakeFiles/behavior_tree_editor.dir/build.make:171: CMakeFiles/behavior_tree_editor.dir/bt_editor/editor_flowscene.cpp.o] Error 1
1 error generated.
[... cut ...]
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
1 error generated.
make[2]: *** [CMakeFiles/behavior_tree_editor.dir/build.make:157: CMakeFiles/behavior_tree_editor.dir/bt_editor/mainwindow.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:188: CMakeFiles/behavior_tree_editor.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
---
Failed   <<< groot [4.23s, exited with code 2]

Summary: 0 packages finished [4.82s]
  1 package failed: groot
  1 package had stderr output: groot
```

Let me know if there is anything else you need, thanks

Tobias-Fischer, 2024-04-10:
Can you please clean the build and provide the full output?

SebastianBitsch, 2024-04-10:
Sorry - i am kinda new to this, here is a clean build:
```bash
(ros2_env) sebastianbitsch@MBPtilhebastian ros2_ws % rm -fr  build install log
(ros2_env) sebastianbitsch@MBPtilhebastian ros2_ws % colcon build --packages-select groot > build.log 2>&1
```
which gives the output:
```bash
Starting >>> groot
--- stderr: groot
CMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Deprecation Warning at QtNodeEditor/CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Deprecation Warning at test/CMakeLists.txt:1 (CMAKE_MINIMUM_REQUIRED):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

/Users/sebastianbitsch/ros2_ws/src/Groot/QtNodeEditor/src/ConnectionStyle.cpp:169:3: warning: 'qsrand' is deprecated: use QRandomGenerator instead [-Wdeprecated-declarations]
  qsrand(hash);
  ^
/Users/sebastianbitsch/micromamba/envs/ros2_env/include/qt/QtCore/qglobal.h:1279:15: note: 'qsrand' has been explicitly marked deprecated here
Q_CORE_EXPORT QT_DEPRECATED_VERSION_X_5_15("use QRandomGenerator instead") void qsrand(uint seed);
              ^
/Users/sebastianbitsch/micromamba/envs/ros2_env/include/qt/QtCore/qglobal.h:374:45: note: expanded from macro 'QT_DEPRECATED_VERSION_X_5_15'
# define QT_DEPRECATED_VERSION_X_5_15(text) QT_DEPRECATED_X(text)
                                            ^
/Users/sebastianbitsch/micromamba/envs/ros2_env/include/qt/QtCore/qglobal.h:294:33: note: expanded from macro 'QT_DEPRECATED_X'
#  define QT_DEPRECATED_X(text) Q_DECL_DEPRECATED_X(text)
                                ^
/Users/sebastianbitsch/micromamba/envs/ros2_env/include/qt/QtCore/qcompilerdetection.h:675:55: note: expanded from macro 'Q_DECL_DEPRECATED_X'
#    define Q_DECL_DEPRECATED_X(text) __attribute__ ((__deprecated__(text)))
                                                      ^
/Users/sebastianbitsch/ros2_ws/src/Groot/QtNodeEditor/src/ConnectionStyle.cpp:170:21: warning: 'qrand' is deprecated: use QRandomGenerator instead [-Wdeprecated-declarations]
  std::size_t hue = qrand() % hue_range;
                    ^
/Users/sebastianbitsch/micromamba/envs/ros2_env/include/qt/QtCore/qglobal.h:1280:15: note: 'qrand' has been explicitly marked deprecated here
Q_CORE_EXPORT QT_DEPRECATED_VERSION_X_5_15("use QRandomGenerator instead") int qrand();
              ^
/Users/sebastianbitsch/micromamba/envs/ros2_env/include/qt/QtCore/qglobal.h:374:45: note: expanded from macro 'QT_DEPRECATED_VERSION_X_5_15'
# define QT_DEPRECATED_VERSION_X_5_15(text) QT_DEPRECATED_X(text)
                                            ^
/Users/sebastianbitsch/micromamba/envs/ros2_env/include/qt/QtCore/qglobal.h:294:33: note: expanded from macro 'QT_DEPRECATED_X'
#  define QT_DEPRECATED_X(text) Q_DECL_DEPRECATED_X(text)
                                ^
/Users/sebastianbitsch/micromamba/envs/ros2_env/include/qt/QtCore/qcompilerdetection.h:675:55: note: expanded from macro 'Q_DECL_DEPRECATED_X'
#    define Q_DECL_DEPRECATED_X(text) __attribute__ ((__deprecated__(text)))
                                                      ^
2 warnings generated.
/Users/sebastianbitsch/ros2_ws/src/Groot/QtNodeEditor/src/NodeGeometry.cpp:286:44: warning: 'width' is deprecated: Use QFontMetrics::horizontalAdvance [-Wdeprecated-declarations]
    width = std::max(unsigned(_fontMetrics.width(name)), width);
                                           ^
/Users/sebastianbitsch/micromamba/envs/ros2_env/include/qt/QtGui/qfontmetrics.h:105:5: note: 'width' has been explicitly marked deprecated here
    QT_DEPRECATED_X("Use QFontMetrics::horizontalAdvance")
    ^
/Users/sebastianbitsch/micromamba/envs/ros2_env/include/qt/QtCore/qglobal.h:294:33: note: expanded from macro 'QT_DEPRECATED_X'
#  define QT_DEPRECATED_X(text) Q_DECL_DEPRECATED_X(text)
                                ^
[... 70 lines cut ...]
make[2]: *** [CMakeFiles/behavior_tree_editor.dir/build.make:213: CMakeFiles/behavior_tree_editor.dir/bt_editor/graphic_container.cpp.o] Error 1
make[2]: *** [CMakeFiles/behavior_tree_editor.dir/build.make:171: CMakeFiles/behavior_tree_editor.dir/bt_editor/editor_flowscene.cpp.o] Error 1
In file included from /Users/sebastianbitsch/ros2_ws/src/Groot/bt_editor/mainwindow.cpp:1:
In file included from /Users/sebastianbitsch/ros2_ws/src/Groot/bt_editor/mainwindow.h:16:
In file included from /Users/sebastianbitsch/ros2_ws/src/Groot/bt_editor/graphic_container.h:8:
/Users/sebastianbitsch/ros2_ws/src/Groot/bt_editor/bt_editor_base.h:11:10: fatal error: 'behaviortree_cpp_v3/bt_factory.h' file not found
#include <behaviortree_cpp_v3/bt_factory.h>
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
1 error generated.
1 error generated.
make[2]: *** [CMakeFiles/behavior_tree_editor.dir/build.make:185: CMakeFiles/behavior_tree_editor.dir/bt_editor/utils.cpp.o] Error 1
[... cut ...]
1 error generated.
make[2]: *** [CMakeFiles/behavior_tree_editor.dir/build.make:157: CMakeFiles/behavior_tree_editor.dir/bt_editor/mainwindow.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:188: CMakeFiles/behavior_tree_editor.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
---
Failed   <<< groot [21.7s, exited with code 2]

Summary: 0 packages finished [21.8s]
  1 package failed: groot
  1 package had stderr output: groot
```

I hope this is what you mean, if not let me know - thank you for helping!

Tobias-Fischer, 2024-04-10:
Here are the changes that I required to get it to build:
```
diff --git a/CMakeLists.txt b/CMakeLists.txt
index eeda3ea..f52034c 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -12,10 +12,10 @@ find_package(Qt5 COMPONENTS  Core Widgets Gui OpenGL Xml Svg)
 set(CMAKE_MODULE_PATH ${CMAKE_MODULE_PATH}  "${CMAKE_CURRENT_LIST_DIR}/cmake")
 
 if(NOT CMAKE_VERSION VERSION_LESS 3.1)
-    set(CMAKE_CXX_STANDARD 11)
+    set(CMAKE_CXX_STANDARD 14)
     set(CMAKE_CXX_STANDARD_REQUIRED ON)
 else()
-    set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11")
+    set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++14")
 endif()
 
 #############################################################
@@ -60,7 +60,7 @@ set(CMAKE_AUTOUIC ON)
 set(CMAKE_AUTORCC ON)
 
 add_definitions(${Qt5Widgets_DEFINITIONS} -DNODE_EDITOR_STATIC )
-set(CMAKE_CXX_FLAGS "${Qt5Widgets_EXECUTABLE_COMPILE_FLAGS}")
+set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} ${Qt5Widgets_EXECUTABLE_COMPILE_FLAGS}")
 
 #set (CMAKE_CXX_FLAGS_DEBUG "${CMAKE_CXX_FLAGS_DEBUG} -fno-omit-frame-pointer -fsanitize=address")
 #set (CMAKE_LINKER_FLAGS_DEBUG "${CMAKE_STATIC_LINKER_FLAGS_DEBUG} -fno-omit-frame-pointer -fsanitize=address")
diff --git a/test/CMakeLists.txt b/test/CMakeLists.txt
index 126d559..dd51831 100644
--- a/test/CMakeLists.txt
+++ b/test/CMakeLists.txt
@@ -8,7 +8,7 @@ find_package(Qt5Gui REQUIRED)
 SET(CMAKE_AUTOMOC ON)
 SET(CMAKE_AUTORCC ON)
 SET(CMAKE_INCLUDE_CURRENT_DIR ON)
-SET(CMAKE_CXX_STANDARD 11)
+SET(CMAKE_CXX_STANDARD 14)
 SET(CMAKE_CXX_STANDARD_REQUIRED ON)
 ENABLE_TESTING()
 
```

I'd be happy to include `groot` as an additional recipe if you want to prepare one - it shouldn't be hard; see the additional_recipes directory for examples.

SebastianBitsch, 2024-04-11:
Built without problems and runs like a charm - thank you so much. I will look into preparing a recipe

closed 2024-07-23 by Tobias-Fischer