RoboStack/ros-humble #174: Strange behaviour on initialising the env compared to native install of ROS Humble

JackDeSomeTrades, 2024-04-24:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

I see strange behaviour in populating the `LD_LIBRARY_PATH` environment variable with a native ROS2 Humble install and one installed with RoboStack. 

Below is an explanatory image with expected behaviour on the left and current behaviour on the right: 

[image]

This issue is affecting the discovery of  `libgeometry_msgs__rosidl_generator_py.so` which is required when generating custom msgs for a package and can be found in `../miniforge3/envs/<env-name>/lib/`. 

In a native install, sourcing the `setup.ba/z/sh` file populates this path in the `LD_LIBRARY_PATH` while sourcing the same setup file in `../miniforge3/envs/<env-name>/setup.ba/z/sh` does not. 

Exporting the path manually in every terminal seems to solve the discovery issue but this is not expected nor preferred behaviour. 

I can't seem to find what's causing this bug but I hope you may be able to help. 

### Installed packages

```shell
# packages in environment at /home/pavan/miniforge3/envs/carla_humble:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
aiohttp                   3.9.3           py310h2372a71_1    conda-forge
aiosignal                 1.3.1              pyhd8ed1ab_0    conda-forge
alsa-lib                  1.2.8                h166bdaf_0    conda-forge
aom                       3.5.0                h27087fc_0    conda-forge
argcomplete               3.2.2              pyhd8ed1ab_0    conda-forge
assimp                    5.2.5                hf40c2ba_0    conda-forge
astroid                   3.1.0           py310hff52083_0    conda-forge
async-timeout             4.0.3              pyhd8ed1ab_0    conda-forge
atk-1.0                   2.38.0               hd4edc92_1    conda-forge
attr                      2.5.1                h166bdaf_1    conda-forge
attrs                     23.2.0             pyh71513ae_0    conda-forge
blosc                     1.21.5               h0f2a231_0    conda-forge
boost                     1.78.0          py310hcb52e73_5    conda-forge
boost-cpp                 1.78.0               h5adbc97_2    conda-forge
brotli                    1.1.0                hd590300_1    conda-forge
brotli-bin                1.1.0                hd590300_1    conda-forge
bullet                    3.24                 hfdc917e_0    conda-forge
bullet-cpp                3.24                 h769672d_0    conda-forge
bzip2                     1.0.8                hd590300_5    conda-forge
c-ares                    1.26.0               hd590300_0    conda-forge
ca-certificates           2024.2.2             hbcca054_0    conda-forge
cairo                     1.16.0            ha61ee94_1014    conda-forge
catkin_pkg                1.0.0              pyhd8ed1ab_0    conda-forge
certifi                   2024.2.2           pyhd8ed1ab_0    conda-forge
cffi                      1.16.0          py310h2fee648_0    conda-forge
cmake                     3.26.3               h077f3f9_0    conda-forge
colorama                  0.4.6              pyhd8ed1ab_0    conda-forge
console_bridge            1.0.2                h924138e_1    conda-forge
contourpy                 1.2.1           py310hd41b1e2_0    conda-forge
cppcheck                  2.13.2          py310hb32fc57_0    conda-forge
cryptography              42.0.2          py310hb8475ec_0    conda-forge
curl                      7.88.1               hdc1c0ab_1    conda-forge
cycler                    0.12.1             pyhd8ed1ab_0    conda-forge
dav1d                     1.2.1                hd590300_0    conda-forge
dbus                      1.13.6               h5008d03_3    conda-forge
dill                      0.3.8              pyhd8ed1ab_0    conda-forge
distro                    1.9.0              pyhd8ed1ab_0    conda-forge
docutils                  0.21.1             pyhd8ed1ab_0    conda-forge
double-conversion         3.2.0                h27087fc_1    conda-forge
eigen                     3.4.0                h00ab1b0_0    conda-forge
elfutils                  0.189                hde5d1a3_0    conda-forge
empy                      3.3.4              pyh9f0ad1d_1    conda-forge
[... 628 lines cut ...]
                          __cuda=12.2=0
                          __glibc=2.35=0
                          __linux=6.5.0=0
                          __unix=0=0
       base environment : /home/pavan/miniforge3  (writable)
      conda av data dir : /home/pavan/miniforge3/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack-staging/linux-64
                          https://conda.anaconda.org/robostack-staging/noarch
                          https://conda.anaconda.org/conda-forge/linux-64
                          https://conda.anaconda.org/conda-forge/noarch
                          https://repo.anaconda.com/pkgs/main/linux-64
                          https://repo.anaconda.com/pkgs/main/noarch
                          https://repo.anaconda.com/pkgs/r/linux-64
                          https://repo.anaconda.com/pkgs/r/noarch
          package cache : /home/pavan/miniforge3/pkgs
                          /home/pavan/.conda/pkgs
       envs directories : /home/pavan/miniforge3/envs
                          /home/pavan/.conda/envs
               platform : linux-64
             user-agent : conda/23.11.0 requests/2.31.0 CPython/3.10.13 Linux/6.5.0-28-generic ubuntu/22.04.4 glibc/2.35 solver/libmamba conda-libmamba-solver/23.12.0 libmambapy/1.5.5
                UID:GID : 1000:1000
             netrc file : None
           offline mode : False
```

Tobias-Fischer, 2024-04-24:
Many thanks for reporting this! Can you please share code for a minimum working/failing example?

JackDeSomeTrades, 2024-04-24:
Hi, thanks for responding. I'm not sure if I can provide a minimum failing example.. Perhaps you can try to build and run `derived_object_msgs` from here - https://github.com/astuff/astuff_sensor_msgs/tree/master/derived_object_msgs 

It requires `geometry_msgs` to build and run. It builds well,  but using it in any package throws up an error because the package cannot find `libgeometry_msgs__rosidl_generator_py.so` 

The context here is that I'm trying to use the Carla simulator with the ROS_Bridge. The reason I manually built `derived_object_msgs` is because `ros-humble-derived-object-msgs` isn't available on Robostack for Linux-x86-64 (it was going to be my next pull request). 

It could be a good test to see if using the prebuilt package would work better? However, that still doesn't address the core `lib` issue.

Tobias-Fischer, 2024-04-24:
You do not have the compilers package etc installed - please check our install guide and try again, this could be the culprit.

JackDeSomeTrades, 2024-04-25:
Tried with installing the compilers, colcon-common-extensions packages etc and still the same behaviour.

Tobias-Fischer, 2024-04-25:
If you could please report the steps that need to be taken to reproduce the error, that'd be great. E.g. something like
```
mkdir -p ~/ws/src
cd ~/ws/src
git clone ...
colcon build ...
ros2 run ...
```

Tobias-Fischer, 2024-07-02:
Closing due to inactivity.

closed 2024-07-02 by Tobias-Fischer