RoboStack/ros-humble #175: Important:  libprotobuf==4.25.1 needs to be installed for gazebo_ros_plugins to work with ros2-distro-mutex==0.5.0=humble

traversaro, 2024-04-24:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

Similar to https://github.com/RoboStack/ros-noetic/pull/464, also in ros-humble we have the same problem of packages built against libprotobuf but not properly constraining its version. The proper solution is to merge and do a rebuild, but in the meanwhile I opened this issue that hopefully is visible enough for people experiencing this problem.

@matteomastrogiuseppe @STaliani @AleTarsi do you happen to still have around the error you were getting when libprotobuf was missing? If you have it, if you copy&paste the error in the issue it would be easier for future people having this problem to find the solution via search engines or LLMs.

### Installed packages

```shell
.
```

### Environment info

```shell
.
```

matteomastrogiuseppe, 2024-04-24:
I don't have the full error log anymore, but here is the core:
```
libprotobuf.so.25.1.0: cannot open shared object file:
```

For now, it can be fixed by specifying the libprotobuf version in the environment.yaml.

traversaro, 2024-04-24:
Great, thanks!

traversaro, 2024-06-01:
Probably we should put this somewhere in the docs.

Daviesss, 2024-12-14:
After installing  libprotobuf==4.25.1 the laser scan and camera topics are not still coming up.

traversaro, 2024-12-14:
Hello @Daviesss, can you provide the full report of what you are trying to do and what is happening, including your environment, the command that you are running and the error that you are experiencing? Probably it may be easier to just open a new issue (https://github.com/RoboStack/ros-humble/issues/new/choose) and follow the template, thanks!

Daviesss, 2024-12-17:
> Hello @Daviesss, can you provide the full report of what you are trying to do and what is happening, including your environment, the command that you are running and the error that you are [...]

The issue I am having , I was able to resolve it . But now I don't seem to see the laser scan topic when robot is spawned into gazebo . But I get the remaining  topics.

traversaro, 2024-12-17:
Also for this issue we are in a similar situation, if you can provide a full report of the error and the environment you installed in a new issue https://github.com/RoboStack/ros-humble/issues/new/choose ?

Tobias-Fischer, 2025-01-06:
Fixed in #229 which has a constraint on protobuf.

closed 2025-01-06 by Tobias-Fischer