RoboStack/ros-humble #177: ROS Humble not installed correctly on macOS

alvgaona, 2024-04-25:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

I've tried installing ROS Humble in macOS by following the getting started guide — I had installed this in an Ubuntu machine a couple days prior — but I'm having a couple issues with macOS right now.

1. The first thing I noticed when I ran `rviz2` was that the `AMENT_PREFIX_PATH` env var wasn't set properly. I had to manually point it to `/path/to/envs/ros_env`.
2. After doing the ament prefix thing, I tried running `rviz2` again but I encountered this error.
```
libc++abi: terminating due to uncaught exception of type ament_index_cpp::PackageNotFoundError: package 'rviz_common' not found, searching: []
fish: Job 1, 'rviz2' terminated by signal SIGABRT (Abort)
```
3. Then I was wondering if packages had been installed correctly by running `ros2 pkg list` but nothing pops up. In contrast, this works perfectly on Ubuntu 23.10.

There might be some misconfiguration somewhere that makes `ros2` not find the installed pacakges; `mamba list` shows a lot of packages for `ros-humble-*`.

### Installed packages

```shell
# packages in environment at /Users/agaona/.miniconda/envs/ros_env:
#
# Name                    Version                   Build  Channel
aiohttp                   3.9.5           py311he705e18_0    conda-forge
aiosignal                 1.3.1              pyhd8ed1ab_0    conda-forge
aom                       3.9.0                h9bc2f43_0    conda-forge
argcomplete               3.3.0              pyhd8ed1ab_0    conda-forge
assimp                    5.3.1                h460e769_2    conda-forge
atk-1.0                   2.38.0               h4bec284_2    conda-forge
attrs                     23.2.0             pyh71513ae_0    conda-forge
blosc                     1.21.5               hafa3907_1    conda-forge
boost                     1.82.0               h9bdd199_6    conda-forge
brotli                    1.1.0                h0dc2134_1    conda-forge
brotli-bin                1.1.0                h0dc2134_1    conda-forge
bullet                    3.25                 h9bdd199_2    conda-forge
bullet-cpp                3.25                 h7be5568_2    conda-forge
bzip2                     1.0.8                h6c40b1e_5  
c-ares                    1.28.1               h10d778d_0    conda-forge
c-compiler                1.7.0                h282daa2_0    conda-forge
ca-certificates           2024.3.11            hecd8cb5_0  
cairo                     1.18.0               h99e66fa_0    conda-forge
catkin_pkg                1.0.0              pyhd8ed1ab_0    conda-forge
catkin_tools              0.9.4              pyhd8ed1ab_0    conda-forge
cctools                   986                  h40f6528_0    conda-forge
cctools_osx-64            986                  ha1c5b94_0    conda-forge
certifi                   2024.2.2           pyhd8ed1ab_0    conda-forge
cffi                      1.16.0          py311hc0b63fd_0    conda-forge
clang                     16.0.6               hdae98eb_6    conda-forge
clang-16                  16.0.6          default_h7151d67_6    conda-forge
clang_impl_osx-64         16.0.6              h8787910_11    conda-forge
clang_osx-64              16.0.6              hb91bd55_11    conda-forge
clangxx                   16.0.6          default_h7151d67_6    conda-forge
clangxx_impl_osx-64       16.0.6              h6d92fbe_11    conda-forge
clangxx_osx-64            16.0.6              hb91bd55_11    conda-forge
cmake                     3.28.3               h7c85d92_0    conda-forge
colcon-argcomplete        0.3.3              pyhd8ed1ab_0    conda-forge
colcon-bash               0.5.0              pyhd8ed1ab_0    conda-forge
colcon-cd                 0.1.1              pyhd8ed1ab_0    conda-forge
colcon-cmake              0.2.28             pyhd8ed1ab_0    conda-forge
colcon-common-extensions  0.3.0           py311h6eed73b_1    conda-forge
colcon-core               0.16.0             pyhd8ed1ab_0    conda-forge
colcon-defaults           0.2.8              pyhd8ed1ab_0    conda-forge
colcon-devtools           0.3.0              pyhd8ed1ab_0    conda-forge
colcon-library-path       0.2.1                      py_0    conda-forge
colcon-metadata           0.2.5                      py_0    conda-forge
colcon-output             0.2.13             pyhd8ed1ab_0    conda-forge
colcon-package-information 0.4.0              pyhd8ed1ab_0    conda-forge
colcon-package-selection  0.2.10                     py_0    conda-forge
colcon-parallel-executor  0.2.4                      py_0    conda-forge
[... 600 lines cut ...]
          conda version : 24.3.0
    conda-build version : not installed
         python version : 3.12.2.final.0
                 solver : libmamba (default)
       virtual packages : __archspec=1=cannonlake
                          __conda=24.3.0=0
                          __osx=13.6.6=0
                          __unix=0=0
       base environment : /Users/alvgaona/.miniconda  (writable)
      conda av data dir : /Users/alvgaona/.miniconda/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack-staging/osx-64
                          https://conda.anaconda.org/robostack-staging/noarch
                          https://conda.anaconda.org/conda-forge/osx-64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : /Users/alvgaona/.miniconda/pkgs
                          /Users/alvgaona/.conda/pkgs
       envs directories : /Users/alvgaona/.miniconda/envs
                          /Users/alvgaona/.conda/envs
               platform : osx-64
             user-agent : conda/24.3.0 requests/2.31.0 CPython/3.12.2 Darwin/22.6.0 OSX/13.6.6 solver/libmamba conda-libmamba-solver/24.1.0 libmambapy/1.5.8
                UID:GID : 501:20
             netrc file : None
           offline mode : False
```

alvgaona, 2024-04-25:
Just after posting this I've figured I have the fish shell and there are only `setup.bash`, `setup.sh` and `setup.zsh` files which probably are not getting triggered by the `conda activate ros_env`. Therefore, ROS cannont find the packages location.

wolfv, 2024-04-25:
That does sound like an issue. Is `fish` supported by ROS?

alvgaona, 2024-04-25:
I don't think it is. I just confirmed that running ROS under a different shell that is not bash or zsh, it won't load the script, and we'll have an incomplete activated environment 😢

alvgaona, 2024-04-25:
There's a draft PR from 2019 [here](https://github.com/ros2/ros2cli/pull/326).

wolfv, 2024-04-25:
In that case there is not much we can do here  ... the `setup.XX` scripts are generated by ROS. Sorry!

alvgaona, 2024-04-25:
Yes, definitely! I just realized that after posting this.

closed 2024-04-25 by alvgaona

wolfv, 2024-04-25:
One thing you might be able to do is use `pixi run fish` or `micromamba -n myenv run fish`

That would first start a `bash` shell, activate the environment, and then load fish.