RoboStack/ros-humble #180: Package request: libcamera

sonelu, 2024-05-10:
### Package name

libcamera

### Operating System(s)

- [ ] linux-64
- [X] linux-aarch64
- [ ] osx-64
- [ ] osx-arm64
- [ ] win-64

### Additional context (optional)

I see it is in included in the vinca file for aarch64, but does't seem to have been build.

### Links and references (optional)

It's needed for camera_ros, which seems to be the only ros package that work with the new libcamera on Raspberry Pi.

Tobias-Fischer, 2024-07-21:
Hi @sonelu - that's an interesting one. It uses `meson` as build tool, and we don't have a template for that. I just added a warning message to `vinca` for easier debugging in the future (https://github.com/RoboStack/vinca/commit/8109f1288919eaf39037032ec7f47e96da939120).

/cc @traversaro @wolfv - have you built packages using `meson` before? Do you think it would be easy enough to add a template for RoboStack in vinca?

traversaro, 2024-07-22:
> /cc @traversaro @wolfv - have you built packages using `meson` before? Do you think it would be easy enough to add a template for RoboStack in vinca?

Yes, I maintain https://github.com/conda-forge/spral-feedstock/ (and probably a few other, I can't remember). The meson workflow is quite similar to the CMake one, so I would guess it should doable.

Tobias-Fischer, 2024-07-23:
@sonelu  would you be interested in contributing this? It shouldn’t be too hard, but time on the core RoboStack contributors is very constrained at the moment and other things like a rebuild have higher priority.

sonelu, 2024-07-26:
@Tobias-Fischer certainly I could.

The background for this is that [camera_ros](https://github.com/christianrauch/camera_ros) uses libcamera, and is the only ROS package that I could find that works with Raspberry Pi since the change to libcamera. The issue is that apt installed libcamera in the os is not detected by the conda installed ros (ex. from Robostack) and compilation of camera_ros fails. I had managed to get over that by symlinking the /usr/lib/aarch-linux-gnu/pkgconfig/libcamera* defintions to /home/<user>/miniforge3/envs/<env>/lib/pkgconfig. You have more experience in dealing with system libraries when building under a conda environment. Maybe we don't need to package it at all, just fix the compilation of camera_ros (which is the one we need to include in RoboStack).

Alreantively, for libcamera I think it would be better if it would packaged in conda-forge rather than RoboStack. What do you think?

Tobias-Fischer, 2024-07-26:
Yes that’s a good point - it’s probably better in conda-forge and that should be less complicated to do as well; simply follow the same recipe structure as in e.g. https://github.com/conda-forge/spral-feedstock/

traversaro, 2025-06-22:
I guess this was solved by @carlodri in https://github.com/conda-forge/staged-recipes/pull/29477 (thanks a lot!) and also the rasperrypi specific fork is packaged (see https://github.com/conda-forge/libcamera-feedstock/pull/3), so I think we can close the issue here. @sonelu please comment (or open a new one) if the issue is not solved.

fyi @elephantum

closed 2025-06-22 by traversaro