RoboStack/ros-humble #181: Gazebo not publishing topics

etherswangel, 2024-05-16:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

Gazebo not publishing any topics, including /clock and other sensor topics.

### Installed packages

```shell
# packages in environment at /opt/homebrew/Caskroom/miniconda/base/envs/ros2:
#
# Name                    Version                   Build  Channel
aiohttp                   3.9.3           py311h05b510d_1    conda-forge
aiosignal                 1.3.1              pyhd8ed1ab_0    conda-forge
aom                       3.8.2                h078ce10_0    conda-forge
argcomplete               3.2.3              pyhd8ed1ab_0    conda-forge
assimp                    5.3.1                he63ff86_2    conda-forge
atk-1.0                   2.38.0               hcb7b3dd_1    conda-forge
attrs                     23.2.0             pyh71513ae_0    conda-forge
aws-c-auth                0.7.17               h382b9c6_2    conda-forge
aws-c-cal                 0.6.11               hd34e5fa_0    conda-forge
aws-c-common              0.9.15               h93a5062_0    conda-forge
aws-c-compression         0.2.18               hd34e5fa_3    conda-forge
aws-c-event-stream        0.4.2                h247c08a_8    conda-forge
aws-c-http                0.8.1               hf9e830b_10    conda-forge
aws-c-io                  0.14.7               h33d81b3_6    conda-forge
aws-c-mqtt                0.10.3               h5f4abda_4    conda-forge
aws-c-s3                  0.5.7                h606a3d2_1    conda-forge
aws-c-sdkutils            0.1.15               hd34e5fa_3    conda-forge
aws-checksums             0.1.18               hd34e5fa_3    conda-forge
aws-crt-cpp               0.26.6               h13f0230_4    conda-forge
aws-sdk-cpp               1.11.267             h134aaec_6    conda-forge
azure-core-cpp            1.11.1               he231e37_1    conda-forge
azure-storage-blobs-cpp   12.10.0              h2ffa867_1    conda-forge
azure-storage-common-cpp  12.5.0               h09a5875_4    conda-forge
benchmark                 1.8.3                h13dd4ca_0    conda-forge
blosc                     1.21.5               hc338f07_0    conda-forge
boost                     1.82.0               hf3b2ce4_6    conda-forge
boost-cpp                 1.82.0               hca5e981_6    conda-forge
brotli                    1.1.0                hb547adb_1    conda-forge
brotli-bin                1.1.0                hb547adb_1    conda-forge
bullet                    3.25                 hf3b2ce4_2    conda-forge
bullet-cpp                3.25            py311h6e08293_2    conda-forge
bzip2                     1.0.8                h93a5062_5    conda-forge
c-ares                    1.27.0               h93a5062_0    conda-forge
c-compiler                1.7.0                h6aa9301_0    conda-forge
ca-certificates           2024.2.2             hf0a4a13_0    conda-forge
cairo                     1.18.0               hd1e100b_0    conda-forge
catkin_pkg                1.0.0              pyhd8ed1ab_0    conda-forge
catkin_tools              0.9.4              pyhd8ed1ab_0    conda-forge
ccache                    4.9.1                h9b0f9a8_0    conda-forge
cctools                   986                  h4faf515_0    conda-forge
cctools_osx-arm64         986                  h62378fb_0    conda-forge
ceres-solver              2.2.0                h036b7f2_2    conda-forge
certifi                   2024.2.2           pyhd8ed1ab_0    conda-forge
cffi                      1.16.0          py311h4a08483_0    conda-forge
cfitsio                   4.4.0                h808cd33_1    conda-forge
clang                     16.0.6               h30cc82d_6    conda-forge
clang-16                  16.0.6          default_he012953_6    conda-forge
clang_impl_osx-arm64      16.0.6              hc421ffc_11    conda-forge
clang_osx-arm64           16.0.6              h54d7cd3_11    conda-forge
clangxx                   16.0.6          default_h4cf2255_6    conda-forge
clangxx_impl_osx-arm64    16.0.6              hcd7bac0_11    conda-forge
clangxx_osx-arm64         16.0.6              h54d7cd3_11    conda-forge
cmake                     3.29.2               h50fd54c_0    conda-forge
colcon-argcomplete        0.3.3              pyhd8ed1ab_0    conda-forge
colcon-bash               0.5.0              pyhd8ed1ab_0    conda-forge
colcon-cd                 0.1.1              pyhd8ed1ab_0    conda-forge
[... 738 lines cut ...]
          conda version : 24.5.0
    conda-build version : not installed
         python version : 3.11.8.final.0
                 solver : libmamba (default)
       virtual packages : __archspec=1=m1
                          __conda=24.5.0=0
                          __osx=14.4.1=0
                          __unix=0=0
       base environment : /opt/homebrew/Caskroom/miniconda/base  (writable)
      conda av data dir : /opt/homebrew/Caskroom/miniconda/base/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack-staging/osx-arm64
                          https://conda.anaconda.org/robostack-staging/noarch
                          https://conda.anaconda.org/conda-forge/osx-arm64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : /opt/homebrew/Caskroom/miniconda/base/pkgs
                          /Users/walker/.conda/pkgs
       envs directories : /opt/homebrew/Caskroom/miniconda/base/envs
                          /Users/walker/.conda/envs
               platform : osx-arm64
             user-agent : conda/24.5.0 requests/2.31.0 CPython/3.11.8 Darwin/23.4.0 OSX/14.4.1 solver/libmamba conda-libmamba-solver/24.1.0 libmambapy/1.5.6
                UID:GID : 501:20
             netrc file : None
           offline mode : False
```

traversaro, 2024-05-16:
Can you check https://github.com/RoboStack/ros-humble/issues/175 ? Thanks!

etherswangel, 2024-05-16:
Thank you for the quick reply! `conda install libprotobuf==4.25.1` solves the problem. It also reinstalled `gazebo` 

```
gazebo                                 11.14.0-h36c1656_9 --> 11.14.0-h816cae4_7
```

I'll check if this is the point if I got some time :D

traversaro, 2024-05-16:
The problem is not in gazebo itself, but that ros-gazebo plugins have been compiled with libprotobuf==4.25.1, so if you try to load them with libprotobuf==4.25.3 they print an error that they can't find the libprotobuf==4.25.1 libraries. The change in gazebo version is just a consequence of this, as each gazebo version is compiled against a version of libprotobuf.

etherswangel, 2024-05-16:
Thank you for explaining!

closed 2024-05-16 by etherswangel