RoboStack/ros-humble #185: laserscan plugin not found by Gazebo

MannavaVivek, 2024-05-31:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

I am trying to run a Gazebo simulation with a laser scanner, but the scan topic never gets published. When I checked the logs, I got the following error:

```
(1717157020 348620272) [Err] [Plugin.hh:212] Failed to load plugin libgazebo_ros_laser.so: dlopen(libgazebo_ros_laser.so, 0x0009): tried: '/Users/vivek/miniforge3/envs/ros_env/opt/rviz_ogre_vendor/lib/libgazebo_ros_laser.so' (no such file), '/libgazebo_ros_laser.so' (no such file), '/Users/vivek/miniforge3/envs/ros_env/lib/gazebo-11/plugins/libgazebo_ros_laser.so' (no such file), '/libgazebo_ros_laser.so' (no such file), 'libgazebo_ros_laser.so' (no such file), 
```

here is the section where I am loading the plugin
```
      	<plugin name="${name}_controller" filename="libgazebo_ros_laser.so">
      	  <topicName>${ros_topic}</topicName>
      	  <frame_name>${name}_link</frame_name>
        </plugin>
```

### Installed packages

```shell
see below
```

### Environment info

```shell
mamba version : 1.5.8
     active environment : ros_env
    active env location : /Users/vivek/miniforge3/envs/ros_env
            shell level : 1
       user config file : /Users/vivek/.condarc
 populated config files : /Users/vivek/miniforge3/.condarc
                          /Users/vivek/.condarc
                          /Users/vivek/miniforge3/envs/ros_env/.condarc
          conda version : 24.5.0
    conda-build version : not installed
         python version : 3.10.14.final.0
                 solver : libmamba (default)
       virtual packages : __archspec=1=m1
                          __conda=24.5.0=0
                          __osx=14.5=0
                          __unix=0=0
       base environment : /Users/vivek/miniforge3  (writable)
      conda av data dir : /Users/vivek/miniforge3/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack-staging/osx-arm64
                          https://conda.anaconda.org/robostack-staging/noarch
                          https://conda.anaconda.org/conda-forge/osx-arm64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : /Users/vivek/miniforge3/pkgs
                          /Users/vivek/.conda/pkgs
       envs directories : /Users/vivek/miniforge3/envs
                          /Users/vivek/.conda/envs
               platform : osx-arm64
             user-agent : conda/24.5.0 requests/2.31.0 CPython/3.10.14 Darwin/23.5.0 OSX/14.5 solver/libmamba conda-libmamba-solver/24.1.0 libmambapy/1.5.8
                UID:GID : 501:20
             netrc file : None
           offline mode : False
```

MannavaVivek, 2024-05-31:
Here are my packages:
```
# packages in environment at /Users/vivek/miniforge3/envs/ros_env:
#
# Name                    Version                   Build  Channel
aiohttp                   3.9.5           py311h05b510d_0    conda-forge
aiosignal                 1.3.1              pyhd8ed1ab_0    conda-forge
aom                       3.8.2                h078ce10_0    conda-forge
argcomplete               3.3.0              pyhd8ed1ab_0    conda-forge
assimp                    5.3.1                he63ff86_2    conda-forge
atk-1.0                   2.38.0               hcb7b3dd_1    conda-forge
attrs                     23.2.0             pyh71513ae_0    conda-forge
aws-c-auth                0.7.17               h382b9c6_2    conda-forge
aws-c-cal                 0.6.11               hd34e5fa_0    conda-forge
aws-c-common              0.9.15               h93a5062_0    conda-forge
aws-c-compression         0.2.18               hd34e5fa_3    conda-forge
aws-c-event-stream        0.4.2                h247c08a_8    conda-forge
aws-c-http                0.8.1               hf9e830b_10    conda-forge
aws-c-io                  0.14.7               h33d81b3_6    conda-forge
aws-c-mqtt                0.10.3               h5f4abda_4    conda-forge
aws-c-s3                  0.5.7                h606a3d2_1    conda-forge
aws-c-sdkutils            0.1.15               hd34e5fa_3    conda-forge
aws-checksums             0.1.18               hd34e5fa_3    conda-forge
aws-crt-cpp               0.26.6               h13f0230_4    conda-forge
aws-sdk-cpp               1.11.267             h134aaec_6    conda-forge
azure-core-cpp            1.10.3               he231e37_1    conda-forge
azure-storage-blobs-cpp   12.10.0              h6aa02a4_0    conda-forge
azure-storage-common-cpp  12.5.0               h607ffeb_2    conda-forge
benchmark                 1.8.3                h13dd4ca_0    conda-forge
blosc                     1.21.5               hc338f07_0    conda-forge
boost                     1.82.0               hf3b2ce4_6    conda-forge
boost-cpp                 1.82.0               hca5e981_6    conda-forge
brotli                    1.1.0                hb547adb_1    conda-forge
brotli-bin                1.1.0                hb547adb_1    conda-forge
bullet                    3.25                 hf3b2ce4_2    conda-forge
bullet-cpp                3.25            py311h6e08293_2    conda-forge
bzip2                     1.0.8                h93a5062_5    conda-forge
c-ares                    1.28.1               h93a5062_0    conda-forge
c-compiler                1.7.0                h6aa9301_1    conda-forge
ca-certificates           2024.2.2             hf0a4a13_0    conda-forge
cairo                     1.18.0               hd1e100b_0    conda-forge
catkin_pkg                1.0.0              pyhd8ed1ab_0    conda-forge
catkin_tools              0.9.4              pyhd8ed1ab_0    conda-forge
cctools                   986                  h4faf515_0    conda-forge
cctools_osx-arm64         986                  h62378fb_0    conda-forge
ceres-solver              2.2.0                h036b7f2_2    conda-forge
certifi                   2024.2.2           pyhd8ed1ab_0    conda-forge
cffi                      1.16.0          py311h4a08483_0    conda-forge
cfitsio                   4.3.1                h808cd33_0    conda-forge
clang                     16.0.6               h30cc82d_6    conda-forge
clang-16                  16.0.6          default_he012953_6    conda-forge
clang_impl_osx-arm64      16.0.6              hc421ffc_14    conda-forge
clang_osx-arm64           16.0.6              h54d7cd3_14    conda-forge
clangxx                   16.0.6          default_h4cf2255_6    conda-forge
clangxx_impl_osx-arm64    16.0.6              hcd7bac0_14    conda-forge
clangxx_osx-arm64         16.0.6              h54d7cd3_14    conda-forge
cmake                     3.28.3               h50fd54c_0    conda-forge
colcon-argcomplete        0.3.3              pyhd8ed1ab_0    conda-forge
colcon-bash               0.5.0              pyhd8ed1ab_0    conda-forge
colcon-cd                 0.1.1              pyhd8ed1ab_0    conda-forge
colcon-cmake              0.2.28             pyhd8ed1ab_0    conda-forge
colcon-common-extensions  0.3.0           py311h267d04e_1    conda-forge
colcon-core               0.16.0             pyhd8ed1ab_0    conda-forge
colcon-defaults           0.2.8              pyhd8ed1ab_0    conda-forge
colcon-devtools           0.3.0              pyhd8ed1ab_0    conda-forge
colcon-library-path       0.2.1                      py_0    conda-forge
colcon-metadata           0.2.5                      py_0    conda-forge
colcon-output             0.2.13             pyhd8ed1ab_0    conda-forge
colcon-package-information 0.4.0              pyhd8ed1ab_0    conda-forge
colcon-package-selection  0.2.10                     py_0    conda-forge
[... 756 lines cut ...]
xorg-libxdmcp             1.1.3                h27ca646_0    conda-forge
xorg-libxext              1.3.4                h1a8c8d9_2    conda-forge
xorg-libxfixes            5.0.3             h3422bc3_1004    conda-forge
xorg-libxmu               1.1.3                h99b78c6_1    conda-forge
xorg-libxpm               3.5.17               hb547adb_0    conda-forge
xorg-libxrandr            1.5.2                h3422bc3_1    conda-forge
xorg-libxrender           0.9.11               hb547adb_0    conda-forge
[... cut ...]
yaml-cpp                  0.8.0                h13dd4ca_0    conda-forge
yarl                      1.9.4           py311h05b510d_0    conda-forge
zeromq                    4.3.5                hcc0f68c_4    conda-forge
zipp                      3.17.0             pyhd8ed1ab_0    conda-forge
zlib                      1.2.13               h53f4e23_5    conda-forge
zstd                      1.5.6                hb46c0d2_0    conda-forge
zziplib                   0.13.69              he1e0b03_1    conda-forge
```

traversaro, 2024-06-01:
Can you try to pin the protobuf version, see for example https://github.com/RoboStack/ros-humble/issues/175 .

MannavaVivek, 2024-06-02:
I have `libprotobuf==4.25.1` already, and reinstalling confirmed it.

traversaro, 2024-06-02:
> I have `libprotobuf==4.25.1` already, and reinstalling confirmed it.

Sorry, I missed that. Then I guess the problem is similar to https://github.com/RoboStack/ros-noetic/issues/228 and https://github.com/RoboStack/ros-noetic/issues/462, can you run:

~~~
otool -L $CONDA_PREFIX/lib/libgazebo_ros_laser.dylib
~~~

? My guess is that this will show that the loader can't find `libRayPlugin.dylib`, so a quick workaround would be to symlink it in `$CONDA_PREFIX/lib/`:

~~~
ln -s $CONDA_PREFIX/lib/gazebo-11/plugins/libRayPlugin.dylib $CONDA_PREFIX/lib/libRayPlugin.dylib
~~~

As suggested in https://github.com/RoboStack/ros-noetic/issues/228#issuecomment-1035689132 . 

In theory, the problem was fixed upstream by https://github.com/ros-simulation/gazebo_ros_pkgs/pull/974, but I guess something changed in macOS, just to double check can you also report the output of:

~~~
echo $DYLD_LIBRARY_PATH 
~~~

Thanks!

MannavaVivek, 2024-06-02:
I tried with the symlink, but it still can't find the libraries. 
here's the output you requested:
```
echo $DYLD_LIBRARY_PATH
/Users/vivek/miniforge3/envs/ros_env/opt/rviz_ogre_vendor/lib::/Users/vivek/miniforge3/envs/ros_env/lib/gazebo-11/plugins:
```

traversaro, 2024-06-02:
Can you report the output of `otool -L $CONDA_PREFIX/lib/libgazebo_ros_laser.dylib` ? That is the one that should explain the problem.

MannavaVivek, 2024-06-02:
sorry, here you go:
```
otool -L $CONDA_PREFIX/lib/libgazebo_ros_laser.dylib
error: otool: can't open file: /Users/vivek/miniforge3/envs/ros_env/lib/libgazebo_ros_laser.dylib (No such file or directory)
```

traversaro, 2024-06-02:
That is strange, can you try `otool -L $CONDA_PREFIX/lib/libgazebo_ros_laser.so` ?

MannavaVivek, 2024-06-02:
its the same 
```
otool -L $CONDA_PREFIX/lib/libgazebo_ros_laser.so

error: otool: can't open file: /Users/vivek/miniforge3/envs/ros_env/lib/libgazebo_ros_laser.so (No such file or directory)
```

and in case you were wondering, I have RayPlugin in my gazebo-11/plugins

```
ls | grep Ray

libGpuRayPlugin.dylib
libRayPlugin.dylib
libRaySensorNoisePlugin.dylib
```

MannavaVivek, 2024-06-02:
```
ls $CONDA_PREFIX/lib | grep gazebo_ros
gazebo_ros
libgazebo_ros2_control.dylib
libgazebo_ros_ackermann_drive.dylib
libgazebo_ros_bumper.dylib
libgazebo_ros_camera.dylib
libgazebo_ros_diff_drive.dylib
libgazebo_ros_elevator.dylib
libgazebo_ros_factory.dylib
libgazebo_ros_force.dylib
libgazebo_ros_force_system.dylib
libgazebo_ros_ft_sensor.dylib
libgazebo_ros_gps_sensor.dylib
libgazebo_ros_hand_of_god.dylib
libgazebo_ros_harness.dylib
libgazebo_ros_imu_sensor.dylib
libgazebo_ros_init.dylib
libgazebo_ros_joint_pose_trajectory.dylib
libgazebo_ros_joint_state_publisher.dylib
libgazebo_ros_node.dylib
libgazebo_ros_p3d.dylib
libgazebo_ros_planar_move.dylib
libgazebo_ros_projector.dylib
libgazebo_ros_properties.dylib
libgazebo_ros_ray_sensor.dylib
libgazebo_ros_state.dylib
libgazebo_ros_template.dylib
libgazebo_ros_tricycle_drive.dylib
libgazebo_ros_utils.dylib
libgazebo_ros_vacuum_gripper.dylib
libgazebo_ros_video.dylib
libgazebo_ros_wheel_slip.dylib
```

traversaro, 2024-06-02:
Are you sure that `libgazebo_ros_laser.dylib` exists at all in ROS 2 ? It seems a ROS 1 thing at the first glance.

MannavaVivek, 2024-06-02:
I tried this, but with sensor type set to "ray" and this library, gazebo just crashes. But since its not the same issue anymore, I will close this one.

closed 2024-06-02 by MannavaVivek