RoboStack/ros-humble #186: Cannot install ros-humble-ros-gz on MacOS

icosac, 2024-06-20:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

Hello everyone!

I must confess that I'm not the most familiar person with Mamba nor Conda, so the solution may very well be something really trivial and that has nothing to do with RoboStack. Please forgive me if this is the case.

I'm running MacOS on an M1 Pro and I wanted to check if I could run a whole project I've been developing on Ubuntu with ROS2 Humble natively.

I downloaded the repo and run `rosdep` to install the missing packages. All of them installed correctly, except for `ros-humble-ros-gz` that throws the following error:

```
Encountered problems while solving:
  - package ros-humble-rviz-assimp-vendor-11.2.10-py311he51b96d_5 requires assimp >=5.3.1,<5.3.2.0a0, but none of the providers can be installed
```

I run `mamba install -c conda-forge assimp` and it installed the package correctly, yet I cannot install `ros-humble-ros-gz`. 

Does anyone have the same problem or knows a possible solution? Thanks!

### Installed packages

```shell
# Name                    Version                   Build  Channel
aom                       3.9.0                h537b5a7_0    conda-forge
argcomplete               3.3.0              pyhd8ed1ab_0    conda-forge
assimp                    5.3.1                he63ff86_2    conda-forge
atk-1.0                   2.38.0               hd03087b_2    conda-forge
aws-c-auth                0.7.20               h5cf208e_0    conda-forge
aws-c-cal                 0.6.12               h35c0bb2_0    conda-forge
aws-c-common              0.9.17               h03532ee_0    conda-forge
aws-c-compression         0.2.18               h35c0bb2_4    conda-forge
aws-c-event-stream        0.4.2               h7d5773a_10    conda-forge
aws-c-http                0.8.1               h00faecf_13    conda-forge
aws-c-io                  0.14.8               h6dd71cf_0    conda-forge
aws-c-mqtt                0.10.4               h92d7a41_2    conda-forge
aws-c-s3                  0.5.8                he1e208d_3    conda-forge
aws-c-sdkutils            0.1.16               h35c0bb2_0    conda-forge
aws-checksums             0.1.18               h35c0bb2_4    conda-forge
aws-crt-cpp               0.26.8              h34bd0e7_10    conda-forge
aws-sdk-cpp               1.11.267             h108e708_8    conda-forge
azure-core-cpp            1.11.1               he231e37_1    conda-forge
azure-identity-cpp        1.6.0                hd1853d3_1    conda-forge
azure-storage-blobs-cpp   12.10.0              h2ffa867_1    conda-forge
azure-storage-common-cpp  12.5.0               h09a5875_4    conda-forge
benchmark                 1.8.4                h00cdb27_1    conda-forge
blosc                     1.21.5               h9c252e8_1    conda-forge
boost                     1.82.0               hf3b2ce4_6    conda-forge
boost-cpp                 1.82.0               hca5e981_6    conda-forge
bullet-cpp                3.25            py311h6e08293_2    conda-forge
bzip2                     1.0.8                h80987f9_6  
c-ares                    1.28.1               h93a5062_0    conda-forge
c-compiler                1.7.0                h6aa9301_1    conda-forge
ca-certificates           2024.6.2             hf0a4a13_0    conda-forge
cairo                     1.18.0               hd1e100b_0    conda-forge
catkin_pkg                1.0.0              pyhd8ed1ab_0    conda-forge
catkin_tools              0.9.4              pyhd8ed1ab_0    conda-forge
cctools                   986                  h4faf515_0    conda-forge
cctools_osx-arm64         986                  h62378fb_0    conda-forge
ceres-solver              2.2.0                h036b7f2_2    conda-forge
cfitsio                   4.4.0                h808cd33_1    conda-forge
clang                     16.0.6               h30cc82d_6    conda-forge
clang-16                  16.0.6          default_he012953_6    conda-forge
clang_impl_osx-arm64      16.0.6              hc421ffc_14    conda-forge
clang_osx-arm64           16.0.6              h54d7cd3_14    conda-forge
clangxx                   16.0.6          default_h4cf2255_6    conda-forge
clangxx_impl_osx-arm64    16.0.6              hcd7bac0_14    conda-forge
[... 589 lines cut ...]
       user config file : /Users/user/.condarc
 populated config files : /Users/user/.condarc
          conda version : 4.12.0
    conda-build version : not installed
         python version : 3.9.12.final.0
       virtual packages : __osx=14.2.1=0
                          __unix=0=0
                          __archspec=1=arm64
       base environment : /Users/user/opt/miniconda3  (writable)
      conda av data dir : /Users/user/opt/miniconda3/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack-staging/osx-arm64
                          https://conda.anaconda.org/robostack-staging/noarch
                          https://conda.anaconda.org/conda-forge/osx-arm64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : /Users/user/opt/miniconda3/pkgs
                          /Users/user/.conda/pkgs
       envs directories : /Users/user/opt/miniconda3/envs
                          /Users/user/.conda/envs
               platform : osx-arm64
             user-agent : conda/4.12.0 requests/2.29.0 CPython/3.9.12 Darwin/23.2.0 OSX/14.2.1
                UID:GID : 501:20
             netrc file : None
           offline mode : False
```

traversaro, 2024-06-21:
Can you provide the exact commands necessary to reproduce the issue for you?

icosac, 2024-06-21:
Yes, after installing RoboStack, I activate the environment (called ros_env)

```
mamba activate ros_env
```

Then to install all the dependencies I run:

```
rosdep install --from-paths src --rosdistro humble -y 
```

But the package that is giving me problems is `ros-humble-ros-gz`. Indeed, even if I try to install it directly I cannot. The following commands returns the error I mentioned above:

```
mamba install -c conda-forge -c robostack-staging ros-humble-ros-gz 
```

All the others packages installed correctly.

traversaro, 2024-06-21:
~~~
CONDA_SUBDIR=osx-arm64 CONDA_OVERRIDE_OSX=11.0 mamba create -n test186 -c conda-forge -c robostack-staging ros-humble-ros-gz
~~~

work fine (env variables are to reproduce the osx stuff on linux), unfortunately unless you provide how you created the `ros_env` and the packages installed by `rosdep` install, it may be difficult for us to reproduce the problem.

icosac, 2024-06-21:
I tried to run your command and it worked on a new environment, so there must have been some package conflict. 

After that I decided to clean everything and start anew. So I removed all packages, environments and the whole microforge folder. Then I followed the [guide](https://robostack.github.io/GettingStarted.html#__tabbed_1_1) to recreate the environment. 

Once it was ready, I tried to run `rosdep` again, and while I managed to install `ros-humble-ros-gz`, now I have a problem on other packages always relevant to Gazebo. The full lists of packages I'm trying to install with `rosdep` is:

```
ros-humble-nav2-bringup ros-humble-xacro ros-humble-ros-gz ros-humble-joint-state-publisher ros-humble-gazebo-msgs ros-humble-gazebo-ros-pkgs ros-humble-gazebo-ros2-control ros-humble-gazebo-ros 
```

I can install the first 4 without problem, while I cannot install the following packages: 

```
ros-humble-gazebo-ros-pkgs ros-humble-gazebo-ros2-control ros-humble-gazebo-ros
```

But, I should be able to based on [this table](https://robostack.github.io/humble.html), right?

The error that I get when trying to install any of these packages is the following:

```
$ mamba install -c robostack-staging -c conda-forge -y ros-humble-gazebo-ros2-control                         1 ↵

Looking for: ['ros-humble-gazebo-ros2-control']

robostack-staging/osx-arm64                                 Using cache
robostack-staging/noarch                                    Using cache
conda-forge/osx-arm64                                       Using cache
conda-forge/noarch                                          Using cache

Pinned packages:
  - python 3.11.*

warning  libmamba Added empty dependency for problem type SOLVER_RULE_UPDATE
Could not solve for environment specs
The following packages are incompatible
├─ libignition-physics5 is installable with the potential options
│  ├─ libignition-physics5 5.2.0 would require
│  │  └─ dartsim >=6.12.2,<6.13.0a0  with the potential options
│  │     ├─ dartsim [6.12.2|6.13.0] would require
│  │     │  └─ assimp >=5.2.5,<5.2.6.0a0 , which can be installed;
│  │     ├─ dartsim [6.12.2|6.13.2] would require
│  │     │  └─ assimp >=5.4.1,<5.4.2.0a0 , which can be installed;
│  │     ├─ dartsim 6.12.2, which can be installed;
│  │     └─ dartsim [6.12.1|6.12.2] would require
│  │        └─ assimp >=5.2.4,<5.2.5.0a0 , which can be installed;
│  ├─ libignition-physics5 [5.1.0|5.2.0] would require
│  │  └─ dartsim >=6.12.1,<6.13.0a0  with the potential options
│  │     ├─ dartsim [6.12.2|6.13.0], which can be installed (as previously explained);
│  │     ├─ dartsim [6.12.2|6.13.2], which can be installed (as previously explained);
│  │     ├─ dartsim 6.12.2, which can be installed;
│  │     ├─ dartsim [6.12.1|6.12.2], which can be installed (as previously explained);
│  │     ├─ dartsim 6.12.1 would require
│  │     │  └─ assimp >=5.1.3,<5.2.0a0 , which can be installed;
│  │     ├─ dartsim 6.12.1 would require
│  │     │  └─ assimp >=5.2.3,<5.2.4.0a0 , which can be installed;
│  │     └─ dartsim 6.12.1 would require
│  │        └─ assimp >=5.2.0,<5.3.0a0 , which can be installed;
│  ├─ libignition-physics5 5.2.0 would require
│  │  └─ assimp >=5.2.4,<5.2.5.0a0 , which can be installed;
│  ├─ libignition-physics5 5.2.0 would require
│  │  └─ assimp >=5.2.5,<5.2.6.0a0 , which can be installed;
│  └─ libignition-physics5 5.2.0 would require
│     └─ assimp >=5.4.1,<5.4.2.0a0 , which can be installed;
├─ ros-humble-gazebo-ros2-control is installable and it requires
│  └─ ros-humble-gazebo-dev with the potential options
│     ├─ ros-humble-gazebo-dev 3.7.0 would require
│     │  └─ python 3.10.* *_cpython, which can be installed;
│     └─ ros-humble-gazebo-dev 3.7.0 would require
│        └─ gazebo >=11.14.0,<12.0a0  with the potential options
[... 15 lines cut ...]
│                 ├─ dartsim 6.13.1 conflicts with any installable versions previously reported;
│                 ├─ dartsim [6.12.2|6.13.2], which can be installed (as previously explained);
│                 └─ dartsim 6.13.2 conflicts with any installable versions previously reported;
└─ ros-humble-rviz-assimp-vendor is installable with the potential options
   ├─ ros-humble-rviz-assimp-vendor 11.2.10 would require
   │  └─ assimp >=5.3.1,<5.3.2.0a0 , which conflicts with any installable versions previously reported;
   └─ ros-humble-rviz-assimp-vendor [11.2.4|11.2.5] would require
      └─ python [3.10.* |3.10.* *_cpython], which can be installed.
```

And I think it may be still related to assimp for some reasons. 

At the same time, if I run 

```
mamba create -n test -c conda-forge -c robostack-staging ros-humble-gazebo-ros # for example
```

then it installs correctly, whereas if I run:

```
mamba create -n test -c conda-forge -c robostack-staging ros-humble-gazebo-ros ros-humble-ros-gz
```

then I get a similar error as above that I pasted in a [pastbin](https://pastebin.com/G0CS86uy) post as it's much longer.

traversaro, 2024-06-21:
I am afraid there are simply no compatible version of libignition-gazebo6 and gazebo that has been compiled with the same version of dartsim and is compatible with the version of the libraries used by the latest robostack build, see the kind of error you can get from:

~~~
mamba create -n test -c conda-forge -c robostack-staging gazebo libignition-gazebo6 "ros2-distro-mutex==0.5.*" assimp==5.3.1 "tinyxml2==10.*" "libprotobuf==4.24.4"
~~~

For the future, we should add `dartsim` to the conda-forge pinning (both the one in conda-forge and in https://github.com/RoboStack/ros-humble/blob/main/.ci_support/conda_forge_pinnings.yaml) and re-build all robostack packages. Do you think it make sense to rename the issue to track the fact that it is not possible to install `ros-humble-gazebo-ros` and `ros-humble-ros-gz` at the same time, as instead `ros-humble-ros-gz` can be installed on its own without any problem?

traversaro, 2024-06-21:
Actually I am not even sure that would solve the issue, as the problem is that `libignition-gazebo6` requires dartsim 6.12.* and is not compatible with dartsim 6.13.* , that is instead used by gazebo-classic .

icosac, 2024-06-21:
I was actually just about to update the issue. I was reading through this old [one](https://github.com/RoboStack/ros-humble/issues/68#event-11853393049) #68 I posted some time ago and it came to my mind that the problem might have been trying to install both Ignition and Classic.

I'll update the title of the issue right away. Also, I agree that it may not be possible to install them both at the same time, but it may not be a problem as long as it is made clear. It's still possible to have two separate environments, one for Classic and one for Ignition. 

Btw, I had left an old dependency inside my project that should have been removed, I don't really need both Ignition and Classic at the same time and I reckon neither does the great majority of people. Still, if there is a way of fixing this, it would obviously be much better.

traversaro, 2025-01-09:
I think the problem was solved by the last rebuild, i.e. the following command works fine:

~~~
CONDA_SUBDIR=osx-arm64 conda create -n testgazebogz -c conda-forge -c robostack-staging ros-humble-gazebo-ros-pkgs ros-humble-ros-gz-bridge "ros2-distro-mutex==0.6.*"
~~~

closed 2025-01-09 by Tobias-Fischer