RoboStack/ros-humble #189: Fastdds server not working as expected in MacOS

xingruiyang, 2024-07-09:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

I am trying to use fastdds to establish a centralized communication server between two nodes, but  unable to get a working example following [the official doc](https://docs.ros.org/en/humble/Tutorials/Advanced/Discovery-Server/Discovery-Server.html). It seems that no matter what ROS_DISCOVERY_SERVER I set the talker/listener nodes can always find each other. I am using a MacOS running Sonoma 14.5

### Installed packages

```shell
aiohttp                   3.9.5           py311h05b510d_0    conda-forge
aiosignal                 1.3.1              pyhd8ed1ab_0    conda-forge
aom                       3.8.2                h078ce10_0    conda-forge
argcomplete               3.4.0              pyhd8ed1ab_0    conda-forge
assimp                    5.3.1                he63ff86_2    conda-forge
atk-1.0                   2.38.0               hcb7b3dd_1    conda-forge
attrs                     23.2.0             pyh71513ae_0    conda-forge
aws-c-auth                0.7.17               h382b9c6_2    conda-forge
aws-c-cal                 0.6.11               hd34e5fa_0    conda-forge
aws-c-common              0.9.15               h93a5062_0    conda-forge
aws-c-compression         0.2.18               hd34e5fa_3    conda-forge
aws-c-event-stream        0.4.2                h247c08a_8    conda-forge
aws-c-http                0.8.1               hf9e830b_10    conda-forge
aws-c-io                  0.14.7               h33d81b3_6    conda-forge
aws-c-mqtt                0.10.3               h5f4abda_4    conda-forge
aws-c-s3                  0.5.7                h606a3d2_1    conda-forge
aws-c-sdkutils            0.1.15               hd34e5fa_3    conda-forge
aws-checksums             0.1.18               hd34e5fa_3    conda-forge
aws-crt-cpp               0.26.6               h13f0230_4    conda-forge
aws-sdk-cpp               1.11.267             h134aaec_6    conda-forge
azure-core-cpp            1.11.1               he231e37_1    conda-forge
azure-storage-blobs-cpp   12.10.0              h2ffa867_1    conda-forge
azure-storage-common-cpp  12.5.0               h09a5875_4    conda-forge
blosc                     1.21.5               hc338f07_0    conda-forge
boost                     1.82.0               hf3b2ce4_6    conda-forge
brotli                    1.1.0                hb547adb_1    conda-forge
brotli-bin                1.1.0                hb547adb_1    conda-forge
bullet                    3.25                 hf3b2ce4_2    conda-forge
bullet-cpp                3.25            py311h6e08293_2    conda-forge
bzip2                     1.0.8                h80987f9_6  
c-ares                    1.28.1               h93a5062_0    conda-forge
c-compiler                1.7.0                h6aa9301_1    conda-forge
ca-certificates           2024.7.4             hf0a4a13_0    conda-forge
cairo                     1.18.0               hd1e100b_0    conda-forge
catkin_pkg                1.0.0              pyhd8ed1ab_0    conda-forge
catkin_tools              0.9.4              pyhd8ed1ab_0    conda-forge
cctools                   986                  h4faf515_0    conda-forge
cctools_osx-arm64         986                  h62378fb_0    conda-forge
certifi                   2024.6.2           pyhd8ed1ab_0    conda-forge
cffi                      1.16.0          py311h4a08483_0    conda-forge
cfitsio                   4.3.1                h808cd33_0    conda-forge
clang                     16.0.6          default_h675cc0c_9    conda-forge
clang-16                  16.0.6          default_h5c12605_9    conda-forge
clang_impl_osx-arm64      16.0.6              hc421ffc_16    conda-forge
clang_osx-arm64           16.0.6              h54d7cd3_16    conda-forge
clangxx                   16.0.6          default_h675cc0c_9    conda-forge
clangxx_impl_osx-arm64    16.0.6              hcd7bac0_16    conda-forge
clangxx_osx-arm64         16.0.6              h54d7cd3_16    conda-forge
cmake                     3.28.3               h50fd54c_0    conda-forge
colcon-argcomplete        0.3.3              pyhd8ed1ab_0    conda-forge
colcon-bash               0.5.0              pyhd8ed1ab_0    conda-forge
colcon-cd                 0.1.1              pyhd8ed1ab_0    conda-forge
colcon-cmake              0.2.28             pyhd8ed1ab_0    conda-forge
colcon-common-extensions  0.3.0           py311h267d04e_1    conda-forge
colcon-core               0.17.0             pyhd8ed1ab_0    conda-forge
colcon-defaults           0.2.8              pyhd8ed1ab_0    conda-forge
colcon-devtools           0.3.0              pyhd8ed1ab_0    conda-forge
colcon-library-path       0.2.1                      py_0    conda-forge
colcon-metadata           0.2.5                      py_0    conda-forge
[... 658 lines cut ...]
          conda version : 24.5.0
    conda-build version : not installed
         python version : 3.11.8.final.0
                 solver : libmamba (default)
       virtual packages : __archspec=1=m1
                          __conda=24.5.0=0
                          __osx=14.5=0
                          __unix=0=0
       base environment : /Users/xingrui/miniconda3  (writable)
      conda av data dir : /Users/xingrui/miniconda3/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack-staging/osx-arm64
                          https://conda.anaconda.org/robostack-staging/noarch
                          https://conda.anaconda.org/conda-forge/osx-arm64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : /Users/xingrui/miniconda3/pkgs
                          /Users/xingrui/.conda/pkgs
       envs directories : /Users/xingrui/miniconda3/envs
                          /Users/xingrui/.conda/envs
               platform : osx-arm64
             user-agent : conda/24.5.0 requests/2.31.0 CPython/3.11.8 Darwin/23.5.0 OSX/14.5 solver/libmamba conda-libmamba-solver/23.11.1 libmambapy/1.5.8
                UID:GID : 501:20
             netrc file : None
           offline mode : False
```

traversaro, 2024-07-09:
Hello @xingruiyang, it would be easier to debug your issue if you could prepare a minimal reproducible example for your problem, do you think this is possible? Thanks!

Tobias-Fischer, 2024-07-09:
Xref https://github.com/RoboStack/ros-humble/issues/32

closed 2024-08-31 by xingruiyang

Tobias-Fischer, 2024-08-31:
How did you fix this @xingruiyang ?

xingruiyang, 2024-09-03:
I haven't. I just decided to stay with Ubuntu, it is a much easier solution.
Please feel free to reopen the issue if you think it's useful.