RoboStack/ros-humble #192: ModuleNotFoundError: No module named 'setuptools.extern' on new installation

shaunabanana, 2024-07-26:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

Hi! I noticed an issue building Python packages that wasn't present when I first tried about one or to month ago. Any help would be greatly appreciated! :)

**Issue**
Building Python packages raises `ModuleNotFoundError: No module named 'setuptools.extern'`.

**Steps to reproduce:**
1. Create a brand new installation of ROS Humble following the documentation (including installing dev dependencies).
2. Create a new package in an empty `<workspace>/src` using `ros2 pkg create --build-type ament_python demo`.
3. Run `colcon build` in the workspace.
4. Building fails due to the error above.

**System:**
- MacBook Pro M1
- macOS 14.4.1 (23E224)

**Full error logs:**
```
Starting >>> demo
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ModuleNotFoundError: No module named 'setuptools.extern'
--- stderr: demo
Traceback (most recent call last):
  File "/Users/shauna/.miniforge/envs/humble/lib/python3.11/site-packages/colcon_core/executor/__init__.py", line 91, in __call__
    rc = await self.task(*args, **kwargs)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/shauna/.miniforge/envs/humble/lib/python3.11/site-packages/colcon_core/task/__init__.py", line 93, in __call__
    return await task_method(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/shauna/.miniforge/envs/humble/lib/python3.11/site-packages/colcon_ros/task/ament_python/build.py", line 51, in build
    setup_py_data = get_setup_data(self.context.pkg, env)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/shauna/.miniforge/envs/humble/lib/python3.11/site-packages/colcon_core/task/python/__init__.py", line 20, in get_setup_data
    return dict(pkg.metadata[key](env))
                ^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/shauna/.miniforge/envs/humble/lib/python3.11/site-packages/colcon_ros/package_augmentation/ros_ament_python.py", line 57, in getter
    return get_setup_information(
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/shauna/.miniforge/envs/humble/lib/python3.11/site-packages/colcon_python_setup_py/package_identification/python_setup_py.py", line 241, in get_setup_information
    _setup_information_cache[hashable_env] = _get_setup_information(
                                             ^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/shauna/.miniforge/envs/humble/lib/python3.11/site-packages/colcon_python_setup_py/package_identification/python_setup_py.py", line 281, in _get_setup_information
    result = subprocess.run(
             ^^^^^^^^^^^^^^^
  File "/Users/shauna/.miniforge/envs/humble/lib/python3.11/subprocess.py", line 571, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['/Users/shauna/.miniforge/envs/humble/bin/python3.11', '-c', "import sys;from setuptools.extern.packaging.specifiers import SpecifierSet;from distutils.core import run_setup;dist = run_setup(    'setup.py', script_args=('--dry-run',), stop_after='config');skip_keys = ('cmdclass', 'distclass', 'ext_modules', 'metadata');data = {    key: value for key, value in dist.__dict__.items()     if (        not key.startswith('_') and         not callable(value) and         key not in skip_keys and         key not in dist.display_option_names    )};data['metadata'] = {    k: v for k, v in dist.metadata.__dict__.items()     if k not in ('license_files', 'provides_extras')};sys.stdout.buffer.write(repr(data).encode('utf-8'))"]' returned non-zero exit status 1.
---
Failed   <<< demo [0.16s, exited with code 1]

Summary: 0 packages finished [0.29s]
  1 package failed: demo
  1 package had stderr output: demo
Command '['/Users/shauna/.miniforge/envs/humble/bin/python3.11', '-c', "import sys;from setuptools.extern.packaging.specifiers import SpecifierSet;from distutils.core import run_setup;dist = run_setup(    'setup.py', script_args=('--dry-run',), stop_after='config');skip_keys = ('cmdclass', 'distclass', 'ext_modules', 'metadata');data = {    key: value for key, value in dist.__dict__.items()     if (        not key.startswith('_') and         not callable(value) and         key not in skip_keys and         key not in dist.display_option_names    )};data['metadata'] = {    k: v for k, v in dist.metadata.__dict__.items()     if k not in ('license_files', 'provides_extras')};sys.stdout.buffer.write(repr(data).encode('utf-8'))"]' returned non-zero exit status 1.
```

### Installed packages

```shell
# Name                    Version                   Build  Channel
aiohttp                   3.9.5           py311h05b510d_0    conda-forge
aiosignal                 1.3.1              pyhd8ed1ab_0    conda-forge
aom                       3.9.1                h7bae524_0    conda-forge
argcomplete               3.4.0              pyhd8ed1ab_0    conda-forge
[... 642 lines cut ...]
          conda version : 24.5.0
    conda-build version : not installed
         python version : 3.10.14.final.0
                 solver : libmamba (default)
       virtual packages : __archspec=1=m1
                          __conda=24.5.0=0
                          __osx=14.4.1=0
                          __unix=0=0
       base environment : /Users/shauna/.miniforge  (writable)
      conda av data dir : /Users/shauna/.miniforge/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack-staging/osx-arm64
                          https://conda.anaconda.org/robostack-staging/noarch
                          https://conda.anaconda.org/conda-forge/osx-arm64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : /Users/shauna/.miniforge/pkgs
                          /Users/shauna/.conda/pkgs
       envs directories : /Users/shauna/.miniforge/envs
                          /Users/shauna/.conda/envs
               platform : osx-arm64
             user-agent : conda/24.5.0 requests/2.31.0 CPython/3.10.14 Darwin/23.4.0 OSX/14.4.1 solver/libmamba conda-libmamba-solver/24.1.0 libmambapy/1.5.8
                UID:GID : 501:20
             netrc file : None
           offline mode : False
```

Tobias-Fischer, 2024-07-26:
See https://answers.ros.org/question/362883/fail-building-foxy-modulenotfounderror-no-module-named-setuptoolsextern/ - basically setuptools is new.

shaunabanana, 2024-07-26:
Thank you very much for the quick response!

After changing line 251 of `~/colcon_python_setup_py/package_identification/python_setup_py.py` from 
`'from setuptools.extern.packaging.specifiers import SpecifierSet',`
to
`'from packaging.specifiers import SpecifierSet',`
the package indeed compiles as normal.

I'll close this issue for now.

closed 2024-07-26 by shaunabanana