RoboStack/ros-humble #193: Colcon buld failure for a custom message on ROS 2 MacOS M2 (already following FAQ)

danitruc6, 2024-07-28:
I'm already following steps from [FAQ](https://robostack.github.io/FAQ.html).

I started getting the error:
```bash
CMake Error at /Users/danitruc/miniconda3/envs/ros2_env/share/cmake-3.28/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find Python3 (missing: Python3_NumPy_INCLUDE_DIRS NumPy) (found
  version "3.12.4")
```

Then I followed the [FAQ](https://robostack.github.io/FAQ.html) recommendation and got:
```bash
CMake Warning (dev) at /Users/danitruc/miniconda3/envs/ros2_env/share/rosidl_generator_py/cmake/rosidl_generator_py_generate_interfaces.cmake:20 (find_package):
  Policy CMP0148 is not set: The FindPythonInterp and FindPythonLibs modules
  are removed.  Run "cmake --help-policy CMP0148" for policy details.  Use
  the cmake_policy command to set the policy and suppress this warning.
```

For that I upgraded `cmake_minimum_required` from 3.8 to 3.28 to avoid the following error:
```bash
CMake Warning (dev) at /Users/danitruc/miniconda3/envs/ros2_env/share/rosidl_generator_py/cmake/rosidl_generator_py_generate_interfaces.cmake:20 (find_package):
  Policy CMP0148 is not set: The FindPythonInterp and FindPythonLibs modules
  are removed.  Run "cmake --help-policy CMP0148" for policy details.  Use
  the cmake_policy command to set the policy and suppress this warning.
```

And finally this is the error I'm getting and not knowing how to fix it.

```bash
❯ colcon build --packages-select tutorial_interfaces --cmake-args -DPython_EXECUTABLE=$CONDA_PREFIX/bin/python -DPython3_EXECUTABLE=$CONDA_PREFIX/bin/python -DPYTHON_EXECUTABLE=$CONDA_PREFIX/bin/python -DPython3_FIND_STRATEGY=LOCATION -DPython_FIND_STRATEGY=LOCATION
Starting >>> tutorial_interfaces
--- stderr: tutorial_interfaces
CMake Error at /Users/danitruc/miniconda3/envs/ros2_env/share/rosidl_generator_py/cmake/rosidl_generator_py_generate_interfaces.cmake:20 (find_package):
  By not providing "FindPythonInterp.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "PythonInterp", but CMake did not find one.

  Could not find a package configuration file provided by "PythonInterp"
  (requested version 3.6) with any of the following names:

    PythonInterpConfig.cmake
    pythoninterp-config.cmake

  Add the installation prefix of "PythonInterp" to CMAKE_PREFIX_PATH or set
  "PythonInterp_DIR" to a directory containing one of the above files.  If
  "PythonInterp" provides a separate development package or SDK, be sure it
  has been installed.
Call Stack (most recent call first):
  /Users/danitruc/miniconda3/envs/ros2_env/share/ament_cmake_core/cmake/core/ament_execute_extensions.cmake:48 (include)
  /Users/danitruc/miniconda3/envs/ros2_env/share/rosidl_cmake/cmake/rosidl_generate_interfaces.cmake:286 (ament_execute_extensions)
  CMakeLists.txt:13 (rosidl_generate_interfaces)

make: *** [Makefile:747: cmake_check_build_system] Error 1
---
Failed   <<< tutorial_interfaces [1.30s, exited with code 2]

Summary: 0 packages finished [1.62s]
  1 package failed: tutorial_interfaces
  1 package had stderr output: tutorial_interfaces
```

CMakeLists.txt file:
```txt
cmake_minimum_required(VERSION 3.27)
project(tutorial_interfaces)

if(CMAKE_COMPILER_IS_GNUCXX OR CMAKE_CXX_COMPILER_ID MATCHES "Clang")
  add_compile_options(-Wall -Wextra -Wpedantic)
endif()

# find dependencies
[... 677 lines cut ...]
xorg-libxrender           0.9.11               hb547adb_0    conda-forge
xorg-libxt                1.3.0                hb547adb_1    conda-forge
xorg-randrproto           1.5.0             h3422bc3_1001    conda-forge
xorg-renderproto          0.11.1            h27ca646_1002    conda-forge
xorg-xextproto            7.3.0             h1a8c8d9_1003    conda-forge
xorg-xproto               7.0.31            h27ca646_1007    conda-forge
xz                        5.2.6                h57fd34a_0    conda-forge
yaml                      0.2.5                h3422bc3_2    conda-forge
yaml-cpp                  0.8.0                h13dd4ca_0    conda-forge
yarl                      1.9.4           py311h05b510d_0    conda-forge
zipp                      3.19.2             pyhd8ed1ab_0    conda-forge
zlib                      1.2.13               hfb2fe0b_6    conda-forge
zstd                      1.5.6                hb46c0d2_0    conda-forge
zziplib                   0.13.69              he1e0b03_1    conda-forge
```

conda env list
```bash
❯ conda env list
# conda environments:
#
base                     /Users/danitruc/miniconda3
ros2_env              *  /Users/danitruc/miniconda3/envs/ros2_env

```

danitruc6, 2024-07-28:
Oh yes, you are right, the build succeeded.

I guess I can fix the CMP0148 by adding:
```txt
if(POLICY CMP0148) 
  cmake_policy(SET CMP0148 OLD) 
endif()
```

Just before closing this issue, I would like to know the following.

I'm still getting this part of the error, is this expected since I'm on MacOS? It has somethind to do with the code signing.

```bash
❯ colcon build --packages-select tutorial_interfaces --cmake-args -DPython_EXECUTABLE=$CONDA_PREFIX/bin/python -DPython3_EXECUTABLE=$CONDA_PREFIX/bin/python -DPYTHON_EXECUTABLE=$CONDA_PREFIX/bin/python -DPython3_FIND_STRATEGY=LOCATION -DPython_FIND_STRATEGY=LOCATION
Starting >>> tutorial_interfaces
--- stderr: tutorial_interfaces
/Users/danitruc/miniconda3/envs/ros2_env/bin/install_name_tool: warning: changes being made to the file will invalidate the code signature in: /Users/danitruc/ros2_ws/install/tutorial_interfaces/lib/libtutorial_interfaces__rosidl_typesupport_fastrtps_c.dylib
[cctools-port]: generating fake signature for '/Users/danitruc/ros2_ws/install/tutorial_interfaces/lib/libtutorial_interfaces__rosidl_typesupport_fastrtps_c.dylib'
/Users/danitruc/miniconda3/envs/ros2_env/bin/install_name_tool: warning: changes being made to the file will invalidate the code signature in: /Users/danitruc/ros2_ws/install/tutorial_interfaces/lib/libtutorial_interfaces__rosidl_typesupport_introspection_c.dylib
[cctools-port]: generating fake signature for '/Users/danitruc/ros2_ws/install/tutorial_interfaces/lib/libtutorial_interfaces__rosidl_typesupport_introspection_c.dylib'
/Users/danitruc/miniconda3/envs/ros2_env/bin/install_name_tool: warning: changes being made to the file will invalidate the code signature in: /Users/danitruc/ros2_ws/install/tutorial_interfaces/lib/libtutorial_interfaces__rosidl_typesupport_c.dylib
[cctools-port]: generating fake signature for '/Users/danitruc/ros2_ws/install/tutorial_interfaces/lib/libtutorial_interfaces__rosidl_typesupport_c.dylib'
error: /Users/danitruc/miniconda3/envs/ros2_env/bin/install_name_tool: no LC_RPATH load command with path: /Users/danitruc/ros2_ws/build/tutorial_interfaces/rosidl_generator_py/tutorial_interfaces found in: /Users/danitruc/ros2_ws/install/tutorial_interfaces/lib/python3.11/site-packages/tutorial_interfaces/tutorial_interfaces_s__rosidl_typesupport_fastrtps_c.cpython-311-darwin.so (for architecture arm64), required for specified option "-delete_rpath /Users/danitruc/ros2_ws/build/tutorial_interfaces/rosidl_generator_py/tutorial_interfaces"
error: /Users/danitruc/miniconda3/envs/ros2_env/bin/install_name_tool: no LC_RPATH load command with path: /Users/danitruc/ros2_ws/build/tutorial_interfaces found in: /Users/danitruc/ros2_ws/install/tutorial_interfaces/lib/python3.11/site-packages/tutorial_interfaces/tutorial_interfaces_s__rosidl_typesupport_fastrtps_c.cpython-311-darwin.so (for architecture arm64), required for specified option "-delete_rpath /Users/danitruc/ros2_ws/build/tutorial_interfaces"
error: /Users/danitruc/miniconda3/envs/ros2_env/bin/install_name_tool: no LC_RPATH load command with path: /Users/danitruc/ros2_ws/build/tutorial_interfaces/rosidl_generator_py/tutorial_interfaces found in: /Users/danitruc/ros2_ws/install/tutorial_interfaces/lib/python3.11/site-packages/tutorial_interfaces/tutorial_interfaces_s__rosidl_typesupport_introspection_c.cpython-311-darwin.so (for architecture arm64), required for specified option "-delete_rpath /Users/danitruc/ros2_ws/build/tutorial_interfaces/rosidl_generator_py/tutorial_interfaces"
error: /Users/danitruc/miniconda3/envs/ros2_env/bin/install_name_tool: no LC_RPATH load command with path: /Users/danitruc/ros2_ws/build/tutorial_interfaces found in: /Users/danitruc/ros2_ws/install/tutorial_interfaces/lib/python3.11/site-packages/tutorial_interfaces/tutorial_interfaces_s__rosidl_typesupport_introspection_c.cpython-311-darwin.so (for architecture arm64), required for specified option "-delete_rpath /Users/danitruc/ros2_ws/build/tutorial_interfaces"
error: /Users/danitruc/miniconda3/envs/ros2_env/bin/install_name_tool: no LC_RPATH load command with path: /Users/danitruc/ros2_ws/build/tutorial_interfaces/rosidl_generator_py/tutorial_interfaces found in: /Users/danitruc/ros2_ws/install/tutorial_interfaces/lib/python3.11/site-packages/tutorial_interfaces/tutorial_interfaces_s__rosidl_typesupport_c.cpython-311-darwin.so (for architecture arm64), required for specified option "-delete_rpath /Users/danitruc/ros2_ws/build/tutorial_interfaces/rosidl_generator_py/tutorial_interfaces"
error: /Users/danitruc/miniconda3/envs/ros2_env/bin/install_name_tool: no LC_RPATH load command with path: /Users/danitruc/ros2_ws/build/tutorial_interfaces found in: /Users/danitruc/ros2_ws/install/tutorial_interfaces/lib/python3.11/site-packages/tutorial_interfaces/tutorial_interfaces_s__rosidl_typesupport_c.cpython-311-darwin.so (for architecture arm64), required for specified option "-delete_rpath /Users/danitruc/ros2_ws/build/tutorial_interfaces"
error: /Users/danitruc/miniconda3/envs/ros2_env/bin/install_name_tool: no LC_RPATH load command with path: /Users/danitruc/ros2_ws/build/tutorial_interfaces found in: /Users/danitruc/ros2_ws/install/tutorial_interfaces/lib/libtutorial_interfaces__rosidl_generator_py.dylib (for architecture arm64), required for specified option "-delete_rpath /Users/danitruc/ros2_ws/build/tutorial_interfaces"
---
Finished <<< tutorial_interfaces [2.23s]

Summary: 1 package finished [2.39s]
  1 package had stderr output: tutorial_interfaces
```

Tobias-Fischer, 2024-07-28:
I think you can safely ignore these :). They’ve been around for a while, and seeing that the binaries work, it’s too much trouble to debug.

closed 2024-07-28 by Tobias-Fischer

danitruc6, 2024-07-29:
Got it, thank you very much @Tobias-Fischer