RoboStack/ros-humble #194: Error loading server extension jupyros ImportError: Error importing numpy:

USeebi, 2024-08-02:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

I'm not sure if it is a bug, my platform is a portable version of Miniconda3 py311 on Win10, I have no problem to launch ros_noetic but ros_humble seems to be a very different species.

The jupyros extension is installed in the base env, the extension loaded as expected in the base as well as in ros_noetic but got errors below in ros_humble.

I tried the fix from https://stackoverflow.com/questions/61663400/dll-load-failed-multiarray-unmath-when-importing-numpy with
set PATH=%PATH%;K:\Miniconda3\py311\Library\bin
but no help!

What did I miss? What is the correct version of numpy or Visual C++ Redistributable should be used?

[W 15:55:54.850 NotebookApp] Error loading server extension jupyros
    Traceback (most recent call last):
      File "K:\Miniconda3\py311\envs\ros_humble\Lib\site-packages\numpy\core\__init__.py", line 24, in <module>
        from . import multiarray
      File "K:\Miniconda3\py311\envs\ros_humble\Lib\site-packages\numpy\core\multiarray.py", line 10, in <module>
        from . import overrides
      File "K:\Miniconda3\py311\envs\ros_humble\Lib\site-packages\numpy\core\overrides.py", line 8, in <module>
        from numpy.core._multiarray_umath import (
    ImportError: DLL load failed while importing _multiarray_umath: The specified module could not be found.

    During handling of the above exception, another exception occurred:

    Traceback (most recent call last):
      File "K:\Miniconda3\py311\envs\ros_humble\Lib\site-packages\numpy\__init__.py", line 130, in <module>
        from numpy.__config__ import show as show_config
      File "K:\Miniconda3\py311\envs\ros_humble\Lib\site-packages\numpy\__config__.py", line 4, in <module>
        from numpy.core._multiarray_umath import (
      File "K:\Miniconda3\py311\envs\ros_humble\Lib\site-packages\numpy\core\__init__.py", line 50, in <module>
        raise ImportError(msg)
    ImportError:

    IMPORTANT: PLEASE READ THIS FOR ADVICE ON HOW TO SOLVE THIS ISSUE!

    Importing the numpy C-extensions failed. This error can happen for
    many reasons, often due to issues with your setup or how NumPy was
    installed.

    We have compiled some common reasons and troubleshooting tips at:

        https://numpy.org/devdocs/user/troubleshooting-importerror.html

    Please note and check the following:

      * The Python version is: Python3.11 from "K:\Miniconda3\py311\python.exe"
      * The NumPy version is: "1.26.4"

    and make sure that they are the versions you expect.
    Please carefully study the documentation linked above for further help.

    Original error was: DLL load failed while importing _multiarray_umath: The specified module could not be found.

    The above exception was the direct cause of the following exception:

    Traceback (most recent call last):
      File "K:\Miniconda3\py311\Lib\site-packages\notebook\notebookapp.py", line 2047, in init_server_extensions
        mod = importlib.import_module(modulename)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
      File "K:\Miniconda3\py311\Lib\importlib\__init__.py", line 126, in import_module
        return _bootstrap._gcd_import(name[level:], package, level)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
      File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
      File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
      File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
      File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
[... 727 lines cut ...]
       base environment : K:\Miniconda3\py311  (writable)
      conda av data dir : K:\Miniconda3\py311\etc\conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/conda-forge/win-64
                          https://conda.anaconda.org/conda-forge/noarch
                          https://conda.anaconda.org/robostack-staging/win-64
                          https://conda.anaconda.org/robostack-staging/noarch
                          https://repo.anaconda.com/pkgs/main/win-64
                          https://repo.anaconda.com/pkgs/main/noarch
                          https://repo.anaconda.com/pkgs/r/win-64
                          https://repo.anaconda.com/pkgs/r/noarch
                          https://repo.anaconda.com/pkgs/msys2/win-64
                          https://repo.anaconda.com/pkgs/msys2/noarch
          package cache : K:\Miniconda3\py311\pkgs
                          C:\Users\useeb\.conda\pkgs
                          C:\Users\useeb\AppData\Local\conda\conda\pkgs
       envs directories : K:\Miniconda3\py311\envs
                          C:\Users\useeb\.conda\envs
                          C:\Users\useeb\AppData\Local\conda\conda\envs
               platform : win-64
             user-agent : conda/24.7.1 requests/2.32.2 CPython/3.11.9 Windows/10 Windows/10.0.19045 solver/libmamba conda-libmamba-solver/24.1.0 libmambapy/1.5.8 aau/0.4.4 c/. s/. e/.
          administrator : False
             netrc file : None
           offline mode : False
```

USeebi, 2024-08-08:
Finally I sorted out the problem by comparing PYTHONPATH between ros_noetic and ros_humble.  It seems that the jupyros extension loaded some sort of precedent to the numpy=2.0.1 package installed with it while the PYTHONPATH of ros_humble point to numpy=1.26.4 and lead to the DLL load failure.

I installed numpy=2.0.1 in the ros_humble env and it works.  Down the road another offending package is PIL/pillow, I applied the same trick but I got other problems with PIL in Jupyter.  By pure fluke I managed to get it done by deleting ros_humble's pillow package while copying the PIL library from jupyros to the PYTHONPATH of ros_humble.

I just wonder are these just asynchronous in development between jupyros and ros_desktop or I did miss something?  How the precedent arise in the jupyter extension if there is really any such precedent? 

I did try to install jupyros in the same ros_humble env but you still got the same PIL/pillow problem to solve.  What would be other successful Windows installations of current distribution of RoboStack out there?

traversaro, 2024-08-08:
> I did try to install jupyros in the same ros_humble env but you still got the same PIL/pillow problem to solve.

Sorry, which kind of PIL/pillow issue you are having? Thanks!

Tobias-Fischer, 2024-08-08:
Do you mix conda channels or pip and conda any chance?

USeebi, 2024-08-08:
> > I did try to install jupyros in the same ros_humble env but you still got the same PIL/pillow problem to solve.
> 
> Sorry, which kind of PIL/pillow issue you are having? Thanks!

I'm referring to the same problem of replacing pillow=10.3.0 in ros_humble with pillow=10.4.0 in jupyros if you install jupyros after ros_humble_desktop.  I don't have the setup to reproduce the problems but I can illustrate with the correct ones

[image]

This display would be magnified at least 3 - 4 times.

[image]

The Start Stop widgets buttons not running due to some javascript problems.

There are some other issues to run the ipynb in notebook.

USeebi, 2024-08-08:
> Do you mix conda channels or pip and conda any chance?

jupyros was installed by pip install --pre jupyros==0.7.0a0 according to https://github.com/RoboStack/jupyter-ros/issues/157 in the base env

ros_humble_desktop are installed strictly by the Getting Started manual.

What would be other valid combinations?