RoboStack/ros-humble #195: Package gazebo-ros is not starting the service /spawn_entity

danitruc6, 2024-08-07:
Can someone please help me to understand the following issue of why is /spawn_entity not started on my RoboStack env, like is normally started on my Ubuntu VM.

I have installed gazebo classic through the following commands:
```bash
mamba install ros-humble-gazebo-ros 
mamba install ros-humble-gazebo-ros-pkgs
```

Run the robot_state_publisher:
```bash
❯ ros2 run robot_state_publisher robot_state_publisher --ros-args -p robot_description:="$(xacro /Users/danitruc/ros2_ws/src/my_robot_description/urdf/my_robot.urdf.xacro)"
```

If I launch gazebo:
```bash
❯ ros2 launch gazebo_ros gazebo.launch.py
[INFO] [launch]: All log files can be found below /Users/danitruc/.ros/log/2024-08-06-21-13-46-474094-trucs-mac.local-52784
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [gzserver-1]: process started with pid [52795]
[INFO] [gzclient-2]: process started with pid [52796]
[gzserver-1] [INFO] [1723000482.719135357] [gazebo_ros_node]: ROS was initialized without arguments.
```

And I want to spawn my urdf in gazebo using the spawn script, however I'm getting this message:

```bash
❯ ros2 run gazebo_ros spawn_entity.py -topic robot_description -entity my_robot
[INFO] [1723002072.188968220] [spawn_entity]: Spawn Entity started
[INFO] [1723002072.189483118] [spawn_entity]: Loading entity published on topic robot_description
/Users/danitruc/miniforge3/envs/ros2_env/lib/python3.11/site-packages/rclpy/qos.py:307: UserWarning: DurabilityPolicy.RMW_QOS_POLICY_DURABILITY_TRANSIENT_LOCAL is deprecated. Use DurabilityPolicy.TRANSIENT_LOCAL instead.
  warnings.warn(
[INFO] [1723002072.194511184] [spawn_entity]: Waiting for entity xml on robot_description
[INFO] [1723002072.197840937] [spawn_entity]: Waiting for service /spawn_entity, timeout = 30
[INFO] [1723002072.198231507] [spawn_entity]: Waiting for service /spawn_entity
[ERROR] [1723002105.373467090] [spawn_entity]: Service %s/spawn_entity unavailable. Was Gazebo started with GazeboRosFactory?
[ERROR] [1723002105.374930204] [spawn_entity]: Spawn service failed. Exiting.
[ros2run]: Process exited with failure 1
```

If I check the service list:
```bash
❯ ros2 service list
/apply_joint_effort
/apply_link_wrench
/clear_joint_efforts
/clear_link_wrenches
/gazebo/describe_parameters
/gazebo/get_parameter_types
/gazebo/get_parameters
/gazebo/list_parameters
/gazebo/set_parameters
/gazebo/set_parameters_atomically
/spawn_entity/describe_parameters
/spawn_entity/get_parameter_types
/spawn_entity/get_parameters
/spawn_entity/list_parameters
/spawn_entity/set_parameters
/spawn_entity/set_parameters_atomically
```

And this is the gazebo_ros node info:
```bash
❯ ros2 node info /gazebo
/gazebo
  Subscribers:
    /clock: rosgraph_msgs/msg/Clock
    /parameter_events: rcl_interfaces/msg/ParameterEvent
  Publishers:
    /parameter_events: rcl_interfaces/msg/ParameterEvent
    /rosout: rcl_interfaces/msg/Log
[... 753 lines cut ...]
xorg-libx11               1.8.9                h570a39f_0    conda-forge
xorg-libxau               1.0.11               hb547adb_0    conda-forge
xorg-libxaw               1.0.14               h3422bc3_1    conda-forge
xorg-libxdmcp             1.1.3                h27ca646_0    conda-forge
xorg-libxext              1.3.4                h1a8c8d9_2    conda-forge
xorg-libxfixes            5.0.3             h3422bc3_1004    conda-forge
xorg-libxmu               1.1.3                h99b78c6_1    conda-forge
xorg-libxpm               3.5.17               hb547adb_0    conda-forge
xorg-libxrandr            1.5.2                h3422bc3_1    conda-forge
xorg-libxrender           0.9.11               hb547adb_0    conda-forge
xorg-libxt                1.3.0                hb547adb_1    conda-forge
xorg-randrproto           1.5.0             h3422bc3_1001    conda-forge
xorg-renderproto          0.11.1            h27ca646_1002    conda-forge
xorg-xextproto            7.3.0             h1a8c8d9_1003    conda-forge
xorg-xproto               7.0.31            h27ca646_1007    conda-forge
xz                        5.2.6                h57fd34a_0    conda-forge
yaml                      0.2.5                h3422bc3_2    conda-forge
yaml-cpp                  0.8.0                h13dd4ca_0    conda-forge
yarl                      1.9.4           py311h05b510d_0    conda-forge
zeromq                    4.3.5                hcc0f68c_4    conda-forge
zipp                      3.19.2             pyhd8ed1ab_0    conda-forge
zlib                      1.2.13               hfb2fe0b_6    conda-forge
zstd                      1.5.6                hb46c0d2_0    conda-forge
zziplib                   0.13.69              he1e0b03_1    conda-forge
```

traversaro, 2024-08-07:
Could this be related to https://github.com/RoboStack/ros-humble/issues/175 ? Can you try to install `libprotobuf==4.25.1` ?

danitruc6, 2024-08-07:
Oh yes, that ibprotobuf==4.25.1 did the trick, and now I'm getting all of the expected gazebo_services working. 

Thank you very much @traversaro.

### gazebo_ros node info:
As a reference, in case someone is havin the same issue as me, I just ran:
```bash
mamba install libprotobuf==4.25.1
```

And now all the expected services are up and running:
```bash
❯ ros2 node info /gazebo
/gazebo
  Subscribers:
    /clock: rosgraph_msgs/msg/Clock
    /parameter_events: rcl_interfaces/msg/ParameterEvent
  Publishers:
    /clock: rosgraph_msgs/msg/Clock
    /parameter_events: rcl_interfaces/msg/ParameterEvent
    /performance_metrics: gazebo_msgs/msg/PerformanceMetrics
    /rosout: rcl_interfaces/msg/Log
  Service Servers:
    /apply_joint_effort: gazebo_msgs/srv/ApplyJointEffort
    /apply_link_wrench: gazebo_msgs/srv/ApplyLinkWrench
    /clear_joint_efforts: gazebo_msgs/srv/JointRequest
    /clear_link_wrenches: gazebo_msgs/srv/LinkRequest
    /delete_entity: gazebo_msgs/srv/DeleteEntity
    /gazebo/describe_parameters: rcl_interfaces/srv/DescribeParameters
    /gazebo/get_parameter_types: rcl_interfaces/srv/GetParameterTypes
    /gazebo/get_parameters: rcl_interfaces/srv/GetParameters
    /gazebo/list_parameters: rcl_interfaces/srv/ListParameters
    /gazebo/set_parameters: rcl_interfaces/srv/SetParameters
    /gazebo/set_parameters_atomically: rcl_interfaces/srv/SetParametersAtomically
    /get_model_list: gazebo_msgs/srv/GetModelList
    /pause_physics: std_srvs/srv/Empty
    /reset_simulation: std_srvs/srv/Empty
    /reset_world: std_srvs/srv/Empty
    /spawn_entity: gazebo_msgs/srv/SpawnEntity
    /unpause_physics: std_srvs/srv/Empty
  Service Clients:

  Action Servers:

  Action Clients:
```

closed 2024-08-07 by danitruc6

traversaro, 2024-08-08:
Thanks for opening the issue! Unfortunately in robostack we can't fix wrong dependency information for old packages, and a proper fix will require a rebuild of all packages, that will take time.