RoboStack/ros-humble #196: Ros-Humble-Gazebo-Ros-Pkgs Not Installing Gazebo ROS API Plugins on OSX-arm64

alexanderkjones, 2024-08-14:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

When installing ros-humble-gazebo-ros-pkgs from the robostack-staging channel gazebo-ros plugins are not installed in gazebo-11/plugins directory or any directory on host machine. For example when running `gzserver --verbose -s libgazebo_ros_init.so ` the response is

```
[Err] [Plugin.hh:212] Failed to load plugin libgazebo_ros_init.so: dlopen(libgazebo_ros_init.so, 0x0009): tried: '/opt/miniconda3/envs/ros2_humble/opt/rviz_ogre_vendor/lib/libgazebo_ros_init.so' (no such file), '/libgazebo_ros_init.so' (no such file), '/opt/miniconda3/envs/ros2_humble/lib/gazebo-11/plugins/libgazebo_ros_init.so' (no such file), '/libgazebo_ros_init.so'
```

The same is true for `libgazebo_ros_factory.so`. Having done a system wide search I cannot find libgazebo_ros_factory or any of the other plugins in any directory.

I ran across this issue trying to spawn a urdf in Gazebo 11. The spawn_entity.py returns the following error:

```
[INFO] [1723503078.578469654] [spawn_entity]: Spawn Entity started
[INFO] [1723503078.579020789] [spawn_entity]: Loading entity published on topic robot_description
[INFO] [1723503078.585026898] [spawn_entity]: Waiting for entity xml on robot_description
[INFO] [1723503078.588259124] [spawn_entity]: Waiting for service /spawn_entity, timeout = 30
[INFO] [1723503078.588627589] [spawn_entity]: Waiting for service /spawn_entity
[ERROR] [1723503108.975306091] [spawn_entity]: Service %s/spawn_entity unavailable. Was Gazebo started with GazeboRosFactory?
[ERROR] [1723503108.978652402] [spawn_entity]: Spawn service failed. Exiting.
```

Note: showing only robostack-staging packages below due to character limit in posting issue.

### Installed packages

```shell
# packages in environment at /opt/miniconda3/envs/ros2_humble:
#
# Name                    Version                   Build  Channel
ros-humble-ackermann-msgs 2.0.2           py311hbfa132e_5    robostack-staging
ros-humble-ackermann-steering-controller 2.32.0          py311hbfa132e_5    robostack-staging
ros-humble-action-msgs    1.2.1           py311hbfa132e_5    robostack-staging
ros-humble-action-tutorials-cpp 0.20.3          py311hbfa132e_5    robostack-staging
ros-humble-action-tutorials-interfaces 0.20.3          py311hbfa132e_5    robostack-staging
ros-humble-action-tutorials-py 0.20.3          py311hbfa132e_5    robostack-staging
ros-humble-actionlib-msgs 4.2.3           py311hbfa132e_5    robostack-staging
ros-humble-admittance-controller 2.32.0          py311hbfa132e_5    robostack-staging
ros-humble-ament-cmake    1.3.7           py311hbfa132e_5    robostack-staging
ros-humble-ament-cmake-auto 1.3.7           py311hbfa132e_5    robostack-staging
ros-humble-ament-cmake-copyright 0.12.10         py311hbfa132e_5    robostack-staging
ros-humble-ament-cmake-core 1.3.7           py311hbfa132e_5    robostack-staging
ros-humble-ament-cmake-cppcheck 0.12.10         py311hbfa132e_5    robostack-staging
ros-humble-ament-cmake-cpplint 0.12.10         py311hbfa132e_5    robostack-staging
ros-humble-ament-cmake-export-definitions 1.3.7           py311hbfa132e_5    robostack-staging
ros-humble-ament-cmake-export-dependencies 1.3.7           py311hbfa132e_5    robostack-staging
ros-humble-ament-cmake-export-include-directories 1.3.7           py311hbfa132e_5    robostack-staging
ros-humble-ament-cmake-export-interfaces 1.3.7           py311hbfa132e_5    robostack-staging
ros-humble-ament-cmake-export-libraries 1.3.7           py311hbfa132e_5    robostack-staging
ros-humble-ament-cmake-export-link-flags 1.3.7           py311hbfa132e_5    robostack-staging
ros-humble-ament-cmake-export-targets 1.3.7           py311hbfa132e_5    robostack-staging
ros-humble-ament-cmake-flake8 0.12.10         py311hbfa132e_5    robostack-staging
ros-humble-ament-cmake-gen-version-h 1.3.7           py311hbfa132e_5    robostack-staging
ros-humble-ament-cmake-gmock 1.3.7           py311hbfa132e_5    robostack-staging
ros-humble-ament-cmake-gtest 1.3.7           py311hbfa132e_5    robostack-staging
ros-humble-ament-cmake-include-directories 1.3.7           py311hbfa132e_5    robostack-staging
ros-humble-ament-cmake-libraries 1.3.7           py311hbfa132e_5    robostack-staging
ros-humble-ament-cmake-lint-cmake 0.12.10         py311hbfa132e_5    robostack-staging
ros-humble-ament-cmake-pep257 0.12.10         py311hbfa132e_5    robostack-staging
ros-humble-ament-cmake-pytest 1.3.7           py311hbfa132e_5    robostack-staging
ros-humble-ament-cmake-python 1.3.7           py311hbfa132e_5    robostack-staging
ros-humble-ament-cmake-ros 0.10.0          py311hbfa132e_5    robostack-staging
ros-humble-ament-cmake-target-dependencies 1.3.7           py311hbfa132e_5    robostack-staging
ros-humble-ament-cmake-test 1.3.7           py311hbfa132e_5    robostack-staging
ros-humble-ament-cmake-uncrustify 0.12.10         py311hbfa132e_5    robostack-staging
ros-humble-ament-cmake-version 1.3.7           py311hbfa132e_5    robostack-staging
[... 353 lines cut ...]
          conda version : 24.7.1
    conda-build version : not installed
         python version : 3.12.2.final.0
                 solver : libmamba (default)
       virtual packages : __archspec=1=m2
                          __conda=24.7.1=0
                          __osx=14.2.1=0
                          __unix=0=0
       base environment : /opt/miniconda3  (writable)
      conda av data dir : /opt/miniconda3/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack-staging/osx-arm64
                          https://conda.anaconda.org/robostack-staging/noarch
                          https://conda.anaconda.org/conda-forge/osx-arm64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : /opt/miniconda3/pkgs
                          /Users/alexanderjones/.conda/pkgs
       envs directories : /opt/miniconda3/envs
                          /Users/alexanderjones/.conda/envs
               platform : osx-arm64
             user-agent : conda/24.7.1 requests/2.32.2 CPython/3.12.2 Darwin/23.2.0 OSX/14.2.1 solver/libmamba conda-libmamba-solver/24.1.0 libmambapy
/1.5.8 aau/0.4.4 c/. s/. e/.                                                                                                                                          UID:GID : 501:20
             netrc file : None
           offline mode : False
```

traversaro, 2024-08-14:
Please report all the packages in your environment, not only the one related to ROS. I suspect this may be related to https://github.com/RoboStack/ros-humble/issues/175, can you try to install the reported version of libprotobuf?

alexanderkjones, 2024-08-14:
@traversaro thanks for taking a look. Complete package list here.

```
# packages in environment at /opt/miniconda3/envs/ros2_humble:
#
# Name                    Version                   Build  Channel
aiohttp                   3.9.5           py311h05b510d_0    conda-forge
aiosignal                 1.3.1              pyhd8ed1ab_0    conda-forge
aom                       3.8.2                h078ce10_0    conda-forge
archspec                  0.2.3              pyhd8ed1ab_0    conda-forge
argcomplete               3.4.0              pyhd8ed1ab_0    conda-forge
assimp                    5.3.1                he63ff86_2    conda-forge
atk-1.0                   2.38.0               hcb7b3dd_1    conda-forge
attrs                     24.1.0             pyh71513ae_0    conda-forge
aws-c-auth                0.7.17               h382b9c6_2    conda-forge
aws-c-cal                 0.6.11               hd34e5fa_0    conda-forge
aws-c-common              0.9.15               h93a5062_0    conda-forge
aws-c-compression         0.2.18               hd34e5fa_3    conda-forge
aws-c-event-stream        0.4.2                h247c08a_8    conda-forge
aws-c-http                0.8.1               hf9e830b_10    conda-forge
aws-c-io                  0.14.7               h33d81b3_6    conda-forge
aws-c-mqtt                0.10.3               h5f4abda_4    conda-forge
aws-c-s3                  0.5.7                h606a3d2_1    conda-forge
aws-c-sdkutils            0.1.15               hd34e5fa_3    conda-forge
aws-checksums             0.1.18               hd34e5fa_3    conda-forge
aws-crt-cpp               0.26.6               h13f0230_4    conda-forge
aws-sdk-cpp               1.11.267             h134aaec_6    conda-forge
azure-core-cpp            1.11.1               he231e37_1    conda-forge
azure-storage-blobs-cpp   12.10.0              h2ffa867_1    conda-forge
azure-storage-common-cpp  12.5.0               h09a5875_4    conda-forge
benchmark                 1.8.5                h00cdb27_0    conda-forge
blosc                     1.21.5               hc338f07_0    conda-forge
boltons                   24.0.0             pyhd8ed1ab_0    conda-forge
boost                     1.82.0               hf3b2ce4_6    conda-forge
boost-cpp                 1.82.0               hca5e981_6    conda-forge
brotli                    1.1.0                hb547adb_1    conda-forge
brotli-bin                1.1.0                hb547adb_1    conda-forge
brotli-python             1.1.0           py311ha891d26_1    conda-forge
bullet                    3.25                 hf3b2ce4_2    conda-forge
bullet-cpp                3.25            py311h6e08293_2    conda-forge
bzip2                     1.0.8                h99b78c6_7    conda-forge
c-ares                    1.32.3               h99b78c6_0    conda-forge
c-compiler                1.7.0                h6aa9301_1    conda-forge
ca-certificates           2024.7.4             hf0a4a13_0    conda-forge
cairo                     1.18.0               hd1e100b_0    conda-forge
catkin_pkg                1.0.0              pyhd8ed1ab_0    conda-forge
catkin_tools              0.9.4              pyhd8ed1ab_0    conda-forge
cctools                   986                  h4faf515_0    conda-forge
cctools_osx-arm64         986                  h62378fb_0    conda-forge
ceres-solver              2.2.0                h036b7f2_2    conda-forge
certifi                   2024.7.4           pyhd8ed1ab_0    conda-forge
cffi                      1.16.0          py311h4a08483_0    conda-forge
cfitsio                   4.3.1                h808cd33_0    conda-forge
charset-normalizer        3.3.2              pyhd8ed1ab_0    conda-forge
clang                     16.0.6          default_h675cc0c_11    conda-forge
clang-16                  16.0.6          default_h5c12605_11    conda-forge
clang_impl_osx-arm64      16.0.6              hc421ffc_18    conda-forge
clang_osx-arm64           16.0.6              h54d7cd3_18    conda-forge
clangxx                   16.0.6          default_h675cc0c_11    conda-forge
clangxx_impl_osx-arm64    16.0.6              hcd7bac0_18    conda-forge
clangxx_osx-arm64         16.0.6              h54d7cd3_18    conda-forge
cmake                     3.28.3               h50fd54c_0    conda-forge
colcon-argcomplete        0.3.3              pyhd8ed1ab_0    conda-forge
colcon-bash               0.5.0              pyhd8ed1ab_0    conda-forge
colcon-cd                 0.1.1              pyhd8ed1ab_0    conda-forge
colcon-cmake              0.2.28             pyhd8ed1ab_0    conda-forge
colcon-common-extensions  0.3.0           py311h267d04e_1    conda-forge
colcon-core               0.17.1             pyhd8ed1ab_0    conda-forge
colcon-defaults           0.2.8              pyhd8ed1ab_0    conda-forge
colcon-devtools           0.3.0              pyhd8ed1ab_0    conda-forge
[... 761 lines cut ...]
xorg-libxext              1.3.4                h1a8c8d9_2    conda-forge
xorg-libxfixes            5.0.3             h3422bc3_1004    conda-forge
xorg-libxmu               1.1.3                h99b78c6_1    conda-forge
xorg-libxpm               3.5.17               hb547adb_0    conda-forge
xorg-libxrandr            1.5.2                h3422bc3_1    conda-forge
xorg-libxrender           0.9.11               hb547adb_0    conda-forge
xorg-libxt                1.3.0                hb547adb_1    conda-forge
xorg-randrproto           1.5.0             h3422bc3_1001    conda-forge
[... cut ...]
yarl                      1.9.4           py311h05b510d_0    conda-forge
zeromq                    4.3.5                hcc0f68c_4    conda-forge
zipp                      3.19.2             pyhd8ed1ab_0    conda-forge
zlib                      1.2.13               hfb2fe0b_6    conda-forge
zstandard                 0.23.0          py311h4a6b76e_0    conda-forge
zstd                      1.5.6                hb46c0d2_0    conda-forge
zziplib                   0.13.69              he1e0b03_1    conda-forge
```

Tobias-Fischer, 2024-08-14:
As @traversaro suggested, with your current version of libprotobuf, we know that this does not work. Can you please install 4.25.1 and let us know if that helped?

alexanderkjones, 2024-08-15:
@traversaro @Tobias-Fischer, thanks for your help. Using `install libprotobuf=4.25.1` then installing `ros-humble-gazebo-ros-pkgs` was successful.

closed 2024-08-15 by Tobias-Fischer