RoboStack/ros-humble #197: CMake Error at CMakeLists.txt:10 (find_package):   By not providing "Findcatkin.cmake" in CMAKE_MODULE_PATH from colcon build

USeebi, 2024-08-15:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

colcon build failed with package generated from 

catkin_create_pkg mrobot_description urdf xacro

already 

set CMAKE_PREFIX_PATH=K:\Miniconda3\envs\ros_humble

### Installed packages

```shell
# packages in environment at K:\Miniconda3\envs\ros_humble:
#
# Name                    Version                   Build  Channel
aiohappyeyeballs          2.3.5              pyhd8ed1ab_0    conda-forge
aiohttp                   3.10.2          py311he736701_0    conda-forge
aiosignal                 1.3.1              pyhd8ed1ab_0    conda-forge
aom                       3.8.2                h63175ca_0    conda-forge
argcomplete               3.5.0              pyhd8ed1ab_0    conda-forge
assimp                    5.3.1                h81f0834_2    conda-forge
asttokens                 2.4.1              pyhd8ed1ab_0    conda-forge
attrs                     24.2.0             pyh71513ae_0    conda-forge
aws-c-auth                0.7.17               hb40cdec_2    conda-forge
aws-c-cal                 0.6.11               ha21e00f_0    conda-forge
aws-c-common              0.9.15               hcfcfb64_0    conda-forge
aws-c-compression         0.2.18               ha21e00f_3    conda-forge
aws-c-event-stream        0.4.2                hf668b60_8    conda-forge
aws-c-http                0.8.1               hd704247_10    conda-forge
aws-c-io                  0.14.7               h14865c8_6    conda-forge
aws-c-mqtt                0.10.3               h748201e_4    conda-forge
aws-c-s3                  0.5.7                h5da7064_1    conda-forge
aws-c-sdkutils            0.1.15               ha21e00f_3    conda-forge
aws-checksums             0.1.18               ha21e00f_3    conda-forge
aws-crt-cpp               0.26.6               h5bc0ceb_4    conda-forge
aws-sdk-cpp               1.11.267             h5d77392_6    conda-forge
azure-core-cpp            1.11.1               h249a519_1    conda-forge
azure-storage-blobs-cpp   12.10.0              h91493d7_1    conda-forge
azure-storage-common-cpp  12.5.0               h91493d7_4    conda-forge
blosc                     1.21.5               hdccc3a2_0    conda-forge
boost                     1.82.0               h3810d55_6    conda-forge
brotli                    1.1.0                hcfcfb64_1    conda-forge
brotli-bin                1.1.0                hcfcfb64_1    conda-forge
bullet                    3.25                 h3810d55_2    conda-forge
bullet-cpp                3.25                 hf63dbb6_2    conda-forge
bzip2                     1.0.8                h2bbff1b_6
c-ares                    1.32.3               h2466b09_0    conda-forge
c-compiler                1.7.0                hcfcfb64_1    conda-forge
ca-certificates           2024.7.4             h56e8100_0    conda-forge
cairo                     1.18.0               h1fef639_0    conda-forge
catkin_pkg                1.0.0              pyhd8ed1ab_0    conda-forge
catkin_tools              0.8.2              pyhd8ed1ab_0    conda-forge
certifi                   2024.7.4           pyhd8ed1ab_0    conda-forge
cffi                      1.17.0          py311he736701_0    conda-forge
cfitsio                   4.3.1                h9b0cee5_0    conda-forge
clangdev                  5.0.0                   flang_3    conda-forge
cmake                     3.28.3               hf0feee3_0    conda-forge
colcon-cmake              0.2.28             pyhd8ed1ab_0    conda-forge
colcon-common-extensions  0.3.0           py311h1ea47a8_1    conda-forge
colcon-core               0.17.1             pyhd8ed1ab_0    conda-forge
colcon-defaults           0.2.8              pyhd8ed1ab_0    conda-forge
colcon-devtools           0.3.0              pyhd8ed1ab_0    conda-forge
colcon-library-path       0.2.1                      py_0    conda-forge
colcon-metadata           0.2.5                      py_0    conda-forge
colcon-notification       0.3.0           py311h1ea47a8_0    conda-forge
[... 650 lines cut ...]
    conda-build version : not installed
         python version : 3.11.9.final.0
                 solver : libmamba (default)
       virtual packages : __archspec=1=nehalem
                          __conda=24.7.1=0
                          __win=0=0
       base environment : K:\Miniconda3\py311  (writable)
      conda av data dir : K:\Miniconda3\py311\etc\conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack-staging/win-64
                          https://conda.anaconda.org/robostack-staging/noarch
                          https://conda.anaconda.org/conda-forge/win-64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : K:\Miniconda3\py311\pkgs
                          C:\Users\useeb\.conda\pkgs
                          C:\Users\useeb\AppData\Local\conda\conda\pkgs
       envs directories : K:\Miniconda3\py311\envs
                          C:\Users\useeb\.conda\envs
                          C:\Users\useeb\AppData\Local\conda\conda\envs
               platform : win-64
             user-agent : conda/24.7.1 requests/2.32.2 CPython/3.11.9 Windows/10 Windows/10.0.19045 solver/libmamba conda-libmamba-solver/24.1.0 libmambapy/1.5.8
          administrator : False
             netrc file : None
           offline mode : False
```

traversaro, 2024-08-15:
What is the error if you not modify manually the CMAKE_PREFIX_PATH?

traversaro, 2024-08-15:
Are you using Command Prompt?

USeebi, 2024-08-16:
The CMAKE_PREFIX_PATH is empty originally and definitely I'm using cmd not ps.

I know I'm inexperience in building catkin package in ros2 but there are  strange error outputs from 

colcon info

```(ros_humble) K:\Miniconda3>colcon info
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "K:\Miniconda3\envs\ros_humble\Lib\site-packages\setuptools\_distutils\core.py", line 266, in run_setup
    exec(code, g)
  File "<string>", line 37, in <module>
IndexError: list index out of range
[6.318s] colcon.colcon_core.package_identification ERROR Exception in package identification extension 'python_setup_py' in 'envs\ros_humble\Lib\site-packages\adodbapi': Command '['K:\\Miniconda3\\envs\\ros_humble\\python.exe', '-c', "import sys;from packaging.specifiers import SpecifierSet;from distutils.core import run_setup;dist = run_setup(    'setup.py', script_args=('--dry-run',), stop_after='config');skip_keys = ('cmdclass', 'distclass', 'ext_modules', 'metadata');data = {    key: value for key, value in dist.__dict__.items()     if (        not key.startswith('_') and         not callable(value) and         key not in skip_keys and         key not in dist.display_option_names    )};data['metadata'] = {    k: v for k, v in dist.metadata.__dict__.items()     if k not in ('license_files', 'provides_extras')};sys.stdout.buffer.write(repr(data).encode('utf-8'))"]' returned non-zero exit status 1.
Traceback (most recent call last):
  File "K:\Miniconda3\envs\ros_humble\Lib\site-packages\colcon_core\package_identification\__init__.py", line 144, in _identify
    retval = extension.identify(_reused_descriptor_instance)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "K:\Miniconda3\envs\ros_humble\Lib\site-packages\colcon_python_setup_py\package_identification\python_setup_py.py", line 48, in identify
    config = get_setup_information(setup_py)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "K:\Miniconda3\envs\ros_humble\Lib\site-packages\colcon_python_setup_py\package_identification\python_setup_py.py", line 241, in get_setup_information
    _setup_information_cache[hashable_env] = _get_setup_information(
                                             ^^^^^^^^^^^^^^^^^^^^^^^
  File "K:\Miniconda3\envs\ros_humble\Lib\site-packages\colcon_python_setup_py\package_identification\python_setup_py.py", line 283, in _get_setup_information
    result = subprocess.run(
             ^^^^^^^^^^^^^^^
  File "K:\Miniconda3\envs\ros_humble\Lib\subprocess.py", line 571, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['K:\\Miniconda3\\envs\\ros_humble\\python.exe', '-c', "import sys;from packaging.specifiers import SpecifierSet;from distutils.core import run_setup;dist = run_setup(    'setup.py', script_args=('--dry-run',), stop_after='config');skip_keys = ('cmdclass', 'distclass', 'ext_modules', 'metadata');data = {    key: value for key, value in dist.__dict__.items()     if (        not key.startswith('_') and         not callable(value) and         key not in skip_keys and         key not in dist.display_option_names    )};data['metadata'] = {    k: v for k, v in dist.metadata.__dict__.items()     if k not in ('license_files', 'provides_extras')};sys.stdout.buffer.write(repr(data).encode('utf-8'))"]' returned non-zero exit status 1.

<string>:9: DeprecationWarning:

  `numpy.distutils` is deprecated since NumPy 1.23.0, as a result
  of the deprecation of `distutils` itself. It will be removed for
  Python >= 3.12. For older Python versions it will remain present.
  It is recommended to use `setuptools < 60.0` for those Python versions.
  For more details, see:
    https://numpy.org/devdocs/reference/distutils_status_migration.html

Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "K:\Miniconda3\envs\ros_humble\Lib\site-packages\setuptools\_distutils\core.py", line 276, in run_setup
    raise RuntimeError(
RuntimeError: 'distutils.core.setup()' was never called -- perhaps 'setup.py' is not a Distutils setup script?
[12.641s] colcon.colcon_core.package_identification ERROR Exception in package identification extension 'python_setup_py' in 'envs\ros_humble\Lib\site-packages\numpy\_typing': Command '['K:\\Miniconda3\\envs\\ros_humble\\python.exe', '-c', "import sys;from packaging.specifiers import SpecifierSet;from distutils.core import run_setup;dist = run_setup(    'setup.py', script_args=('--dry-run',), stop_after='config');skip_keys = ('cmdclass', 'distclass', 'ext_modules', 'metadata');data = {    key: value for key, value in dist.__dict__.items()     if (        not key.startswith('_') and         not callable(value) and         key not in skip_keys and         key not in dist.display_option_names    )};data['metadata'] = {    k: v for k, v in dist.metadata.__dict__.items()     if k not in ('license_files', 'provides_extras')};sys.stdout.buffer.write(repr(data).encode('utf-8'))"]' returned non-zero exit status 1.
Traceback (most recent call last):
  File "K:\Miniconda3\envs\ros_humble\Lib\site-packages\colcon_core\package_identification\__init__.py", line 144, in _identify
    retval = extension.identify(_reused_descriptor_instance)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "K:\Miniconda3\envs\ros_humble\Lib\site-packages\colcon_python_setup_py\package_identification\python_setup_py.py", line 48, in identify
    config = get_setup_information(setup_py)
[... cut ...]
  File "K:\Miniconda3\envs\ros_humble\Lib\site-packages\pkg_resources\__init__.py", line 3102, in _get_metadata
    if self.has_metadata(name):
       ^^^^^^^^^^^^^^^^^^^^^^^
  File "K:\Miniconda3\envs\ros_humble\Lib\site-packages\pkg_resources\__init__.py", line 1668, in has_metadata
    return self._has(path)
           ^^^^^^^^^^^^^^^
  File "K:\Miniconda3\envs\ros_humble\Lib\site-packages\pkg_resources\__init__.py", line 1876, in _has
    return os.path.exists(path)
           ^^^^^^^^^^^^^^^^^^^^
  File "<frozen genericpath>", line 19, in exists```

USeebi, 2024-08-16:
The above error outputs have to be stopped by Crtl-C.

Did I miss anything?

Tobias-Fischer, 2024-08-16:
You are mixing packages from pypi and conda-forge. Please only use conda/mamba to install packages, otherwise weird stuff like this can happen. I recommend creating a new environment and avoiding pip as much as possible.

Tobias-Fischer, 2024-08-16:
Feel free to reopen if the problem persists

closed 2024-08-16 by Tobias-Fischer