RoboStack/ros-humble #198: ros-humble-gazebo-ros2-control : Gazebo Ros2 Control Plugin Not Loading

alexanderkjones, 2024-08-15:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

Installing ros-humble-gazebo-ros-pkgs and ros-humble-gazebo-ros2-control results in gazebo plugins installing in the miniconda3/env/[name]/lib folder. I have made sure to install `libprotobuf=4.25.1`. When launching Gazebo with plugin args it appropriately finds gazebo-ros plugins but will not load gazebo-ros2-control plugins in the same directory. 

For example, `gzserver --verbose -s libgazebo_ros_init.so -s libgazebo_ros_factory.so` yields no errors and launches gazebo.

However, `gzserver --verbose  -s libgazebo_ros2_control.so` yields: 
```
[Err] [Plugin.hh:212] 
Failed to load plugin libgazebo_ros2_control.so: dlopen(libgazebo_ros2_control.so 0x0009): tried: ...
Failed to load plugin libgazebo_ros2_control.dylib: dlopen(libgazebo_ros2_control.dylib
0x0009): tried: ...
```

In an attempt to debug the issue I added the dylib file to the `DYLD_FALLBACK_LIBRARY_PATH`
```
(ros_humble) (base) alexanderjones@Alexanders-MacBook-Air dev_ws % echo $DYLD_FALLBACK_LIBRARY_PATH
/Users/alexanderjones/miniconda3/envs/ros_humble/lib
```

Which yielded:
```
(ros_humble) (base) alexanderjones@Alexanders-MacBook-Air dev_ws % gzserver --verbose  -s libgazebo_ros2_control.so
Gazebo multi-robot simulator, version 11.14.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Err] [gazebo_shared.cc:46] System is attempting to load a plugin, but detected an incorrect plugin type. Plugin filename[libgazebo_ros2_control.so].
zsh: segmentation fault  gzserver --verbose -s libgazebo_ros2_control.so
```

I can confirm the file exists
```
(ros_humble) (base) alexanderjones@Alexanders-MacBook-Air dev_ws % test -f /Users/alexanderjones/miniconda3/envs/ros_humble/lib/libgazebo_ros2_control.dylib && echo "File exists" || echo "File does not exist"
File exists
```

### Installed packages

```shell
# packages in environment at /Users/alexanderjones/miniconda3/envs/ros_humble:
#
# Name                    Version                   Build  Channel
aiohappyeyeballs          2.3.5              pyhd8ed1ab_0    conda-forge
aiohttp                   3.10.3          py311hd3f4193_0    conda-forge
aiosignal                 1.3.1              pyhd8ed1ab_0    conda-forge
aom                       3.8.2                h078ce10_0    conda-forge
argcomplete               3.5.0              pyhd8ed1ab_0    conda-forge
assimp                    5.3.1                he63ff86_2    conda-forge
atk-1.0                   2.38.0               hcb7b3dd_1    conda-forge
attrs                     24.2.0             pyh71513ae_0    conda-forge
aws-c-auth                0.7.16               h51b92d1_6    conda-forge
aws-c-cal                 0.6.10               hf888d4c_1    conda-forge
aws-c-common              0.9.13               h93a5062_0    conda-forge
aws-c-compression         0.2.18               hf888d4c_1    conda-forge
aws-c-event-stream        0.4.2                h1b95c93_4    conda-forge
aws-c-http                0.8.1                heb97226_5    conda-forge
aws-c-io                  0.14.5               hb51f26e_1    conda-forge
aws-c-mqtt                0.10.2               hd37591b_4    conda-forge
aws-c-s3                  0.5.2                h09f5dbc_0    conda-forge
aws-c-sdkutils            0.1.15               hf888d4c_1    conda-forge
aws-checksums             0.1.18               hf888d4c_1    conda-forge
aws-crt-cpp               0.26.2               h06549c9_7    conda-forge
aws-sdk-cpp               1.11.267             h73c0887_1    conda-forge
azure-core-cpp            1.10.3               he231e37_1    conda-forge
[... 685 lines cut ...]
          conda version : 24.5.0
    conda-build version : not installed
         python version : 3.12.4.final.0
                 solver : libmamba (default)
       virtual packages : __archspec=1=m2
                          __conda=24.5.0=0
                          __osx=14.2.1=0
                          __unix=0=0
       base environment : /Users/alexanderjones/miniconda3  (writable)
      conda av data dir : /Users/alexanderjones/miniconda3/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack-staging/osx-arm64
                          https://conda.anaconda.org/robostack-staging/noarch
                          https://conda.anaconda.org/conda-forge/osx-arm64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : /Users/alexanderjones/miniconda3/pkgs
                          /Users/alexanderjones/.conda/pkgs
       envs directories : /Users/alexanderjones/miniconda3/envs
                          /Users/alexanderjones/.conda/envs
               platform : osx-arm64
             user-agent : conda/24.5.0 requests/2.32.2 CPython/3.12.4 Darwin/23.2.0 OSX/14.2.1 solver/libmamba conda-libmamba-solver/24.1.0 libmambapy/1.5.
8 aau/0.4.4 c/. s/. e/.                                                                                                                                                    UID:GID : 501:20
             netrc file : None
           offline mode : False
```

traversaro, 2024-08-16:
Can you try to report of otool -L for the problematic plugins?

alexanderkjones, 2024-08-16:
@traversaro, thanks for taking a look. Results here for `libgazebo_ros2_control.dylib`. 

```
(ros_humble) (base) alexanderjones@Alexanders-MacBook-Air dev_ws % otool -L /Users/alexanderjones/miniconda3/envs/ros_humble/lib/libgazebo_ros2_control.dylib
/Users/alexanderjones/miniconda3/envs/ros_humble/lib/libgazebo_ros2_control.dylib:
        @rpath/libgazebo_ros2_control.dylib (compatibility version 0.0.0, current version 0.0.0)
        @rpath/libcontroller_manager.dylib (compatibility version 0.0.0, current version 0.0.0)
        @rpath/libament_index_cpp.dylib (compatibility version 0.0.0, current version 0.0.0)
        @rpath/libgazebo_ros_node.dylib (compatibility version 0.0.0, current version 0.0.0)
        @rpath/librclcpp.dylib (compatibility version 0.0.0, current version 0.0.0)
        @rpath/libgazebo_physics.11.dylib (compatibility version 11.0.0, current version 11.14.0)
        @rpath/libgazebo_common.11.dylib (compatibility version 11.0.0, current version 11.14.0)
        @rpath/libsdformat9.9.dylib (compatibility version 9.0.0, current version 9.8.0)
        @rpath/libhardware_interface.dylib (compatibility version 0.0.0, current version 0.0.0)
        @rpath/libconsole_bridge.1.0.dylib (compatibility version 1.0.0, current version 0.0.0)
        @rpath/libclass_loader.dylib (compatibility version 0.0.0, current version 0.0.0)
        @rpath/libtinyxml2.10.dylib (compatibility version 10.0.0, current version 10.0.0)
        @rpath/librcl.dylib (compatibility version 0.0.0, current version 0.0.0)
        @rpath/librclcpp_lifecycle.dylib (compatibility version 0.0.0, current version 0.0.0)
        @rpath/librcpputils.dylib (compatibility version 0.0.0, current version 0.0.0)
        @rpath/librcutils.dylib (compatibility version 0.0.0, current version 0.0.0)
        /usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1336.0.0)
        @rpath/libignition-math6.6.dylib (compatibility version 6.0.0, current version 6.15.1)
        @rpath/libc++.1.dylib (compatibility version 1.0.0, current version 1.0.0)
```

traversaro, 2024-08-19:
Sorry, I am back at the laptop and I was able to check better. I think it is expected that `gzserver --verbose  -s libgazebo_ros2_control.so` fails, as `libgazebo_ros2_control.so` is a modelplugin and not a system plugin, see https://github.com/ros-controls/gazebo_ros2_control/blob/b1b555f9af6bf56e07aa210001fe9cd3c46526ce/gazebo_ros2_control/include/gazebo_ros2_control/gazebo_ros2_control_plugin.hpp#L51 .

Just to understand, why are you trying to load the plugin as `gzserver --verbose  -s libgazebo_ros2_control.so` ?

alexanderkjones, 2024-08-19:
@traversaro thanks for pointing that out. To clarify, I am attempting to launch gazebo with the ros2_control plugin. Gazebo successfully launches and spawns my URDF, but fails to start a ros2_control manager. Having run the launch file with verbose output I was able to identify the error `[gzserver-2] [Err] [Plugin.hh:212] Failed to load plugin libgazebo_ros2_control.so` which is what lead me to believe debugging at the gzserver level would be productive. I'm attaching below the full output. Highlighted plugin error message below with 

[gzserver-2] [Err] [Plugin.hh:212] Failed to load plugin libgazebo_ros2_control.so: dlopen(libgazebo_ros2_control.so, 0x0009): tried: '/Users/alexanderjones/dev_ws/install/interface_examples/lib/libgazebo_ros2_control.so' (no such file), '/Users/alexanderjones/miniconda3/envs/ros_env/lib/gazebo-11/plugins/libgazebo_ros2_control.so' (no such file), '/Users/alexanderjones/miniconda3/envs/ros_env/opt/rviz_ogre_vendor/lib/libgazebo_ros2_control.so' (no such file), '/Users/alexanderjones/miniconda3/envs/ros_humble/opt/rviz_ogre_vendor/lib/libgazebo_ros2_control.so' (no such file), '/libgazebo_ros2_control.so' (no such file), '/Users/alexanderjones/miniconda3/envs/ros_humble/lib/gazebo-11/plugins/libgazebo_ros2_control.so' (no such file), '/libgazebo_ros2_control.so' (no such file), 'libgazebo_ros2_control.so' (no such file), '/System/Volumes/Preboot/Cryptexes/OSlibgazebo_ros2_control.so' (no such file), '/Users/alexanderjones/miniconda3/envs/ros_env/lib/libgazebo_ros2_control.so' (no such file), '/Users/alexanderjones/miniconda3/envs/ros_env/bin/../lib/libgazebo_ros2_control.so' (no such file), '/usr/lib/libgazebo_ros2_control.so' (no such file, not in dyld cache), 'libgazebo_ros2_control.so' (no such file), '/usr/local/lib/libgazebo_ros2_control.so' (no such file), '/usr/lib/libgazebo_ros2_control.so' (no such file, not in dyld cache)
[gzserver-2] Failed to load plugin libgazebo_ros2_control.dylib: dlopen(libgazebo_ros2_control.dylib, 0x0009): tried: '/Users/alexanderjones/dev_ws/install/interface_examples/lib/libgazebo_ros2_control.dylib' (no such file), '/Users/alexanderjones/miniconda3/envs/ros_env/lib/gazebo-11/plugins/libgazebo_ros2_control.dylib' (no such file), '/Users/alexanderjones/miniconda3/envs/ros_env/opt/rviz_ogre_vendor/lib/libgazebo_ros2_control.dylib' (no such file), '/Users/alexanderjones/miniconda3/envs/ros_humble/opt/rviz_ogre_vendor/lib/libgazebo_ros2_control.dylib' (no such file), '/libgazebo_ros2_control.dylib' (no such file), '/Users/alexanderjones/miniconda3/envs/ros_humble/lib/gazebo-11/plugins/libgazebo_ros2_control.dylib' (no such file), '/libgazebo_ros2_control.dylib' (no such file), 'libgazebo_ros2_control.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OSlibgazebo_ros2_control.dylib' (no such file), '/Users/alexanderjones/miniconda3/envs/ros_env/lib/libgazebo_ros2_control.dylib' (no such file), '/Users/alexanderjones/miniconda3/envs/ros_env/bin/../lib/libgazebo_ros2_control.dylib' (no such file), '/usr/lib/libgazebo_ros2_control.dylib' (no such file, not in dyld cache), 'libgazebo_ros2_control.dylib' (no such file), '/usr/local/lib/libgazebo_ros2_control.dylib' (no such file), '/usr/lib/libgazebo_ros2_control.dylib' (no such file, not in dyld cache)

[INFO] [spawn_entity.py-4]: process has finished cleanly [pid 33868]
[DEBUG] [launch.launch_context]: emitting event: 'launch.events.process.ProcessExited'
[DEBUG] [launch]: processing event: '<launch.events.process.process_exited.ProcessExited object at 0x103287ad0>'
[DEBUG] [launch]: processing event: '<launch.events.process.process_exited.ProcessExited object at 0x103287ad0>' ✓ '<launch.event_handlers.on_process_exit.OnProcessExit object at 0x1032fd850>'
[DEBUG] [launch]: processing event: '<launch.events.process.process_exited.ProcessExited object at 0x103287ad0>' ✓ '<launch.event_handlers.on_process_exit.OnProcessExit object at 0x1032ffbd0>'
[DEBUG] [launch.launch_context]: emitting event synchronously: 'launch.events.process.ProcessStdout'
[DEBUG] [launch.launch_context]: emitting event synchronously: 'launch.events.process.ProcessStderr'
[DEBUG] [launch]: processing event: '<launch.events.process.process_stdout.ProcessStdout object at 0x10396b8d0>'
[DEBUG] [launch]: processing event: '<launch.events.process.process_stdout.ProcessStdout object at 0x10396b8d0>' ✓ '<launch.event_handlers.on_process_io.OnProcessIO object at 0x1032f4790>'
[gzclient-3] [Msg] Waiting for master.
[gzclient-3] [Msg] Connected to gazebo master @ http://127.0.0.1:11345
[gzclient-3] [Msg] Publicized address: 10.0.0.72
[DEBUG] [launch]: processing event: '<launch.events.process.process_stderr.ProcessStderr object at 0x1032eb550>'
[DEBUG] [launch]: processing event: '<launch.events.process.process_stderr.ProcessStderr object at 0x1032eb550>' ✓ '<launch.event_handlers.on_process_io.OnProcessIO object at 0x1032f4790>'
[gzclient-3] [Wrn] [GuiIface.cc:114] Populating font family aliases took 153 ms. Replace uses of missing font family "Sans" with one that exists to avoid this cost. 

```

alexanderkjones, 2024-08-19:
@traversaro update to this issue:

I tried using `export DYLD_FALLBACK_LIBRARY_PATH=/Users/alexanderjones/miniconda3/envs/ros_humble/lib` and this resolved the initial issue of the `failed to load plugin` error. Not sure why gazebo-ros is loading plugins from that folder but gazebo-ros2-control is not?

Additionally, fixing the lib issue has resulted in exposing a secondary issue in gazebo-ros2-control when trying to launch. The controller manager is never launched due to this parsing error. Not sure if it's related to the lib/path issues.

```
[gzserver-2] [INFO] [1724096010.041434763] [gazebo_ros2_control]: Received urdf from param server, parsing...
[ERROR] [gzserver-2]: process has died [pid 53807, exit code -11, cmd 'gzserver --verbose -slibgazebo_ros_init.so -slibgazebo_ros_factory.so -slibgazebo_ros_force_system.so'].
```

traversaro, 2024-08-19:
For the parsing error, which locale are you using? As far as I remember, `ros2_control` had some locale issue that should be solved, but you may try to set a C locale to understand if that fixes the problem, see https://github.com/ros-controls/ros2_control/pull/921 for more info.

alexanderkjones, 2024-08-19:
@traversaro thanks for all the pointers here. 

Checked current locale settings:
```
(ros_humble) (base) alexanderjones@Alexanders-MacBook-Air dev_ws % locale
LANG="en_US.UTF-8"
LC_COLLATE="en_US.UTF-8"
LC_CTYPE="en_US.UTF-8"
LC_MESSAGES="en_US.UTF-8"
LC_MONETARY="en_US.UTF-8"
LC_NUMERIC="en_US.UTF-8"
LC_TIME="en_US.UTF-8"
LC_ALL=
```
Then applied C to all
```
(ros_humble) (base) alexanderjones@Alexanders-MacBook-Air dev_ws % export LC_ALL=C
(ros_humble) (base) alexanderjones@Alexanders-MacBook-Air dev_ws % locale         
LANG="en_US.UTF-8"
LC_COLLATE="C"
LC_CTYPE="C"
LC_MESSAGES="C"
LC_MONETARY="C"
LC_NUMERIC="C"
LC_TIME="C"
LC_ALL="C"
```
And I get the same output below with gserver failing right after showing an attempt to parse the URDF. Are there any ways to further log debug between the parsing step and the server failing?

```
(ros_humble) (base) alexanderjones@Alexanders-MacBook-Air dev_ws % ros2 launch diff_drive_robot_example diff_drive_robot_example_ros2_control.launch.py -a -d
[DEBUG] [launch.launch_context]: emitting event synchronously: 'launch.events.IncludeLaunchDescription'
[INFO] [launch]: All log files can be found below /Users/alexanderjones/.ros/log/2024-08-19-14-16-14-822882-Alexanders-MacBook-Air.local-65163
[INFO] [launch]: Default logging verbosity is set to DEBUG
[DEBUG] [launch]: processing event: '<launch.events.include_launch_description.IncludeLaunchDescription object at 0x102cd1250>'
[DEBUG] [launch]: processing event: '<launch.events.include_launch_description.IncludeLaunchDescription object at 0x102cd1250>' ✓ '<launch.event_handlers.on_include_launch_description.OnIncludeLaunchDescription object at 0x102cf6390>'
[DEBUG] [launch.launch_description_source]: Traceback (most recent call last):
  File "/Users/alexanderjones/miniconda3/envs/ros_env/lib/python3.11/site-packages/launch/launch_description_source.py", line 68, in try_get_launch_description_without_context
    perform_substitutions(context, self.__location)
  File "/Users/alexanderjones/miniconda3/envs/ros_env/lib/python3.11/site-packages/launch/utilities/perform_substitutions_impl.py", line 26, in perform_substitutions
    return ''.join([context.perform_substitution(sub) for sub in subs])
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/alexanderjones/miniconda3/envs/ros_env/lib/python3.11/site-packages/launch/utilities/perform_substitutions_impl.py", line 26, in <listcomp>
    return ''.join([context.perform_substitution(sub) for sub in subs])
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/alexanderjones/miniconda3/envs/ros_env/lib/python3.11/site-packages/launch/launch_context.py", line 240, in perform_substitution
    return substitution.perform(self)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/alexanderjones/miniconda3/envs/ros_env/lib/python3.11/site-packages/launch/substitutions/this_launch_file_dir.py", line 56, in perform
    raise SubstitutionFailure(
launch.substitutions.substitution_failure.SubstitutionFailure: ThisLaunchFileDir used outside of a launch file (in a script)

[DEBUG] [launch.launch_description_source]: Failed to load the launch file without a context: ThisLaunchFileDir used outside of a launch file (in a script)
[DEBUG] [launch.launch_description_source]: Traceback (most recent call last):
  File "/Users/alexanderjones/miniconda3/envs/ros_env/lib/python3.11/site-packages/launch/launch_description_source.py", line 68, in try_get_launch_description_without_context
    perform_substitutions(context, self.__location)
  File "/Users/alexanderjones/miniconda3/envs/ros_env/lib/python3.11/site-packages/launch/utilities/perform_substitutions_impl.py", line 26, in perform_substitutions
    return ''.join([context.perform_substitution(sub) for sub in subs])
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/alexanderjones/miniconda3/envs/ros_env/lib/python3.11/site-packages/launch/utilities/perform_substitutions_impl.py", line 26, in <listcomp>
    return ''.join([context.perform_substitution(sub) for sub in subs])
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/alexanderjones/miniconda3/envs/ros_env/lib/python3.11/site-packages/launch/launch_context.py", line 240, in perform_substitution
    return substitution.perform(self)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/alexanderjones/miniconda3/envs/ros_env/lib/python3.11/site-packages/launch/substitutions/this_launch_file_dir.py", line 56, in perform
    raise SubstitutionFailure(
launch.substitutions.substitution_failure.SubstitutionFailure: ThisLaunchFileDir used outside of a launch file (in a script)

[DEBUG] [launch.launch_description_source]: Failed to load the launch file without a context: ThisLaunchFileDir used outside of a launch file (in a script)
[... 243 lines cut ...]
[DEBUG] [launch]: processing event: '<launch.events.process.process_stderr.ProcessStderr object at 0x1032332d0>' ✓ '<launch.event_handlers.on_process_io.OnProcessIO object at 0x103918750>'
[spawner-5] [INFO] [1724102183.890386443] [spawner_diff_drive_controller]: Waiting for '/controller_manager' node to exist
[DEBUG] [launch.launch_context]: emitting event synchronously: 'launch.events.process.ProcessStderr'
[DEBUG] [launch]: processing event: '<launch.events.process.process_stderr.ProcessStderr object at 0x103231e10>'
[... cut ...]
[DEBUG] [launch]: processing event: '<launch.events.process.process_exited.ProcessExited object at 0x103918450>'
[DEBUG] [launch]: processing event: '<launch.events.process.process_exited.ProcessExited object at 0x103918450>' ✓ '<launch.event_handlers.on_process_exit.OnProcessExit object at 0x10391bb10>'
[DEBUG] [launch]: processing event: '<launch.events.process.process_exited.ProcessExited object at 0x103918450>' ✓ '<launch.event_handlers.on_process_exit.OnProcessExit object at 0x10391bd90>'
```

alexanderkjones, 2024-08-19:
@traversaro additional information, I attempted to load the gazebo-ros2-control-demos to investigate wether my launch files were the culprit. I am getting the same error on the demos as I am on my own package:

```
(ros_humble) (base) alexanderjones@Alexanders-MacBook-Air dev_ws % ros2 launch gazebo_ros2_control_demos cart_example_position.launch.py

[INFO] [launch]: All log files can be found below /Users/alexanderjones/.ros/log/2024-08-19-14-50-25-082477-Alexanders-MacBook-Air.local-72717
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [gzserver-1]: process started with pid [72718]
[INFO] [gzclient-2]: process started with pid [72719]
[INFO] [robot_state_publisher-3]: process started with pid [72720]
[INFO] [spawn_entity.py-4]: process started with pid [72721]
[robot_state_publisher-3] [INFO] [1724104225.541490094] [robot_state_publisher]: got segment cart
[robot_state_publisher-3] [INFO] [1724104225.541894388] [robot_state_publisher]: got segment slideBar
[robot_state_publisher-3] [INFO] [1724104225.541921180] [robot_state_publisher]: got segment world
[spawn_entity.py-4] [INFO] [1724104225.831507459] [spawn_entity]: Spawn Entity started
[spawn_entity.py-4] [INFO] [1724104225.832134921] [spawn_entity]: Loading entity published on topic robot_description
[spawn_entity.py-4] /Users/alexanderjones/miniconda3/envs/ros_env/lib/python3.11/site-packages/rclpy/qos.py:307: UserWarning: DurabilityPolicy.RMW_QOS_POLICY_DURABILITY_TRANSIENT_LOCAL is deprecated. Use DurabilityPolicy.TRANSIENT_LOCAL instead.
[spawn_entity.py-4]   warnings.warn(
[spawn_entity.py-4] [INFO] [1724104225.872751557] [spawn_entity]: Waiting for entity xml on robot_description
[spawn_entity.py-4] [INFO] [1724104225.875730993] [spawn_entity]: Waiting for service /spawn_entity, timeout = 30
[spawn_entity.py-4] [INFO] [1724104225.876180079] [spawn_entity]: Waiting for service /spawn_entity
[spawn_entity.py-4] [INFO] [1724104227.516024920] [spawn_entity]: Calling service /spawn_entity
[spawn_entity.py-4] [INFO] [1724104227.573785375] [spawn_entity]: Spawn status: SpawnEntity: Successfully spawned entity [cart]
[gzserver-1] [INFO] [1724104227.647184139] [gazebo_ros2_control]: Loading gazebo_ros2_control plugin
[gzserver-1] [INFO] [1724104227.648616773] [gazebo_ros2_control]: Starting gazebo_ros2_control plugin in namespace: /
[gzserver-1] [INFO] [1724104227.648638315] [gazebo_ros2_control]: Starting gazebo_ros2_control plugin in ros 2 node: gazebo_ros2_control
[gzserver-1] [INFO] [1724104227.648800899] [gazebo_ros2_control]: Loading parameter files /Users/alexanderjones/dev_ws/install/diff_drive_robot_example/share/diff_drive_robot_example/config/diff_drive_robot_example_controllers_config.yaml
[gzserver-1] [INFO] [1724104227.650371826] [gazebo_ros2_control]: connected to service!! robot_state_publisher
[gzserver-1] [INFO] [1724104227.650765287] [gazebo_ros2_control]: Received urdf from param server, parsing...
[ERROR] [gzserver-1]: process has died [pid 72718, exit code -11, cmd 'gzserver -slibgazebo_ros_init.so -slibgazebo_ros_factory.so -slibgazebo_ros_force_system.so'].
[INFO] [spawn_entity.py-4]: process has finished cleanly [pid 72721]
[INFO] [ros2-5]: process started with pid [72739]
[ros2-5] Could not contact service /controller_manager/load_controller
[ERROR] [ros2-5]: process has died [pid 72739, exit code 1, cmd 'ros2 control load_controller --set-state active joint_state_broadcaster'].
[INFO] [ros2-6]: process started with pid [72776]
[ros2-6] Could not contact service /controller_manager/load_controller
[ERROR] [ros2-6]: process has died [pid 72776, exit code 1, cmd 'ros2 control load_controller --set-state active joint_trajectory_controller'].

```

alexanderkjones, 2024-08-20:
@traversaro last issue here I hope.. I have gazebo-ros2-control-demos working out of the box with the fixes we've identified here. However, I'm not getting any of the demo control scripts running. Controller manager loads, all urdf and launch files are running appropriateI, and I can command trajectories through the terminal. However, I cannot get the following basic demo scripts to work. Not sure if you've seen this before as with the other issues:

[Cart on Rail Demo](https://control.ros.org/humble/doc/gazebo_ros2_control/doc/index.html#cart-on-rail)
```
(ros_humble) (base) alexanderjones@Alexanders-MacBook-Air dev_ws % ros2 run gazebo_ros2_control_demos example_position

dyld[99783]: symbol not found in flat namespace '_PyExc_RuntimeError'
[ros2run]: Abort trap: 6
(ros_humble) (base) alexanderjones@Alexanders-MacBook-Air dev_ws % ros2 run gazebo_ros2_control_demos example_velocity

dyld[99820]: symbol not found in flat namespace '_PyExc_RuntimeError'
[ros2run]: Abort trap: 6
```

cmoyates, 2024-08-26:
@alexanderkjones Did you manage to get this working after? I've been dealing with basically the same issue for about a week and a half and it seems like you're closer than I managed to get

alexanderkjones, 2024-08-26:
@cmoyates are you referring to the `_PyExc_RuntimeError` issue? I have not had success there. I have only gotten this error when running the gazebo_ros2_control_demos example_position/velocy/etc nodes. The launch files from ros2_control_demos have worked fine and I have been able to send trajectory commands successfully. 

For example: 

I can successfully launch `ros2 launch gazebo_ros2_control_demos cart_example_position.launch.py`

However, I get the `_PyExc_RuntimeError` when attempting to run `ros2 run gazebo_ros2_control_demos example_position
`

That said, the demo seems to be working because I can send trajectory commands directly to the example as such:

```
ros2 action send_goal /joint_trajectory_controller/follow_joint_trajectory control_msgs/action/FollowJointTrajectory "{trajectory: {joint_names: ['slider_to_cart'], points: [{positions: [1.0], time_from_start: {sec: 1, nanosec: 0}}]}}"
```

```
ros2 action send_goal /joint_trajectory_controller/follow_joint_trajectory control_msgs/action/FollowJointTrajectory "{trajectory: {joint_names: ['slider_to_cart'], points: [{positions: [1.0], velocities: [0.5], accelerations: [0.1], time_from_start: {sec: 1, nanosec: 0}}, {positions: [0.5], velocities: [0.2], accelerations: [0.05], time_from_start: {sec: 2, nanosec: 0}}, {positions: [1.5], velocities: [0.3], accelerations: [0.1], time_from_start: {sec: 3, nanosec: 0}}, {positions: [0.0], velocities: [0.1], accelerations: [0.05], time_from_start: {sec: 5, nanosec: 0}}]}}"
```

My next step has been to write my own control node in place of of the example_position/velocity as I believe there's simply some issue with the build parameters of the demos files. They seem simple to replicate just sending the above commands to the appropriate actions servers. However I have not had time to implement to see if I still get the `_PyExc_RuntimeError`.

@traversaro would it make sense to close this issue regarding the gazebo-ros2-control and ros2-control issues above and, if so, what is the policy for opening an issue around the `_PyExc_RuntimeError` which is actually ros2-control-demos package issue and as I understand it not considered a supported package under Robostack's "Available Packages"

traversaro, 2025-01-11:
> @traversaro would it make sense to close this issue regarding the gazebo-ros2-control and ros2-control issues above and, if so, what is the policy for opening an issue around the [...]

Sorry, I had missed this message. Feel free to open any issue about any package, there is no strict distinction between "available packages" and "not available packages". The only difference is the effort put in contributor to make it work. Anyhow, we recently completed a full rebuilt of all packages (you can be sure to install new packages by installing `ros2-distro-mutex==0.6.*`), if you are still interested can you check if the problem is still present in new packages? Thanks!

traversaro, 2025-04-10:
> Anyhow, we recently completed a full rebuilt of all packages (you can be sure to install new packages by installing `ros2-distro-mutex==0.6.*`), if you are still interested can you check if the problem is still present in new packages? Thanks!

Given the lack of feedback I would close the issue. @alexanderkjones @cmoyates if you still experience this in new packages, feel free to open a new issue, thanks!

closed 2025-04-10 by traversaro