RoboStack/ros-humble #199: Ament tools no longer in bin directory, and therefore not on path

tony-p, 2024-08-16:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

A number of the ament tools are no longer installed to the bin directory (since the py3.11 rebuild) and as a consequence are not on the path causing builds requiring them to fail. Adding them to the path solves the issue.

```sh
#!/bin/bash

export PATH=$PATH:$CONDA_PREFIX/lib/ament_lint_cmake/:$CONDA_PREFIX/lib/ament_xmllint/:$CONDA_PREFIX/lib/ament_cpplint/
```

### Installed packages

```shell
(Truncated to ros only due to char limit)

ros-humble-action-msgs                            1.2.1         py311hb335429_6        101.6 KiB  conda  ros-humble-action-msgs-1.2.1-py311hb335429_6.tar.bz2
ros-humble-actionlib-msgs                          4.2.3         py311hb335429_6        87.2 KiB   conda  ros-humble-actionlib-msgs-4.2.3-py311hb335429_6.tar.bz2
ros-humble-ament-clang-format                     0.12.10       py311hb335429_6        25.6 KiB   conda  ros-humble-ament-clang-format-0.12.10-py311hb335429_6.tar.bz2
ros-humble-ament-cmake                            1.3.7         py311hb335429_6        14.9 KiB   conda  ros-humble-ament-cmake-1.3.7-py311hb335429_6.tar.bz2
ros-humble-ament-cmake-auto                        1.3.7         py311hb335429_6        14.8 KiB   conda  ros-humble-ament-cmake-auto-1.3.7-py311hb335429_6.tar.bz2
ros-humble-ament-cmake-catch2                     1.2.1         py311hb335429_6        15.4 KiB   conda  ros-humble-ament-cmake-catch2-1.2.1-py311hb335429_6.tar.bz2
ros-humble-ament-cmake-copyright                   0.12.10       py311hb335429_6        11.7 KiB   conda  ros-humble-ament-cmake-copyright-0.12.10-py311hb335429_6.tar.bz2
ros-humble-ament-cmake-core                        1.3.7         py311hb335429_6        33.2 KiB   conda  ros-humble-ament-cmake-core-1.3.7-py311hb335429_6.tar.bz2
ros-humble-ament-cmake-cppcheck                    0.12.10       py311hb335429_6        12.7 KiB   conda  ros-humble-ament-cmake-cppcheck-0.12.10-py311hb335429_6.tar.bz2
ros-humble-ament-cmake-cpplint                     0.12.10       py311hb335429_6        12 KiB     conda  ros-humble-ament-cmake-cpplint-0.12.10-py311hb335429_6.tar.bz2
ros-humble-ament-cmake-export-definitions          1.3.7         py311hb335429_6        11.5 KiB   conda  ros-humble-ament-cmake-export-definitions-1.3.7-py311hb335429_6.tar.bz2
ros-humble-ament-cmake-export-dependencies         1.3.7         py311hb335429_6        12.3 KiB   conda  ros-humble-ament-cmake-export-dependencies-1.3.7-py311hb335429_6.tar.bz2
ros-humble-ament-cmake-export-include-directories  1.3.7         py311hb335429_6        11.8 KiB   conda  ros-humble-ament-cmake-export-include-directories-1.3.7-py311hb335429_6.tar.bz2
ros-humble-ament-cmake-export-interfaces           1.3.7         py311hb335429_6        12 KiB     conda  ros-humble-ament-cmake-export-interfaces-1.3.7-py311hb335429_6.tar.bz2
ros-humble-ament-cmake-export-libraries            1.3.7         py311hb335429_6        13.4 KiB   conda  ros-humble-ament-cmake-export-libraries-1.3.7-py311hb335429_6.tar.bz2
ros-humble-ament-cmake-export-link-flags           1.3.7         py311hb335429_6        11.4 KiB   conda  ros-humble-ament-cmake-export-link-flags-1.3.7-py311hb335429_6.tar.bz2
ros-humble-ament-cmake-export-targets              1.3.7         py311hb335429_6        11.9 KiB   conda  ros-humble-ament-cmake-export-targets-1.3.7-py311hb335429_6.tar.bz2
ros-humble-ament-cmake-flake8                     0.12.10       py311hb335429_6        11.9 KiB   conda  ros-humble-ament-cmake-flake8-0.12.10-py311hb335429_6.tar.bz2
ros-humble-ament-cmake-gen-version-h               1.3.7         py311hb335429_6        13.8 KiB   conda  ros-humble-ament-cmake-gen-version-h-1.3.7-py311hb335429_6.tar.bz2
ros-humble-ament-cmake-gmock                       1.3.7         py311hb335429_6        13.2 KiB   conda  ros-humble-ament-cmake-gmock-1.3.7-py311hb335429_6.tar.bz2
ros-humble-ament-cmake-gtest                      1.3.7         py311hb335429_6        17.4 KiB   conda  ros-humble-ament-cmake-gtest-1.3.7-py311hb335429_6.tar.bz2
ros-humble-ament-cmake-include-directories         1.3.7         py311hb335429_6        11.4 KiB   conda  ros-humble-ament-cmake-include-directories-1.3.7-py311hb335429_6.tar.bz2
ros-humble-ament-cmake-libraries                   1.3.7         py311hb335429_6        11.8 KiB   conda  ros-humble-ament-cmake-libraries-1.3.7-py311hb335429_6.tar.bz2
ros-humble-ament-cmake-lint-cmake                 0.12.10       py311hb335429_6        15.1 KiB   conda  ros-humble-ament-cmake-lint-cmake-0.12.10-py311hb335429_6.tar.bz2
ros-humble-ament-cmake-pep257                      0.12.10       py311hb335429_6        11.6 KiB   conda  ros-humble-ament-cmake-pep257-0.12.10-py311hb335429_6.tar.bz2
ros-humble-ament-cmake-pytest                      1.3.7         py311hb335429_6        22.3 KiB   conda  ros-humble-ament-cmake-pytest-1.3.7-py311hb335429_6.tar.bz2
ros-humble-ament-cmake-python                     1.3.7         py311hb335429_6        13.7 KiB   conda  ros-humble-ament-cmake-python-1.3.7-py311hb335429_6.tar.bz2
ros-humble-ament-cmake-ros                        0.10.0        py311hb335429_6        14.4 KiB   conda  ros-humble-ament-cmake-ros-0.10.0-py311hb335429_6.tar.bz2
ros-humble-ament-cmake-target-dependencies         1.3.7         py311hb335429_6        12.9 KiB   conda  ros-humble-ament-cmake-target-dependencies-1.3.7-py311hb335429_6.tar.bz2
ros-humble-ament-cmake-test                        1.3.7         py311hb335429_6        30.3 KiB   conda  ros-humble-ament-cmake-test-1.3.7-py311hb335429_6.tar.bz2
ros-humble-ament-cmake-uncrustify                  0.12.10       py311hb335429_6        12.1 KiB   conda  ros-humble-ament-cmake-uncrustify-0.12.10-py311hb335429_6.tar.bz2
ros-humble-ament-cmake-version                     1.3.7         py311hb335429_6        11.2 KiB   conda  ros-humble-ament-cmake-version-1.3.7-py311hb335429_6.tar.bz2
ros-humble-ament-cmake-xmllint                    0.12.10       py311hb335429_6        11.7 KiB   conda  ros-humble-ament-cmake-xmllint-0.12.10-py311hb335429_6.tar.bz2
ros-humble-ament-copyright                         0.12.10       py311hb335429_6        65.4 KiB   conda  ros-humble-ament-copyright-0.12.10-py311hb335429_6.tar.bz2
ros-humble-ament-cppcheck                          0.12.10       py311hb335429_6        23.9 KiB   conda  ros-humble-ament-cppcheck-0.12.10-py311hb335429_6.tar.bz2
ros-humble-ament-cpplint                           0.12.10       py311hb335429_6        195.8 KiB  conda  ros-humble-ament-cpplint-0.12.10-py311hb335429_6.tar.bz2
ros-humble-ament-flake8                            0.12.10       py311hb335429_6        25 KiB     conda  ros-humble-ament-flake8-0.12.10-py311hb335429_6.tar.bz2
ros-humble-ament-index-cpp                        1.4.0         py311hb335429_6        30.7 KiB   conda  ros-humble-ament-index-cpp-1.4.0-py311hb335429_6.tar.bz2
ros-humble-ament-index-python                      1.4.0         py311hb335429_6        22.5 KiB   conda  ros-humble-ament-index-python-1.4.0-py311hb335429_6.tar.bz2
ros-humble-ament-lint                              0.12.10       py311hb335429_6        9.1 KiB    conda  ros-humble-ament-lint-0.12.10-py311hb335429_6.tar.bz2
ros-humble-ament-lint-auto                        0.12.10       py311hb335429_6        14.9 KiB   conda  ros-humble-ament-lint-auto-0.12.10-py311hb335429_6.tar.bz2
ros-humble-ament-lint-cmake                        0.12.10       py311hb335429_6        38 KiB     conda  ros-humble-ament-lint-cmake-0.12.10-py311hb335429_6.tar.bz2
ros-humble-ament-lint-common                       0.12.10       py311hb335429_6        11.6 KiB   conda  ros-humble-ament-lint-common-0.12.10-py311hb335429_6.tar.bz2
ros-humble-ament-package                           0.14.0        py311hb335429_6        45.1 KiB   conda  ros-humble-ament-package-0.14.0-py311hb335429_6.tar.bz2
ros-humble-ament-pep257                            0.12.10       py311hb335429_6        20.7 KiB   conda  ros-humble-ament-pep257-0.12.10-py311hb335429_6.tar.bz2
ros-humble-ament-uncrustify                        0.12.10       py311hb335429_6        44.9 KiB   conda  ros-humble-ament-uncrustify-0.12.10-py311hb335429_6.tar.bz2
ros-humble-ament-xmllint                           0.12.10       py311hb335429_6        21.8 KiB   conda  ros-humble-ament-xmllint-0.12.10-py311hb335429_6.tar.bz2
[... cut ...] e, pkg-config, make, ninja, colcon-common-extensions, catkin_tools, vcstool, catch2, pytest, setuptools, flake8, flake8-docstrings, flake8-isort, flake8-black, clang-format, clangdev, pre-commit, ros-humble-generate-parameter-library, zeromq, gtest, gmock, sqlite, asyncua, open62541, nlohmann_json, ros-humble-rosbridge-suite, curl, requests, transformations, ros-humble-tf2, python-orocos-kdl, fastapi, uvicorn, gxx, gdb, libxcrypt, binutils, ros-humble-rqt
  Target platforms: win-64, linux-64
```

Tobias-Fischer, 2024-12-22:
Hi @tony-p - did you find a solution to this that we could incorporate into #229?

tony-p, 2024-12-22:
> Hi @tony-p - did you find a solution to this that we could incorporate into #229?

As I use pixi I just have an activation script that adds the deeper paths to the path. I’ve been so busy I haven’t been able to look further than that

Tobias-Fischer, 2024-12-22:
Would you be willing to share the pixi activation script? I think it’ll make sense to “enforce” pixi for RoboStack with the rebuild we’re currently doing; any helper scripts etc would be very welcome!

tony-p, 2024-12-22:
there might be more I’ve missed but this what I’ve encountered so far

```sh

#!/bin/bash

export PATH=$PATH:$CONDA_PREFIX/lib/ament_lint_cmake/:$CONDA_PREFIX/lib/ament_xmllint/:$CONDA_PREFIX/lib/ament_cpplint/:$CONDA_PREFIX/lib/ament_copyright/
```

traversaro, 2024-12-22:
At a first glance the latest rebuild in https://github.com/RoboStack/ros-humble/pull/229 the packages are not affected by this problem, see:

~~~
2024-12-22T05:31:18.9342433Z  │ │ Files in package:
2024-12-22T05:31:18.9342969Z  │ │   - bin/ament_lint_cmake
2024-12-22T05:31:18.9343479Z  │ │   - lib/python3.11/site-packages/ament_lint_cmake/__init__.py
2024-12-22T05:31:18.9344166Z  │ │   - lib/python3.11/site-packages/ament_lint_cmake/__pycache__/__init__.cpython-311.pyc
2024-12-22T05:31:18.9344923Z  │ │   - lib/python3.11/site-packages/ament_lint_cmake/__pycache__/cmakelint.cpython-311.pyc
2024-12-22T05:31:18.9345693Z  │ │   - lib/python3.11/site-packages/ament_lint_cmake/__pycache__/main.cpython-311.pyc
2024-12-22T05:31:18.9346329Z  │ │   - lib/python3.11/site-packages/ament_lint_cmake/cmakelint.py
2024-12-22T05:31:18.9346855Z  │ │   - lib/python3.11/site-packages/ament_lint_cmake/main.py
2024-12-22T05:31:18.9347449Z  │ │   - lib/python3.11/site-packages/ament_lint_cmake-0.12.11.dist-info/INSTALLER
2024-12-22T05:31:18.9348097Z  │ │   - lib/python3.11/site-packages/ament_lint_cmake-0.12.11.dist-info/METADATA
2024-12-22T05:31:18.9348698Z  │ │   - lib/python3.11/site-packages/ament_lint_cmake-0.12.11.dist-info/RECORD
2024-12-22T05:31:18.9349282Z  │ │   - lib/python3.11/site-packages/ament_lint_cmake-0.12.11.dist-info/REQUESTED
2024-12-22T05:31:18.9349902Z  │ │   - lib/python3.11/site-packages/ament_lint_cmake-0.12.11.dist-info/WHEEL
2024-12-22T05:31:18.9350912Z  │ │   - lib/python3.11/site-packages/ament_lint_cmake-0.12.11.dist-info/direct_url.json
2024-12-22T05:31:18.9351649Z  │ │   - lib/python3.11/site-packages/ament_lint_cmake-0.12.11.dist-info/entry_points.txt
2024-12-22T05:31:18.9352666Z  │ │   - lib/python3.11/site-packages/ament_lint_cmake-0.12.11.dist-info/top_level.txt
2024-12-22T05:31:18.9353313Z  │ │   - lib/python3.11/site-packages/ament_lint_cmake-0.12.11.dist-info/zip-safe
2024-12-22T05:31:18.9354047Z  │ │   - share/ament_index/resource_index/packages/ament_lint_cmake
2024-12-22T05:31:18.9354525Z  │ │   - share/ament_lint_cmake/package.xml
2024-12-22T05:31:18.9354841Z  │ │   - info/about.json
2024-12-22T05:31:18.9355115Z  │ │   - info/hash_input.json
2024-12-22T05:31:18.9355437Z  │ │   - info/index.json
2024-12-22T05:31:18.9355658Z  │ │   - info/paths.json
2024-12-22T05:31:18.9355976Z  │ │   - info/recipe/bld_ament_python.bat
2024-12-22T05:31:18.9356350Z  │ │   - info/recipe/build_ament_python.sh
2024-12-22T05:31:18.9356637Z  │ │   - info/recipe/recipe.yaml
2024-12-22T05:31:18.9356900Z  │ │   - info/recipe/rendered_recipe.yaml
2024-12-22T05:31:18.9357216Z  │ │   - info/recipe/variant_config.yaml
2024-12-22T05:31:18.9357568Z  │ │   - info/tests/tests.yaml
2024-12-22T05:31:18.9357998Z  │ │ Creating target folder '/home/runner/work/ros-humble/ros-humble/output/linux-64'
2024-12-22T05:31:18.9358396Z  │ │ Compressing archive...
2024-12-22T05:31:18.9565404Z  │ │ Archive written to '/home/runner/work/ros-humble/ros-humble/output/linux-64/ros-humble-ament-lint-cmake-0.12.11-np126py311h15c07e3_7.conda'
~~~

However, to make sure we do not have regressions in this, I also added some smoke tests to check if the executables you mentioned in this issue can be correctly found in https://github.com/RoboStack/ros-humble/pull/229/commits/1511e1225b465db6a2fbce81e2acad360d998133 , so hopefully we should not have further regressions related to this.

Tobias-Fischer, 2025-01-06:
Fixed in #229

closed 2025-01-06 by Tobias-Fischer