RoboStack/ros-humble #203: Package request: PACKAGE_NAME: ros2_control, ros2_controllers --version bump--

sonelu, 2024-08-22:
### Package name

ros2_control, ros2_controllers

### Operating System(s)

- [X] linux-64
- [X] linux-aarch64
- [X] osx-64
- [X] osx-arm64
- [X] win-64

### Additional context (optional)

How can I bump the version for these packages:
- ros2_control 2. 38 > 4.16
- ros2_controllers 2.32 > 4.13

They are already on the vinca list and I don't know how to trigger the rebuild.

Tobias-Fischer, 2024-08-22:
You can locally increase the build number which will trigger a rebuild. We currently don’t have the capacity to rebuild all packages in CI; I’ll try to do one in two months once the teaching semester is over.

sonelu, 2024-08-22:
An I have to comment all other packages out so that they don't get rebuilt?

I assume if I change the build number as it is now, all the packages that are uncommented on the list will be rebuilt?

sonelu, 2024-08-27:
We'll be doing the full rebuilt as discussed.

closed 2024-08-27 by sonelu