RoboStack/ros-humble #215: ros_env uses home-brew python and numpy fails

cemoke, 2024-10-09:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

I get the following error when I try to import numpy in mamba ros_env using ros-humble

Python 3.11.8 | packaged by conda-forge | (main, Feb 16 2024, 20:49:36) [Clang 16.0.6 ] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import numpy
Traceback (most recent call last):
  File "/opt/homebrew/lib/python3.12/site-packages/numpy/_core/__init__.py", line 23, in <module>
    from . import multiarray
  File "/opt/homebrew/lib/python3.12/site-packages/numpy/_core/multiarray.py", line 10, in <module>
    from . import overrides
  File "/opt/homebrew/lib/python3.12/site-packages/numpy/_core/overrides.py", line 8, in <module>
    from numpy._core._multiarray_umath import (
ModuleNotFoundError: No module named 'numpy._core._multiarray_umath'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/opt/homebrew/lib/python3.12/site-packages/numpy/__init__.py", line 114, in <module>
    from numpy.__config__ import show as show_config
  File "/opt/homebrew/lib/python3.12/site-packages/numpy/__config__.py", line 4, in <module>
    from numpy._core._multiarray_umath import (
  File "/opt/homebrew/lib/python3.12/site-packages/numpy/_core/__init__.py", line 49, in <module>
    raise ImportError(msg)
ImportError:

IMPORTANT: PLEASE READ THIS FOR ADVICE ON HOW TO SOLVE THIS ISSUE!

Importing the numpy C-extensions failed. This error can happen for
many reasons, often due to issues with your setup or how NumPy was
installed.

We have compiled some common reasons and troubleshooting tips at:

    https://numpy.org/devdocs/user/troubleshooting-importerror.html

Please note and check the following:

  * The Python version is: Python3.11 from "/Users/cembilaloglu/miniforge3/envs/ros_env/bin/python"
  * The NumPy version is: "2.1.1"

and make sure that they are the versions you expect.
Please carefully study the documentation linked above for further help.

Original error was: No module named 'numpy._core._multiarray_umath'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/opt/homebrew/lib/python3.12/site-packages/numpy/__init__.py", line 119, in <module>
    raise ImportError(msg) from e
ImportError: Error importing numpy: you should not try to import numpy from
        its source directory; please exit the numpy source tree, and relaunch
        your python interpreter from there.
        
for some reason, when I use mamba to activate ros_env, it adds homebrew's python to my PYTHONPATH, and then different versions of numpy result in this issue. Can you please help me to resolve the issue?

### Installed packages

```shell
# packages in environment at /Users/cembilaloglu/miniforge3/envs/ros_env:
#
# Name                    Version                   Build  Channel
aiohappyeyeballs          2.4.3              pyhd8ed1ab_0    conda-forge
[... 647 lines cut ...]
          conda version : 24.9.1
    conda-build version : not installed
         python version : 3.12.6.final.0
                 solver : libmamba (default)
       virtual packages : __archspec=1=m1
                          __conda=24.9.1=0
                          __osx=14.6.1=0
                          __unix=0=0
       base environment : /Users/cembilaloglu/miniforge3  (writable)
      conda av data dir : /Users/cembilaloglu/miniforge3/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack-staging/osx-arm64
                          https://conda.anaconda.org/robostack-staging/noarch
                          https://conda.anaconda.org/conda-forge/osx-arm64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : /Users/cembilaloglu/miniforge3/pkgs
                          /Users/cembilaloglu/.conda/pkgs
       envs directories : /Users/cembilaloglu/miniforge3/envs
                          /Users/cembilaloglu/.conda/envs
               platform : osx-arm64
             user-agent : conda/24.9.1 requests/2.32.3 CPython/3.12.6 Darwin/23.6.0 OSX/14.6.1 solver/libmamba conda-libmamba-solver/24.7.0 libmambapy/1.5.10
                UID:GID : 501:20
             netrc file : None
           offline mode : False
```

traversaro, 2024-10-09:
Are you sure that is the `mamba activate` that adds homebrew's site-packages to your `PYTHONPATH`? Can you report the ouput of the `env` command before and after the activation (make sure that you remove any confidential env variables such as token or secres).

cemoke, 2024-10-11:
You are right. After more investigation, I realized the problem was not related to the robostack environment but to my setup. So, I am closing the issue.

closed 2024-10-11 by cemoke