RoboStack/ros-humble #216: urdf/model.h out of date for ROS2 Humble?

jpmartin42, 2024-10-12:
System info: ROS2 Humble / Ubuntu 22.04; Mamba env using Python 3.10

I was getting some odd errors in C++ when referencing the `urdf/model.h` library. This was in ros-agnostic libraries that stored URDF's using this library, and had worked fine when I had written the code in ROS1 and when I had used apt-installed ROS2 instead of the RoboStack equivalent.

Long story short, it looks like the `urdf/model.h` version downloaded via the [the install guide](https://robostack.github.io/GettingStarted.html) is the slightly outdated [Galactic version](https://github.com/ros2/urdf/blob/galactic/urdf/include/urdf/model.h) and not the [Humble version](https://github.com/ros2/urdf/blob/humble/urdf/include/urdf/model.h). They're pretty much identical, but the Humble version of the library contains a few extra functions:

```
  URDF_EXPORT Model(const Model & other);
  URDF_EXPORT Model & operator=(const Model & other);
  URDF_EXPORT Model(Model && other) noexcept;
  URDF_EXPORT Model & operator=(Model && other)noexcept;
```

... that my code (and, presumably, other C++ code that referenced the library) needed in a few areas. Adding these few functions to the `urdf/model.h` file allowed my code to build, though I have yet to finish setting it up to run to see if that quick fix did what I needed it to. I've never installed Galactic or any non-Humble version of ROS on this PC in Mamba or via apt.

I doubt it's an issue for most folks, but just wanted to throw this up here in case it became relevant to someone else!

Tobias-Fischer, 2025-01-06:
Should be fixed in the recent rebuild. If not, please feel free to reopen.

closed 2025-01-06 by Tobias-Fischer

traversaro, 2025-01-06:
Indeed, the rebuild in https://github.com/RoboStack/ros-humble/pull/229 contains urdf 2.6.1 that should contain the PR https://github.com/ros2/urdf/pull/41 that contains the backport of the fix to humble.