RoboStack/ros-humble #219: ign gazebo empty.sdf fails to load and find library

mhl787156, 2024-10-29:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

Hi, I've been trying to set up a gazebo environment on my machine (X86 Windows + WSL2 with micromamba), but I'm having issues getting (ign) gazebo fortress to load properly. 

My installation environment is simply:
```yaml
name: gazebo_test
channels:
- conda-forge
- robostack-staging
dependencies:
- ros-humble-desktop
- ros-humble-ros-gz 
```

After activating `gazebo_test`, I run `ign gazebo empty.sdf` which should load an empty world, but I get the following:

```bash
$ ign gazebo empty.sdf
Library error: Problem running [gazebo]() from /home/mickey/micromamba/envs/gazebo_test/lib/libignition-gazebo6-ign.so.6.12.0.
```

Investigating more, it seems incapable of finding `empty.sdf`. Even if I give it the full path to `empty.sdf`

```bash
$ ign gazebo /home/mickey/micromamba/envs/gazebo_test/share/ignition/ignition-gazebo6/worlds/empty.sdf 
Library error: Problem running [gazebo]() from /home/mickey/micromamba/envs/gazebo_test/lib/libignition-gazebo6-ign.so.6.12.0.
```

**However! If I just run `ign gazebo`, the gazebo gui pops up and I can manually start empty.sdf** 

I thought it might be a CLI problem, I tried running `ign gazebo --verbose empty.sdf` but still same error with no more verbosensess. 

Do you know what might be going on? I had a look through other issues and found no leads. 
I'm not sure if repoducable as I do not have another machine to test on. 

Let me know if you want any other information. 

### Installed packages

```shell
See first response, exceeded character limit or something
```

### Environment info

```shell
libmamba version : 2.0.0
     micromamba version : 2.0.0
           curl version : libcurl/8.9.1 OpenSSL/3.3.2 zlib/1.3.1 zstd/1.5.6 libssh2/1.11.0 nghttp2/1.58.0
     libarchive version : libarchive 3.7.4 zlib/1.2.13 bz2lib/1.0.8 libzstd/1.5.6
       envs directories : /home/mickey/micromamba/envs
          package cache : /home/mickey/micromamba/pkgs
                          /home/mickey/.mamba/pkgs
            environment : gazebo_test (active)
           env location : /home/mickey/micromamba/envs/gazebo_test
      user config files : /home/mickey/.mambarc
 populated config files : /home/mickey/.condarc
       virtual packages : __unix=0=0
                          __linux=5.15.153=0
                          __glibc=2.35=0
                          __archspec=1=x86_64_v3
                          __cuda=12.5=0
               channels : https://conda.anaconda.org/conda-forge/linux-64
                          https://conda.anaconda.org/conda-forge/noarch
                          https://conda.anaconda.org/nodefaults/linux-64
                          https://conda.anaconda.org/nodefaults/noarch
       base environment : /home/mickey/micromamba
               platform : linux-64
```

mhl787156, 2024-10-29:
```
List of packages in environment: "/home/mickey/micromamba/envs/gazebo_test"

  Name                                               Version       Build                     Channel                                                                                                    
──────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
  _libgcc_mutex                                      0.1           conda_forge               conda-forge/linux-64/_libgcc_mutex-0.1-conda_forge.tar.bz2                                                 
  _openmp_mutex                                      4.5           2_gnu                     conda-forge/linux-64/_openmp_mutex-4.5-2_gnu.tar.bz2                                                       
  aiohappyeyeballs                                   2.4.3         pyhd8ed1ab_0              conda-forge/noarch/aiohappyeyeballs-2.4.3-pyhd8ed1ab_0.conda                                               
  aiohttp                                            3.10.10       py311h2dc5d0c_0           conda-forge/linux-64/aiohttp-3.10.10-py311h2dc5d0c_0.conda                                                 
  aiosignal                                          1.3.1         pyhd8ed1ab_0              conda-forge/noarch/aiosignal-1.3.1-pyhd8ed1ab_0.tar.bz2                                                    
  alsa-lib                                           1.2.12        h4ab18f5_0                conda-forge/linux-64/alsa-lib-1.2.12-h4ab18f5_0.conda                                                      
  aom                                                3.9.1         hac33072_0                conda-forge/linux-64/aom-3.9.1-hac33072_0.conda                                                            
  argcomplete                                        3.5.1         pyhd8ed1ab_0              conda-forge/noarch/argcomplete-3.5.1-pyhd8ed1ab_0.conda                                                    
  assimp                                             5.3.1         hfb0e8fe_2                conda-forge/linux-64/assimp-5.3.1-hfb0e8fe_2.conda                                                         
  atk-1.0                                            2.38.0        h04ea711_2                conda-forge/linux-64/atk-1.0-2.38.0-h04ea711_2.conda                                                       
  attr                                               2.5.1         h166bdaf_1                conda-forge/linux-64/attr-2.5.1-h166bdaf_1.tar.bz2                                                         
  attrs                                              24.2.0        pyh71513ae_0              conda-forge/noarch/attrs-24.2.0-pyh71513ae_0.conda                                                         
  blosc                                              1.21.5        hc2324a3_1                conda-forge/linux-64/blosc-1.21.5-hc2324a3_1.conda                                                         
  boost                                              1.82.0        h781c19f_6                conda-forge/linux-64/boost-1.82.0-h781c19f_6.conda                                                         
  brotli                                             1.1.0         hb9d3cd8_2                conda-forge/linux-64/brotli-1.1.0-hb9d3cd8_2.conda                                                         
  brotli-bin                                         1.1.0         hb9d3cd8_2                conda-forge/linux-64/brotli-bin-1.1.0-hb9d3cd8_2.conda                                                     
  bullet                                             3.25          h781c19f_3                conda-forge/linux-64/bullet-3.25-h781c19f_3.conda                                                          
  bullet-cpp                                         3.25          h6dcdc2f_3                conda-forge/linux-64/bullet-cpp-3.25-h6dcdc2f_3.conda                                                      
  bzip2                                              1.0.8         h4bc722e_7                conda-forge/linux-64/bzip2-1.0.8-h4bc722e_7.conda                                                          
  c-ares                                             1.34.2        heb4867d_0                conda-forge/linux-64/c-ares-1.34.2-heb4867d_0.conda                                                        
  ca-certificates                                    2024.8.30     hbcca054_0                conda-forge/linux-64/ca-certificates-2024.8.30-hbcca054_0.conda                                            
  cairo                                              1.18.0        h3faef2a_0                conda-forge/linux-64/cairo-1.18.0-h3faef2a_0.conda                                                         
  catkin_pkg                                         1.0.0         pyhd8ed1ab_0              conda-forge/noarch/catkin_pkg-1.0.0-pyhd8ed1ab_0.conda                                                     
  certifi                                            2024.8.30     pyhd8ed1ab_0              conda-forge/noarch/certifi-2024.8.30-pyhd8ed1ab_0.conda                                                    
[... cut ...]
  zlib                                               1.2.13        h4ab18f5_6                conda-forge/linux-64/zlib-1.2.13-h4ab18f5_6.conda                                                          
  zstd                                               1.5.6         ha6fb4c9_0                conda-forge/linux-64/zstd-1.5.6-ha6fb4c9_0.conda                                                           
  zziplib                                            0.13.69       h27826a3_1                conda-forge/linux-64/zziplib-0.13.69-h27826a3_1.tar.bz2                                                    
```

Tobias-Fischer, 2025-01-09:
Hi @mhl787156 - could you please try whether this works with the new packages (`ros2-distro-mutex=0.6`)?

Tobias-Fischer, 2025-01-09:
I did a bit of digging - I was able to reproduce locally and found we need to backport https://github.com/gazebosim/gz-sim/pull/1884 to cmdgazebo6.rb in https://github.com/conda-forge/libignition-gazebo-feedstock

@mhl787156 - are you happy to open a pull request?

Tobias-Fischer, 2025-01-09:
/cc @traversaro

traversaro, 2025-01-09:
Unfortunately there was already a protobuf migration libignition-gazebo, so beside backporting the patch we also need to do somehow at least a build with protobuf 5.28.2 .

traversaro, 2025-01-09:
> I did a bit of digging - I was able to reproduce locally and found we need to backport [gazebosim/gz-sim#1884](https://github.com/gazebosim/gz-sim/pull/1884) to cmdgazebo6.rb in https://github.com/conda-forge/libignition-gazebo-feedstock
> 
> @mhl787156 - are you happy to open a pull request?

Actually we do not need to backport it, we just need to release a new version of ignition-gazebo6 .

mhl787156, 2025-01-15:
Hi @Tobias-Fischer and @traversaro thanks for looking into this. If I understand this right, not further actions are needed and just need to wait for the next release of robostack ignition-gazebo6? 

If thats right, is there an eta for release? 

We're teaching students this term and hoping this might solve our MacOSX woes...

traversaro, 2025-01-15:
Sorry, the release of libignition-gazebo required a release of libignition-gui, that required a release of libignition-rendering, and so on so forth. The actions necessary are doing these releases, hopefully we are reaching the end of it.

traversaro, 2025-01-15:
> We're teaching students this term and hoping this might solve our MacOSX woes...

Anyhow Ignition Fortress (i.e. ignition-gazebo6) had its own problems with macOS, proper macOS support was only added in Gazebo Garden (i.e. gz-sim7), see https://github.com/gazebosim/gz-sim/pull/1225 . Even in new version of Modern Gazebo however, there is unfortunately still no support upstream for `gz sim` command, but you need to run `gz sim -s` and `gz sim -g` separately, see https://github.com/gazebosim/gz-sim/issues/44 . I do not think that is impossible to achieve, but it would still require some work.

Tobias-Fischer, 2025-01-21:
Hi @mhl787156 - we've released the new version of ignition-gazebo6. Could you please take a look whether this resolved your issues? Feel free to re-open if not :).

closed 2025-01-21 by Tobias-Fischer