RoboStack/ros-humble #220: [ERROR] rviz2 error

wangyan-hlab, 2024-10-30:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

I installed ros2 humble according to the installation guide, but it raised the following error when I tried to test the installation with the command "rviz2":

[ERROR] [1730292401.569491564] [rviz2]: InternalErrorException: Could not load dynamic library /home/yan/miniforge3/envs/ros_env/opt/rviz_ogre_vendor/lib/OGRE/RenderSystem_GL.  System Error: /home/yan/miniforge3/envs/ros_env/opt/rviz_ogre_vendor/lib/OGRE/RenderSystem_GL.so.1.12.1: cannot open shared object file: No such file or directory in DynLib::load at ./.obj-x86_64-linux-gnu/ogre-v1.12.1-prefix/src/ogre-v1.12.1/OgreMain/src/OgreDynLib.cpp (line 113)

terminate called after throwing an instance of 'Ogre::InternalErrorException'
  what():  InternalErrorException: Could not load dynamic library /home/yan/miniforge3/envs/ros_env/opt/rviz_ogre_vendor/lib/OGRE/RenderSystem_GL.  System Error: /home/yan/miniforge3/envs/ros_env/opt/rviz_ogre_vendor/lib/OGRE/RenderSystem_GL.so.1.12.1: cannot open shared object file: No such file or directory in DynLib::load at ./.obj-x86_64-linux-gnu/ogre-v1.12.1-prefix/src/ogre-v1.12.1/OgreMain/src/OgreDynLib.cpp (line 113)
Aborted (core dumped)

### Installed packages

```shell
# packages in environment at /home/yan/miniforge3/envs/ros_env:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
aiohappyeyeballs          2.4.3              pyhd8ed1ab_0    conda-forge
aiohttp                   3.10.10         py310h89163eb_0    conda-forge
aiosignal                 1.3.1              pyhd8ed1ab_0    conda-forge
alsa-lib                  1.2.8                h166bdaf_0    conda-forge
aom                       3.5.0                h27087fc_0    conda-forge
argcomplete               3.5.1              pyhd8ed1ab_0    conda-forge
assimp                    5.2.5                hf40c2ba_0    conda-forge
async-timeout             4.0.3              pyhd8ed1ab_0    conda-forge
atk-1.0                   2.38.0               h04ea711_2    conda-forge
attr                      2.5.1                h166bdaf_1    conda-forge
attrs                     24.2.0             pyh71513ae_0    conda-forge
binutils                  2.43                 h4852527_2    conda-forge
binutils_impl_linux-64    2.43                 h4bf12b8_2    conda-forge
binutils_linux-64         2.43                 h4852527_2    conda-forge
boost                     1.78.0          py310hcb52e73_5    conda-forge
boost-cpp                 1.78.0               h5adbc97_2    conda-forge
brotli                    1.1.0                hb9d3cd8_2    conda-forge
brotli-bin                1.1.0                hb9d3cd8_2    conda-forge
bullet                    3.24                 hfdc917e_0    conda-forge
bullet-cpp                3.24                 h769672d_0    conda-forge
bzip2                     1.0.8                h4bc722e_7    conda-forge
c-ares                    1.34.2               heb4867d_0    conda-forge
c-compiler                1.8.0                h2b85faf_1    conda-forge
ca-certificates           2024.8.30            hbcca054_0    conda-forge
cairo                     1.16.0            ha61ee94_1014    conda-forge
catkin_pkg                1.0.0              pyhd8ed1ab_0    conda-forge
catkin_tools              0.9.4              pyhd8ed1ab_0    conda-forge
certifi                   2024.8.30          pyhd8ed1ab_0    conda-forge
cffi                      1.17.1          py310h8deb56e_0    conda-forge
cmake                     3.26.3               h077f3f9_0    conda-forge
colcon-argcomplete        0.3.3              pyhd8ed1ab_0    conda-forge
colcon-bash               0.5.0              pyhd8ed1ab_0    conda-forge
colcon-cd                 0.1.1              pyhd8ed1ab_0    conda-forge
colcon-cmake              0.2.28             pyhd8ed1ab_0    conda-forge
colcon-common-extensions  0.3.0           py310hff52083_2    conda-forge
colcon-core               0.18.2             pyhd8ed1ab_0    conda-forge
colcon-defaults           0.2.8              pyhd8ed1ab_0    conda-forge
colcon-devtools           0.3.0              pyhd8ed1ab_0    conda-forge
colcon-library-path       0.2.1                      py_0    conda-forge
colcon-metadata           0.2.5                      py_0    conda-forge
colcon-output             0.2.13             pyhd8ed1ab_0    conda-forge
colcon-package-information 0.4.0              pyhd8ed1ab_0    conda-forge
colcon-package-selection  0.2.10                     py_0    conda-forge
colcon-parallel-executor  0.3.0              pyhd8ed1ab_0    conda-forge
colcon-pkg-config         0.1.0                      py_0    conda-forge
colcon-powershell         0.4.0              pyhd8ed1ab_0    conda-forge
colcon-python-setup-py    0.2.9              pyhff2d567_0    conda-forge
colcon-recursive-crawl    0.2.3              pyhd8ed1ab_0    conda-forge
[... 625 lines cut ...]
         python version : 3.10.14.final.0
                 solver : libmamba (default)
       virtual packages : __archspec=1=skylake
                          __conda=24.3.0=0
                          __cuda=12.4=0
                          __glibc=2.35=0
                          __linux=6.8.0=0
                          __unix=0=0
       base environment : /home/yan/miniforge3  (writable)
      conda av data dir : /home/yan/miniforge3/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack-staging/linux-64
                          https://conda.anaconda.org/robostack-staging/noarch
                          https://conda.anaconda.org/conda-forge/linux-64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : /home/yan/miniforge3/pkgs
                          /home/yan/.conda/pkgs
       envs directories : /home/yan/miniforge3/envs
                          /home/yan/.conda/envs
               platform : linux-64
             user-agent : conda/24.3.0 requests/2.31.0 CPython/3.10.14 Linux/6.8.0-47-generic ubuntu/22.04.5 glibc/2.35 solver/libmamba conda-libmamba-solver/24.1.0 libmambapy/1.5.8
                UID:GID : 1000:1000
             netrc file : /home/yan/.netrc
           offline mode : False
```

traversaro, 2024-10-30:
Can you post the output of `env` before and after you acivate the environment? Thanks!

wangyan-hlab, 2024-11-01:
> Can you post the output of `env` before and after you acivate the environment? Thanks!

Thank you for your reply. Here are the outputs: 
**Before:** `SHELL=/bin/bash ROS_VERSION=2 SESSION_MANAGER=local/ubuntu:@/tmp/.ICE-unix/2456,unix/ubuntu:/tmp/.ICE-unix/2456 QT_ACCESSIBILITY=1 COLORTERM=truecolor XDG_CONFIG_DIRS=/etc/xdg/xdg-ubuntu:/etc/xdg SSH_AGENT_LAUNCHER=gnome-keyring XDG_MENU_PREFIX=gnome- no_proxy=localhost,127.0.0.0/8,::1 GNOME_DESKTOP_SESSION_ID=this-is-deprecated GTK_IM_MODULE=fcitx CONDA_EXE=/home/ubuntu/miniforge3/bin/conda _CE_M= ROS_PYTHON_VERSION=3 LC_ADDRESS=zh_CN.UTF-8 GNOME_SHELL_SESSION_MODE=ubuntu LC_NAME=zh_CN.UTF-8 SSH_AUTH_SOCK=/run/user/1000/keyring/ssh ISAACSIM_PATH=/home/ubuntu/.local/share/ov/pkg/isaac-sim-4.0.0 XMODIFIERS=@im=fcitx DESKTOP_SESSION=ubuntu LC_MONETARY=zh_CN.UTF-8 LD_LIBRARY=/usr/local/cuda/lib64: XML_CATALOG_FILES=file:///home/ubuntu/miniforge3/etc/xml/catalog file:///etc/xml/catalog GTK_MODULES=gail:atk-bridge PWD=/home/ubuntu XDG_SESSION_DESKTOP=ubuntu LOGNAME=ubuntu XDG_SESSION_TYPE=x11 CONDA_PREFIX=/home/ubuntu/miniforge3 GPG_AGENT_INFO=/run/user/1000/gnupg/S.gpg-agent:0:1 ISAAC_ROS_WS=/home/ubuntu/workspaces/isaac_ros-dev/ SYSTEMD_EXEC_PID=2738 XAUTHORITY=/run/user/1000/gdm/Xauthority ftp_proxy=http://127.0.0.1:8889/ WINDOWPATH=2 HOME=/home/ubuntu USERNAME=ubuntu IM_CONFIG_PHASE=1 LC_PAPER=zh_CN.UTF-8 LANG=en_US.UTF-8 ISAACSIM_PYTHON_EXE=/home/ubuntu/.local/share/ov/pkg/isaac-sim-4.0.0/python.sh LS_COLORS=rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:mi=00:su=37;41:sg=30;43:ca=30;41:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arc=01;31:*.arj=01;31:*.taz=01;31:*.lha=01;31:*.lz4=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.tzo=01;31:*.t7z=01;31:*.zip=01;31:*.z=01;31:*.dz=01;31:*.gz=01;31:*.lrz=01;31:*.lz=01;31:*.lzo=01;31:*.xz=01;31:*.zst=01;31:*.tzst=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.alz=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.cab=01;31:*.wim=01;31:*.swm=01;31:*.dwm=01;31:*.esd=01;31:*.jpg=01;35:*.jpeg=01;35:*.mjpg=01;35:*.mjpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.webp=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=00;36:*.au=00;36:*.flac=00;36:*.m4a=00;36:*.mid=00;36:*.midi=00;36:*.mka=00;36:*.mp3=00;36:*.mpc=00;36:*.ogg=00;36:*.ra=00;36:*.wav=00;36:*.oga=00;36:*.opus=00;36:*.spx=00;36:*.xspf=00;36: XDG_CURRENT_DESKTOP=ubuntu:GNOME VTE_VERSION=6800 CONDA_PROMPT_MODIFIER=(base) AMENT_PREFIX_PATH=/opt/ros/humble GNOME_TERMINAL_SCREEN=/org/gnome/Terminal/screen/03e8f259_e8b6_411e_aeaf_b0a41699a418 https_proxy=http://127.0.0.1:8889/ CLUTTER_IM_MODULE=fcitx LESSCLOSE=/usr/bin/lesspipe %s %s XDG_SESSION_CLASS=user PYTHONPATH=/opt/ros/humble/lib/python3.10/site-packages:/opt/ros/humble/local/lib/python3.10/dist-packages TERM=xterm-256color LC_IDENTIFICATION=zh_CN.UTF-8 _CE_CONDA= LESSOPEN=| /usr/bin/lesspipe %s USER=ubuntu NO_PROXY=localhost,127.0.0.0/8,::1 FTP_PROXY=http://127.0.0.1:8889/ GNOME_TERMINAL_SERVICE=:1.57150 CONDA_SHLVL=1 DISPLAY=:1 SHLVL=1 LC_TELEPHONE=zh_CN.UTF-8 HTTPS_PROXY=http://127.0.0.1:8889/ HTTP_PROXY=http://127.0.0.1:8889/ QT_IM_MODULE=fcitx LC_MEASUREMENT=zh_CN.UTF-8 http_proxy=http://127.0.0.1:8889/ CONDA_PYTHON_EXE=/home/ubuntu/miniforge3/bin/python LD_LIBRARY_PATH=/opt/ros/humble/opt/rviz_ogre_vendor/lib:/opt/ros/humble/lib/x86_64-linux-gnu:/opt/ros/humble/lib::/home/ubuntu/.mujoco/mujoco210/bin:/usr/lib/nvidia:usr/local/lib/ XDG_RUNTIME_DIR=/run/user/1000 ROS_LOCALHOST_ONLY=0 CONDA_DEFAULT_ENV=base ALL_PROXY=socks://127.0.0.1:1089/ LC_TIME=zh_CN.UTF-8 LC_ALL=en_US.UTF-8 GTK3_MODULES=xapp-gtk3-module XDG_DATA_DIRS=/usr/share/ubuntu:/usr/share/gnome:/usr/local/share/:/usr/share/:/var/lib/snapd/desktop all_proxy=socks://127.0.0.1:1089/ PATH=/opt/ros/humble/bin:/usr/local/cuda/bin:/home/ubuntu/miniforge3/bin:/home/ubuntu/miniforge3/condabin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/snap/bin GDMSESSION=ubuntu DBUS_SESSION_BUS_ADDRESS=unix:path=/run/user/1000/bus ROS_DISTRO=humble RMW_IMPLEMENTATION=rmw_cyclonedds_cpp LC_NUMERIC=zh_CN.UTF-8 _=/usr/bin/env `
> 
> **After:** `SHELL=/bin/bash DEBUG_FFLAGS=-march=nocona -mtune=haswell -ftree-vectorize -fPIC -fstack-protector-strong -fno-plt -O2 -ffunction-sections -pipe -isystem /home/ubuntu/miniforge3/envs/ros_env/include -march=nocona -mtune=haswell -ftree-vectorize -fPIC -fstack-protector-all -fno-plt -Og -g -Wall -Wextra -fcheck=all -fbacktrace -fimplicit-none -fvar-tracking-assignments -ffunction-sections -pipe ROS_VERSION=2 SESSION_MANAGER=local/ubuntu:@/tmp/.ICE-unix/2456,unix/ubuntu:/tmp/.ICE-unix/2456 QT_ACCESSIBILITY=1 COLORTERM=truecolor GCC_RANLIB=/home/ubuntu/miniforge3/envs/ros_env/bin/x86_64-conda-linux-gnu-gcc-ranlib XDG_CONFIG_DIRS=/etc/xdg/xdg-ubuntu:/etc/xdg SSH_AGENT_LAUNCHER=gnome-keyring XDG_MENU_PREFIX=gnome- no_proxy=l [... cut ...] me/ubuntu/miniforge3/envs/ros_env/bin/x86_64-conda-linux-gnu-c++filt DBUS_SESSION_BUS_ADDRESS=unix:path=/run/user/1000/bus BUILD=x86_64-conda-linux-gnu CONDA_PREFIX_1=/home/ubuntu/miniforge3 ROS_DISTRO=humble RANLIB=/home/ubuntu/miniforge3/envs/ros_env/bin/x86_64-conda-linux-gnu-ranlib CONDA_BUILD_SYSROOT=/home/ubuntu/miniforge3/envs/ros_env/x86_64-conda-linux-gnu/sysroot RMW_IMPLEMENTATION=rmw_cyclonedds_cpp LC_NUMERIC=zh_CN.UTF-8 PROJ_NETWORK=ON _=/home/ubuntu/miniforge3/envs/ros_env/bin/env `

Tobias-Fischer, 2024-12-22:
You are mixing system-ROS with Robostack-ROS, which is not supported. Please remove all `source` commands for the ROS in `/opt` from your `~/.bashrc` etc., and then activate your RoboStack-env from a "clean" environment.

Feel free to reopen if the problem still persists.

closed 2024-12-22 by Tobias-Fischer

wangyan-hlab, 2025-01-02:
> You are mixing system-ROS with Robostack-ROS, which is not supported. Please remove all `source` commands for the ROS in `/opt` from your `~/.bashrc` etc., and then activate your RoboStack-env from a "clean" environment.
> 
> Feel free to reopen if the problem still persists.

It works! Thank you very much.