RoboStack/ros-humble #221: Unable to build franka description

richardrl, 2024-11-08:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

I tried to build https://github.com/frankaemika/franka_description according to the instructions on Figure 3 here: https://arxiv.org/pdf/2104.12910

I got this error at the `boa build .` step:
```
RuntimeError: Solver could not find solution.Mamba failed to solve:
 - ros-humble-ament-cmake
 - ros-humble-ament-lint-auto
 - ros-humble-ament-lint-common
 - ros-humble-ament-cmake-pytest
 - libstdcxx-ng >=11.2.0
 - libgcc-ng >=11.2.0
 - libgcc-ng >=11.2.0
 - __glibc >=2.17,<3.0.a0

```

Here is my recipe.yaml:
```
package:
  name: ros
  version: 2024.11.08

source:
  - path: ''
    folder: ros-humble-franka-description/src/work

build:
  number: 0

about:
  home: https://www.ros.org/
  license: BSD-3-Clause
  summary: |
    Robot Operating System

extra:
  recipe-maintainers:
    - ros-forge

outputs:
  - package:
      name: ros-humble-franka-description
      version: 0.3.0
    about:
      home: https://www.ros.org/
      license:
        - Apache 2.0
      summary: franka_description contains URDF files and meshes of Franka robots
      maintainers:
        - Franka Robotics GmbH
    extra:
      recipe-maintainers:
        - robostack
    build:
      number: 0
      script:
        sel(unix): build_catkin.sh
        sel(win): build_catkin.bat
    requirements:
      build:
        - "{{ compiler('cxx') }}"
        - "{{ compiler('c') }}"
        - sel(linux64): sysroot_linux-64 2.17
        - ninja
        - sel(unix): make
        - sel(unix): coreutils
        - cmake
        - sel(build_platform != target_platform): python
        - sel(build_platform != target_platform): cross-python_{{ target_platform }}
        - sel(build_platform != target_platform): cython
        - sel(build_platform != target_platform): numpy
        - sel(build_platform != target_platform): pybind11
      host:
        - ros-humble-ament-cmake
        - ros-humble-ament-lint-auto
        - ros-humble-ament-lint-common
        - ros-humble-ament-cmake-pytest
      run:
        - ros-humble-xacro
        - ros-humble-rviz2
        - ros-humble-joint-state-publisher-gui
```

### Installed packages

```shell
Too large to list.
```

### Environment info

```shell
(/samsung4tb/conda_envs/ros_env) rli14@fitzroy:/samsung4tb/franka_description$ micromamba info

       libmamba version : 1.5.8
     micromamba version : 1.5.8
           curl version : libcurl/8.6.0 OpenSSL/3.2.1 zlib/1.2.13 zstd/1.5.5 libssh2/1.11.0 nghttp2/1.58.0
     libarchive version : libarchive 3.7.2 zlib/1.2.13 bz2lib/1.0.8 libzstd/1.5.5
       envs directories : /samsung4tb/conda_envs
          package cache : /home/rli14/micromamba/pkgs
                          /home/rli14/.mamba/pkgs
            environment : ros_env (active)
           env location : /samsung4tb/conda_envs/ros_env
      user config files : /home/rli14/.mambarc
 populated config files : /home/rli14/.condarc
       virtual packages : __unix=0=0
                          __linux=5.4.0=0
                          __glibc=2.31=0
                          __archspec=1=x86_64-v4
                          __cuda=12.4=0
               channels : 
       base environment : /home/rli14/micromamba
               platform : linux-64
```

Tobias-Fischer, 2024-11-08:
It looks like your channels aren’t properly functioning - can you show your condarc / mambarc?

Tobias-Fischer, 2025-01-06:
Closing due to inactivity.

closed 2025-01-06 by Tobias-Fischer