RoboStack/ros-humble #222: Package request: ros-humble-ur-robot-driver, ros-humble-ur-description, ros-humble-ur-dashboard-msgs

EzraBrooks, 2024-11-08:
### Package name

ros-humble-ur-robot-driver, ros-humble-ur-description, ros-humble-ur-dashboard-msgs

### Operating System(s)

- [ ] linux-64
- [X] linux-aarch64
- [ ] osx-64
- [ ] osx-arm64
- [ ] win-64

### Additional context (optional)

Hi RoboStack team!

I've been excited to try RoboStack for a while, and am finally digging into it. I'm trying to get [MoveIt Pro's example workspace](https://github.com/picknikrobotics/moveit_studio_ur_ws), which uses Universal Robots arms in all its examples, building w/ RoboStack and Pixi.

I'm an Apple Silicon user (building inside a VM as we do not yet support macOS natively in our application) and there are a few dependencies of this workspace that are not present in RoboStack for linux-aarch64, notably including these few UR-specific driver and description packages.

traversaro, 2024-11-27:
Hi @EzraBrooks ! As at this point you may have noticed, the idea is that we can do a full rebuild of all packages of robostack, so that we can easily add new packages after that.

The two PRs to track are:
* https://github.com/RoboStack/ros-humble/pull/206
* https://github.com/RoboStack/ros-humble/pull/209

closed 2025-01-07 by Tobias-Fischer

traversaro, 2025-01-07:
@EzraBrooks this was fixed in https://github.com/RoboStack/ros-humble/pull/235, feel free to open new issue if you still have problems!