RoboStack/ros-humble #223: ros-humble-webots-ros2-driver-2023.1.1-py311hb335429_5.tar.bz2 wants libpython3.12.so.1.0

husnoo, 2024-11-09:
Is this an inconsistency, or am I missing something? 
```
[nawal@heisenbug:~/data/software-exercises/ros2/my_package]$ python --version
Python 3.11.0
```

```
[nawal@heisenbug:~/data/software-exercises/ros2/my_package]$ micromamba search ros-humble-webots-ros2-driver
Getting repodata from channels...

conda-forge/linux-64                                        Using cache
conda-forge/noarch                                          Using cache
robostack-staging/linux-64                                  Using cache
robostack-staging/noarch                                    Using cache

ros-humble-webots-ros2-driver 2023.1.1 py311hb335429_5 (+ 1 builds)
───────────────────────────────────────────────────────────────────

 Name            ros-humble-webots-ros2-driver
 Version         2023.1.1
 Build           py311hb335429_5
 Size            1260 kB
 License         BSD-3-Clause
 Subdir          linux-64
 File Name       ros-humble-webots-ros2-driver-2023.1.1-py311hb335429_5.tar.bz2
 URL             https://conda.anaconda.org/robostack-staging/linux-64/ros-humble-webots-ros2-driver-2023.1.1-py311hb335429_5.tar.bz2
 MD5             e0dc67c34fb61e034aefc4d464e10a57
 SHA256          fb2229dc92dc92b1b89cb6f1c2293b3bdf3b93dcc7a57087d067a95d9dd7f71b

 Dependencies:
  - python
  - ros-humble-ros-workspace
  - ros-humble-rclcpp
  - ros-humble-std-msgs
  - ros-humble-geometry-msgs
  - ros-humble-pluginlib
  - ros-humble-rclpy
  - ros-humble-sensor-msgs
  - ros-humble-tf2-ros
  - ros-humble-tf2-geometry-msgs
  - ros-humble-tinyxml2-vendor
  - ros-humble-webots-ros2-msgs
  - ros-humble-vision-msgs
  - ros-humble-webots-ros2-importer
  - libgcc-ng >=12
  - libstdcxx-ng >=12
  - __glibc >=2.17,<3.0.a0
  - python_abi 3.11.* *_cp311
  - numpy >=1.23.5,<2.0a0
  - ros2-distro-mutex 0.5.* humble
  - ros2-distro-mutex 0.5.*
  - yaml-cpp >=0.8.0,<0.9.0a0

 Other Versions (2):

  Version  Build            
 ────────────────────────────
  2023.0.1 py310h7c61026_3  
  2023.0.0 py310h413e681_2  

```

```
[nawal@heisenbug:~/data/software-exercises/ros2/my_package]$ ldd  /home/nawal/data/tmp/micromamba/envs/ros_env/lib/webots_ros2_driver/driver | grep python
        libpython3.12.so.1.0 => not found
```

rpribau, 2024-11-19:
Did you try specify the Python version to 3.12? Since the problem says ```libpython3.12.```, probably you'll need to change your dependencies for Python. The command is something like:

```mamba install python=3.12``` or ```micromamba install -c conda-forge python=3.12```. 

I also recommend creating a ```.yaml``` file for specify all the dependencies you're going at the very beginning. I also had this issue and by specifying in the beginning the problems stopped.

Tobias-Fischer, 2024-11-20:
Looks like this needs a rebuild of the package – unfortunately we maintainers are short of time at the moment, so I am not sure when we'll get to that.

husnoo, 2024-11-20:
> Did you try specify the Python version to 3.12? Since the problem says `libpython3.12.`, probably you'll need to change your dependencies for Python. The command is something like:
>
> `mamba install [...]

unfortunately every version of python led to the solver picking the version of this package that insisted on a different version of python. I tried all three that showed up on the list. 

anyway I managed to install just that package from source.

closed 2024-11-20 by husnoo