RoboStack/ros-humble #230: Laser scan topic not showing when using "ros2 topic list"

Daviesss, 2024-12-19:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

Hi,

I have an issue of not seeing the laser scan topic when checking the lit of topics . When the robot is spawned into gazebo. 
[image]
[image]

### Installed packages

```shell
Package                              Version
------------------------------------ -----------
action_msgs                          1.2.1
action_tutorials_interfaces          0.20.3
action_tutorials_py                  0.20.3
actionlib_msgs                       4.2.3
aiohappyeyeballs                     2.4.4
aiohttp                              3.11.10
aiosignal                            1.3.2
altgraph                             0.17.2
ament_cmake_test                     1.3.7
ament_copyright                      0.12.10
ament_cppcheck                       0.12.10
ament_cpplint                        0.12.10
ament_flake8                         0.12.10
ament_index_python                   1.4.0
ament_lint                           0.12.10
ament_lint_cmake                     0.12.10
ament_package                        0.14.0
ament_pep257                         0.12.10
ament_uncrustify                     0.12.10
ament_xmllint                        0.12.10
angles                               1.15.0
argcomplete                          3.5.2
attrs                                24.3.0
builtin_interfaces                   1.2.1
catkin_pkg                           1.0.0
cffi                                 1.17.1
colcon-argcomplete                   0.3.3
colcon-bash                          0.5.0
colcon-cd                            0.1.1
colcon-cmake                         0.2.28
colcon-common-extensions             0.3.0
colcon-core                          0.18.2
colcon-defaults                      0.2.8
colcon-devtools                      0.3.0
colcon-library-path                  0.2.1
colcon-metadata                      0.2.5
colcon-output                        0.2.13
colcon-package-information           0.4.0
colcon-package-selection             0.2.10
colcon-parallel-executor             0.3.0
colcon-pkg-config                    0.1.0
colcon-powershell                    0.4.0
colcon-python-setup-py               0.2.9
colcon-recursive-crawl               0.2.3
colcon-ros                           0.5.0
colcon-test-result                   0.3.8
colcon-zsh                           0.5.0
colorama                             0.4.6
coloredlogs                          15.0.1
composition_interfaces               1.2.1
contourpy                            1.3.1
coverage                             7.6.9
cppcheck-htmlreport                  2.16.0
[... 169 lines cut ...]
tf2_py                               0.25.5
tf2_ros_py                           0.25.5
tf2_tools                            0.25.5
toml                                 0.10.2
tomli                                2.2.1
topic_monitor                        0.20.3
trajectory_msgs                      4.2.3
turtlesim                            1.4.2
unicodedata2                         15.1.0
unique_identifier_msgs               2.2.1
visualization_msgs                   4.2.3
vtk                                  9.2.6
wheel                                0.37.0
wslink                               2.2.1
xacro                                2.0.8
yarl                                 1.18.3
zipp                                 3.21.0
```

### Environment info

```shell
Ubuntu 22.04 
MacOS
```

traversaro, 2024-12-19:
Can you report the list of installed packages with `pixi list` instead of `pip list` ?

Furthermore, we have a bit of undocumented problem with laser sensors, can you try the following workaround:

~~~
pixi shell
cp $CONDA_PREFIX/lib/gazebo-11/plugins/libCameraPlugin.dylib $CONDA_PREFIX/lib/libCameraPlugin.dylib
cp $CONDA_PREFIX/lib/gazebo-11/plugins/libCameraPlugin.dylib $CONDA_PREFIX/lib/libCameraPlugin.dylib
cp $CONDA_PREFIX/lib/gazebo-11/plugins/libGpuRayPlugin.dylib $CONDA_PREFIX/lib/libGpuRayPlugin.dylib
cp $CONDA_PREFIX/lib/gazebo-11/plugins/libRayPlugin.dylib $CONDA_PREFIX/lib/libRayPlugin.dylib
~~~

see https://github.com/conda-forge/gazebo-feedstock/blob/9ae0b160ac88aa3ba7c269211a41aca06df8c695/recipe/build.sh#L80-L87 for more details.

traversaro, 2025-01-07:
@Daviesss can you try with the new ros2-distro-mutex==0.6.* rebuild packages? Thanks!

closed 2025-01-13 by Tobias-Fischer