RoboStack/ros-humble #233: Package request: package ros-humble-moveit deps

mora200217, 2025-01-04:
### Package name

ros-humble-moveit deps

### Operating System(s)

- [ ] linux-64
- [ ] linux-aarch64
- [X] osx-64
- [X] osx-arm64
- [ ] win-64

### Additional context (optional)

When building in osx it appear conflicts in packages such us 
-  ros-humble-moveit-planners,
-  ros-humble-pilz-industrial-motion-planner

etc 

Either mutex problem (Currently with py3.11 and mutex 0.5) or other dependency that is not working. 

I would love to have moveit from `mamba install ros-humble-moveit` and have all the required dependencies. 

Thanks

traversaro, 2025-01-05:
> I would love to have moveit from `mamba install ros-humble-moveit` and have all the required dependencies.

~~~
CONDA_SUBDIR=osx-arm64 conda create -n testdavies ros-humble-moveit
~~~

works fine. Can you report the exact command you are trying to execute in creating a new environment, and the verbatim errors you are getting? Thanks!

Tobias-Fischer, 2025-01-05:
This works fine for me: `micromamba create -n test ros-humble-moveit-planners ros-humble-pilz-industrial-motion-planner` - please feel free to reopen with more info :)

closed 2025-01-05 by Tobias-Fischer

mora200217, 2025-01-06:
@Tobias-Fischer @traversaro Which are your env specs ? Im using humble with python=3.11

mora200217, 2025-01-06:
When running `mamba install ros-humble-moveit`
```
Could not solve for environment specs
Encountered problems while solving:
  - package ros-humble-moveit-2.5.1-py39heb88ffd_0 requires ros-humble-moveit-planners, but none of the providers can be installed
```

Tobias-Fischer, 2025-01-06:
Can you please create a fresh environment instead of trying to add to an existing environment?

mora200217, 2025-01-06:
It keeps loading for like an hour 

[image]

traversaro, 2025-01-06:
Can you try with mamba? By the way, the fact that conda takes so much time may indicate you have quite an old conda, see https://www.anaconda.com/blog/a-faster-conda-for-a-growing-community  .

mora200217, 2025-01-06:
ugh! got it. When running tha `mamba create -n moveit_env ros-humble-moveit` it works with ease. However using `ros2-distro-mutex=0.3`. My main ros env is working with 0.5. Is planned an update for moveit to work with 0.5 mutex ? 

### PD: Conda info output: 
```
     active environment : humble
    active env location : /usr/local/Caskroom/miniforge/base/envs/humble
            shell level : 1
       user config file : /Users/amoralesma/.condarc
 populated config files : /usr/local/Caskroom/miniforge/base/.condarc
                          /Users/amoralesma/.condarc
          conda version : 22.11.1
    conda-build version : not installed
         python version : 3.10.8.final.0
       virtual packages : __archspec=1=x86_64
                          __osx=14.5=0
                          __unix=0=0
       base environment : /usr/local/Caskroom/miniforge/base  (writable)
      conda av data dir : /usr/local/Caskroom/miniforge/base/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack-staging/osx-64
                          https://conda.anaconda.org/robostack-staging/noarch
                          https://conda.anaconda.org/conda-forge/osx-64
                          https://conda.anaconda.org/conda-forge/noarch
                          https://conda.anaconda.org/robostack-experimental/osx-64
                          https://conda.anaconda.org/robostack-experimental/noarch
                          https://conda.anaconda.org/robostack-humble/osx-64
                          https://conda.anaconda.org/robostack-humble/noarch
                          https://conda.anaconda.org/robostack/osx-64
                          https://conda.anaconda.org/robostack/noarch
          package cache : /usr/local/Caskroom/miniforge/base/pkgs
                          /Users/amoralesma/.conda/pkgs
       envs directories : /usr/local/Caskroom/miniforge/base/envs
                          /Users/amoralesma/.conda/envs
               platform : osx-64
             user-agent : conda/22.11.1 requests/2.28.1 CPython/3.10.8 Darwin/23.5.0 OSX/14.5
                UID:GID : 501:20
             netrc file : None
           offline mode : False
```

traversaro, 2025-01-06:
Good catch:

~~~
traversaro@IITBMP014LW012:~$ CONDA_SUBDIR=osx-64 conda create -n newenv -c conda-forge -c robostack-staging ros-humble-m
oveit ros2-distro-mutex==0.5.*
Channels:
 - conda-forge
 - robostack-staging
Platform: osx-64
Collecting package metadata (repodata.json): done
Solving environment: failed

LibMambaUnsatisfiableError: Encountered problems while solving:
  - package ros-humble-moveit-2.5.4-py310h5aa156f_2 requires ros2-distro-mutex 0.2.* humble, but none of the providers can be installed

Could not solve for environment specs
The following packages are incompatible
├─ ros-humble-moveit is installable with the potential options
│  ├─ ros-humble-moveit 2.5.4 would require
│  │  └─ ros2-distro-mutex 0.2.* humble, which can be installed;
│  └─ ros-humble-moveit 2.5.4 would require
│     └─ ros2-distro-mutex 0.3.* , which can be installed;
└─ ros2-distro-mutex 0.5**  is not installable because it conflicts with any installable versions previously reported.
~~~

There is no plan to add moveit with `ros2-distro-mutex 0.5.* `, but https://github.com/RoboStack/ros-humble/pull/229 will add a `ros2-distro-mutex 0.6.*` that will include moveit also on osx-64 with recetn versions of the dependencies.

mora200217, 2025-01-06:
Niceeee. Thanks!