RoboStack/ros-humble #249: Package request: PACKAGE_NAME

puat133, 2025-01-20:
### Package name

ros-humble-twist-mux 

### Operating System(s)

- [x] linux-64
- [x] linux-aarch64
- [ ] osx-64
- [ ] osx-arm64
- [x] win-64

### Additional context (optional)

Hi,

I am new to robotic and currently starting to learn ROS. In one of the tutorial that I follow, twist-mux package is used. It would be helpful if this package is available for ros-humble. 
Current setup (if this matter):
Ubuntu : 24.04
ROS: Humble.

Thanks!

### Links and references (optional)

https://index.ros.org/p/twist_mux/#humble

traversaro, 2025-01-20:
Hello @puat133, are you interested in opening a PR to add this package? See https://robostack.github.io/Contributing.html . If we are lucky, the only change necessary is to add the package in the `vinca_*.yaml` files.

puat133, 2025-01-20:
Thanks @traversaro . I wish that I know how to do this, but at the moment, it is beyond my capability.

traversaro, 2025-01-21:
> Thanks [@traversaro](https://github.com/traversaro) . I wish that I know how to do this, but at the moment, it is beyond my capability.

No problem! The issue was fixed in https://github.com/RoboStack/ros-humble/pull/250, so you should now be able to find the requested packages in `robostack-staging` channel, thanks for the request.

puat133, 2025-01-21:
Great! Will check it out on my system and let you know.

puat133, 2025-01-22:
I tested it. Now it shows some dependency conflicts:
```
$ mamba install ros-humble-twist-mux
conda-forge/linux-64                                        Using cache
conda-forge/noarch                                          Using cache

Pinned packages:

  - python=3.10

error    libmamba Could not solve for environment specs
    The following packages are incompatible
    ├─ pin on python 3.10.* =* * is installable and it requires
    │  └─ python =3.10 *, which can be installed;
    └─ ros-humble-twist-mux =* * is not installable because it requires
       ├─ ros-humble-diagnostic-updater =* * with the potential options
       │  ├─ ros-humble-diagnostic-updater 4.0.0 would require
       │  │  └─ ros-humble-rclpy =* * with the potential options
       │  │     ├─ ros-humble-rclpy 3.3.15 would require
       │  │     │  └─ ros-humble-rmw-implementation =* * with the potential options
       │  │     │     ├─ ros-humble-rmw-implementation 2.8.4 would require
       │  │     │     │  └─ ros-humble-rmw-fastrtps-dynamic-cpp =* * with the potential options
       │  │     │     │     ├─ ros-humble-rmw-fastrtps-dynamic-cpp 6.2.7 would require
       │  │     │     │     │  └─ ros-humble-fastrtps =* * with the potential options
       │  │     │     │     │     ├─ ros-humble-fastrtps 2.6.9 would require
       │  │     │     │     │     │  └─ openssl >=3.4.0,<4.0a0 *, which can be installed;
       │  │     │     │     │     ├─ ros-humble-fastrtps 2.6.3 would require
       │  │     │     │     │     │  └─ ros2-distro-mutex =0.2 *, which can be installed;
       │  │     │     │     │     ├─ ros-humble-fastrtps 2.6.4 would require
       │  │     │     │     │     │  └─ ros2-distro-mutex =0.3 *, which can be installed;
       │  │     │     │     │     └─ ros-humble-fastrtps 2.6.7 would require
       │  │     │     │     │        └─ ros2-distro-mutex =0.5 humble, which can be installed;
       │  │     │     │     ├─ ros-humble-rmw-fastrtps-dynamic-cpp 6.2.2 would require
       │  │     │     │     │  └─ ros2-distro-mutex =0.2 *, which can be installed;
       │  │     │     │     ├─ ros-humble-rmw-fastrtps-dynamic-cpp 6.2.2 would require
       │  │     │     │     │  └─ ros2-distro-mutex =0.3 *, which can be installed;
       │  │     │     │     └─ ros-humble-rmw-fastrtps-dynamic-cpp 6.2.6 would require
       │  │     │     │        └─ ros2-distro-mutex =0.5 humble, which can be installed;
       │  │     │     ├─ ros-humble-rmw-implementation 2.8.1 would require
       │  │     │     │  └─ ros2-distro-mutex =0.2 *, which can be installed;
       │  │     │     ├─ ros-humble-rmw-implementation 2.8.2 would require
       │  │     │     │  └─ ros2-distro-mutex =0.3 *, which can be installed;
       │  │     │     └─ ros-humble-rmw-implementation 2.8.2 would require
       │  │     │        └─ ros2-distro-mutex =0.5 humble, which can be installed;
       │  │     ├─ ros-humble-rclpy 3.3.5 would require
       │  │     │  └─ ros2-distro-mutex =0.2 *, which can be installed;
       │  │     ├─ ros-humble-rclpy 3.3.7 would require
       │  │     │  └─ ros2-distro-mutex =0.3 *, which can be installed;
       │  │     └─ ros-humble-rclpy 3.3.11 would require
       │  │        └─ ros2-distro-mutex =0.5 humble, which can be installed;
       │  ├─ ros-humble-diagnostic-updater 3.0.0 would require
       │  │  └─ ros2-distro-mutex =0.2 *, which can be installed;
       │  ├─ ros-humble-diagnostic-updater 3.1.0 would require
       │  │  └─ ros2-distro-mutex =0.3 *, which can be installed;
       │  └─ ros-humble-diagnostic-updater 3.1.2 would require
       │     └─ ros2-distro-mutex =0.5 humble, which can be installed;
       ├─ ros-humble-rclcpp =* * with the potential options
       │  ├─ ros-humble-rclcpp 16.0.11 would require
       │  │  └─ ros-humble-rosidl-typesupport-c =* * with the potential options
       │  │     ├─ ros-humble-rosidl-typesupport-c 2.0.2 would require
       │  │     │  └─ ros-humble-ament-cmake-core =* * with the potential options
       │  │     │     ├─ ros-humble-ament-cmake-core 1.3.11 would require
       │  │     │     │  ├─ python =* * with the potential options
       │  │     │     │  │  ├─ python [3.10.0|3.10.1|...|3.10.9], which can be installed;
       │  │     │     │  │  ├─ python [3.10.7|3.10.8] would require
       │  │     │     │  │  │  └─ openssl >=1.1.1s,<1.1.2a *, which conflicts with any installable versions previously reported;
       │  │     │     │  │  ├─ python [3.10.0|3.10.1|3.10.2] would require
       │  │     │     │  │  │  └─ openssl >=1.1.1l,<1.1.2a *, which conflicts with any installable versions previously reported;
       │  │     │     │  │  ├─ python 3.10.4 would require
       │  │     │     │  │  │  └─ openssl >=1.1.1n,<1.1.2a *, which conflicts with any installable versions previously reported;
[... 34 lines cut ...]
       │  │     │     │     ├─ ros-humble-ament-package 0.14.0 would require
       │  │     │     │     │  └─ ros2-distro-mutex =0.3 *, which can be installed;
       │  │     │     │     └─ ros-humble-ament-package 0.14.0 would require
       │  │     │     │        └─ ros2-distro-mutex =0.5 humble, which can be installed;
       │  │     │     ├─ ros-humble-ament-cmake-core 1.3.3 would require
       │  │     │     │  └─ ros2-distro-mutex =0.2 *, which can be installed;
       │  │     │     ├─ ros-humble-ament-cmake-core 1.3.3 would require
       │  │     │     │  └─ ros2-distro-mutex =0.3 *, which can be installed;
[... cut ...]
       │  ├─ ros-humble-rclcpp 16.0.2 would require
       │  │  └─ ros2-distro-mutex =0.2 *, which can be installed;
       │  ├─ ros-humble-rclcpp 16.0.3 would require
       │  │  └─ ros2-distro-mutex =0.3 *, which can be installed;
       │  └─ ros-humble-rclcpp 16.0.8 would require
       │     └─ ros2-distro-mutex =0.5 humble, which can be installed;
       └─ ros2-distro-mutex =0.6 humble_*, which conflicts with any installable versions previously reported.
critical libmamba Could not solve for environment specs
```

traversaro, 2025-01-22:
The latest packages are available for Python 3.11, can you try to create a new environment?

puat133, 2025-01-22:
Ok. I will do. Thanks.