RoboStack/ros-humble #254: Package request: ros-humble-ur-robot-driver

RedBowtie, 2025-01-23:
### Package name

ros-humble-ur-robot-driver

### Operating System(s)

- [ ] linux-64
- [ ] linux-aarch64
- [x] osx-64
- [x] osx-arm64
- [ ] win-64

### Additional context (optional)

Hi RoboStack team!

I just tried robostack for a while and it's amazing. Some of the project I used require this package, I noticed that the package support for linux system is already done few weeks ago, not sure if it is possible to make them compatible with osx as well. That would be really helpful.

traversaro, 2025-01-23:
Hello @RedBowtie ! I was working on something similar in Jazzy in https://github.com/RoboStack/ros-jazzy/pull/19, I initially skipped `ur` packages on `osx` but @Tobias-Fischer correctly mentioned an old patch in https://github.com/RoboStack/ros-noetic/blob/main/patch/ros-noetic-ur-client-library.patch that we could use to ensure the `ur-client-library works on macos, not sure if ur-robot-driver in particular has more macos-specific problems.

traversaro, 2025-01-23:
Some notes if someone wants to look on this. Beside following https://robostack.github.io/Contributing.html, one should remove ur packages from skipped dependencies in https://github.com/RoboStack/ros-humble/blob/main/vinca_osx_arm64.yaml#L41-L45 .

traversaro, 2025-01-23:
> following https://robostack.github.io/Contributing.html,

You may also need to check https://github.com/RoboStack/robostack.github.io/pull/69, sorry but the contributing page was outdated.

PR #262 (2025-01-30): feat: remove ur skip for osx
closes #254
closed 2025-01-30 by Tobias-Fischer