RoboStack/ros-humble #263: build_win jobs are failing

traversaro, 2025-01-30:
### Solution to issue cannot be found in the documentation.

- [x] I checked the documentation.

### Issue

See https://github.com/RoboStack/ros-humble/actions/workflows/win.yml .

The failure is:

~~~

pixi install --locked
  C:\Users\runneradmin\.pixi\bin\pixi.exe install --locked --manifest-path pixi.toml --color always
  
    × lock-file not up-to-date with the project
  
Error: The process 'C:\Users\runneradmin\.pixi\bin\pixi.exe' failed with exit code 1
~~~

### Installed packages

```shell
.
```

### Environment info

```shell
.
```

traversaro, 2025-01-30:
It seems that we are fixing pixi to 0.39.4 in the ci (see https://github.com/RoboStack/vinca/blob/cbb8eba834ce3834df552977d6b08c325a30768e/vinca/generate_gha.py#L398), but we probably updated the pixi.lock file with pixi 0.40.3, and the lock file generated by pixi 0.40.3 does not seems to be compatible with 0.39.4 . Perhaps this is expected? Anyhow in this specific case can update the pixi version in vinca.

Tobias-Fischer, 2025-01-30:
Ah apologies I added the workaround before I realised I simply need to pixi update; we can remove the pin on pixi in vinca

Tobias-Fischer, 2025-01-30:
Should be all fixed :)

closed 2025-01-30 by Tobias-Fischer