RoboStack/ros-humble #265: Turtlebot3 ROS Humble and Gazebo on osx-arm64 (Mac M2)

bchax, 2025-02-02:
### Solution to issue cannot be found in the documentation.

- [x] I checked the documentation.

### Issue

Hello. Thanks for the support of this amazing project that Robostack is.

I am currently trying to run the simulation of a Turtlebot3 with ROS Humble in Gazebo with my M2 Mac but it fails.

The error when launching `ros2 launch turtlebot3_gazebo empty_world.launch.py` is the following:

```
(ros_env) bauti@arm64-apple-darwin20 ~ % ros2 launch turtlebot3_gazebo empty_world.launch.py          
[INFO] [launch]: All log files can be found below /Users/bauti/.ros/log/2025-02-02-15-03-56-399194-Bautis-MacBook-Air.local-43944
[INFO] [launch]: Default logging verbosity is set to INFO
urdf_file_name : turtlebot3_burger.urdf
urdf_file_name : turtlebot3_burger.urdf
[INFO] [gzserver-1]: process started with pid [43957]
[INFO] [gzclient-2]: process started with pid [43958]
[INFO] [robot_state_publisher-3]: process started with pid [43959]
[INFO] [spawn_entity.py-4]: process started with pid [43960]
	[spawn_entity.py-4] [INFO] [1738519455.103014108] [spawn_entity]: Spawn Entity started
[spawn_entity.py-4] [INFO] [1738519455.103645395] [spawn_entity]: Loading entity XML from file /Users/bauti/miniforge3/envs/ros_env/share/turtlebot3_gazebo/models/turtlebot3_burger/model.sdf
[spawn_entity.py-4] [INFO] [1738519455.105304675] [spawn_entity]: Waiting for service /spawn_entity, timeout = 30
[spawn_entity.py-4] [INFO] [1738519455.105513132] [spawn_entity]: Waiting for service /spawn_entity
[robot_state_publisher-3] [INFO] [1738519462.016838467] [robot_state_publisher]: got segment base_footprint
[robot_state_publisher-3] [INFO] [1738519462.017039466] [robot_state_publisher]: got segment base_link
[robot_state_publisher-3] [INFO] [1738519462.017078716] [robot_state_publisher]: got segment base_scan
[robot_state_publisher-3] [INFO] [1738519462.017102507] [robot_state_publisher]: got segment caster_back_link
[robot_state_publisher-3] [INFO] [1738519462.017124257] [robot_state_publisher]: got segment imu_link
[robot_state_publisher-3] [INFO] [1738519462.017145423] [robot_state_publisher]: got segment wheel_left_link
[robot_state_publisher-3] [INFO] [1738519462.017166548] [robot_state_publisher]: got segment wheel_right_link
[gzserver-1] libc++abi: terminating due to uncaught exception of type boost::wrapexcept<boost::lock_error>: boost: mutex lock failed in pthread_mutex_lock: Invalid argument [generic:22]
[ERROR] [gzserver-1]: process has died [pid 43957, exit code -6, cmd 'gzserver /Users/bauti/miniforge3/envs/ros_env/share/turtlebot3_gazebo/worlds/empty_world.world -slibgazebo_ros_init.so -slibgazebo_ros_factory.so -slibgazebo_ros_force_system.so'].
[spawn_entity.py-4] [INFO] [1738519483.532990952] [spawn_entity]: Calling service /spawn_entity
[gzclient-2] 2025-02-02 15:04:57.351 gzclient[43958:783523] +[IMKClient subclass]: chose IMKClient_Legacy
[gzclient-2] 2025-02-02 15:04:57.351 gzclient[43958:783523] +[IMKInputSession subclass]: chose IMKInputSession_Legacy
```

Though if I simply run `gazebo` it runs smoothly.

I have already tried downgrading Libprotobuf but any would remove my ros-humble-desktop, among many others.

If I run `gzserver --verbose -s libgazebo_ros_init.so -s libgazebo_ros_factory.so`, I get:

```
(ros_env) bauti@arm64-apple-darwin20 ~ % gzserver --verbose -s libgazebo_ros_init.so -s libgazebo_ros_factory.so
Gazebo multi-robot simulator, version 11.15.1
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

#     # ####### ####### ###  #####  #######
##    # #     #    #     #  #     # #
# #   # #     #    #     #  #       #
#  #  # #     #    #     #  #       #####
#   # # #     #    #     #  #       #
#    ## #     #    #     #  #     # #
#     # #######    #    ###  #####  #######

This version of Gazebo, now called Gazebo classic, reaches end-of-life
in January 2025. Users are highly encouraged to migrate to the new Gazebo
using our migration guides (https://gazebosim.org/docs/latest/gazebo_classic_migration/)

[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.1.42
[Msg] Loading world file [/Users/bauti/miniforge3/envs/ros_env/share/gazebo-11/worlds/empty.world]
libc++abi: terminating due to uncaught exception of type boost::wrapexcept<boost::lock_error>: boost: mutex lock failed in pthread_mutex_lock: Invalid argument [generic:22]
zsh: abort      gzserver --verbose -s libgazebo_ros_init.so -s libgazebo_ros_factory.so
```

Any help is welcome.

### Installed packages

```shell
Mamba list is capped by characters and mine does not fit.
```

### Environment info

```shell
(ros_env) bauti@arm64-apple-darwin20 ~ % mamba info                                                             

       libmamba version : 2.0.5
          mamba version : 2.0.5
           curl version : libcurl/8.11.1 OpenSSL/3.4.0 (SecureTransport) zlib/1.3.1 zstd/1.5.6 libssh2/1.11.1 nghttp2/1.64.0
     libarchive version : libarchive 3.7.4 zlib/1.2.13 liblzma/5.2.6 bz2lib/1.0.8 liblz4/1.9.3 libzstd/1.5.6
       envs directories : /Users/bauti/miniforge3/envs
          package cache : /Users/bauti/miniforge3/pkgs
                          /Users/bauti/.mamba/pkgs
            environment : ros_env (active)
           env location : /Users/bauti/miniforge3/envs/ros_env
      user config files : /Users/bauti/.mambarc
 populated config files : /Users/bauti/miniforge3/envs/ros_env/.condarc
                          /Users/bauti/.condarc
                          /Users/bauti/miniforge3/.condarc
       virtual packages : __unix=0=0
                          __osx=15.0=0
                          __archspec=1=arm64
               channels : https://conda.anaconda.org/robostack-staging/noarch
                          https://conda.anaconda.org/robostack-staging/osx-arm64
                          https://conda.anaconda.org/conda-forge/noarch
                          https://conda.anaconda.org/conda-forge/osx-arm64
                          https://conda.anaconda.org/robostack/noarch
                          https://conda.anaconda.org/robostack/osx-arm64
       base environment : /Users/bauti/miniforge3
               platform : osx-arm64
```

traversaro, 2025-02-02:
Can you provide the `mamba list` or `conda list` somehow? Even if it too much for the issue char list, can you either attach it as .txt file or link a gist?

bchax, 2025-02-02:
Hi, thank you for your fast reply.

Sure, I did not think about it. I attach my mamba list as well as the steps I did into creating my ros_env, the same taken right from the Getting Started Git of Robostack [https://robostack.github.io/GettingStarted.html](url), with a bunch of packages.

[mambaList.txt](https://github.com/user-attachments/files/18633880/mambaList.txt)

[steps.txt](https://github.com/user-attachments/files/18633883/steps.txt)

traversaro, 2025-02-02:
Thanks! I reproduce a similar issue on `linux-64` (I do not have access to a `osx-arm64` system at the moment) at https://github.com/traversaro/reproduce-turtlebot-spawn-issue :

~~~
traversaro@IITBMP014LW012:~$ git clone https://github.com/traversaro/reproduce-turtlebot-spawn-issue
traversaro@IITBMP014LW012:~$ cd reproduce-turtlebot-spawn-issue
traversaro@IITBMP014LW012:~/reproduce-turtlebot-spawn-issue$ pixi run -e mutex06 launch_sim
✨ Pixi task (launch_sim in mutex06): ros2 launch turtlebot3_gazebo empty_world.launch.py
[INFO] [launch]: All log files can be found below /home/traversaro/.ros/log/2025-02-02-22-31-49-215134-IITBMP014LW012-39207
[INFO] [launch]: Default logging verbosity is set to INFO
urdf_file_name : turtlebot3_burger.urdf
urdf_file_name : turtlebot3_burger.urdf
[INFO] [gzserver-1]: process started with pid [39238]
[INFO] [gzclient-2]: process started with pid [39240]
[INFO] [robot_state_publisher-3]: process started with pid [39242]
[INFO] [spawn_entity.py-4]: process started with pid [39244]
[robot_state_publisher-3] [INFO] [1738531911.221083092] [robot_state_publisher]: got segment base_footprint
[robot_state_publisher-3] [INFO] [1738531911.221221030] [robot_state_publisher]: got segment base_link
[robot_state_publisher-3] [INFO] [1738531911.221237232] [robot_state_publisher]: got segment base_scan
[robot_state_publisher-3] [INFO] [1738531911.221243759] [robot_state_publisher]: got segment caster_back_link
[robot_state_publisher-3] [INFO] [1738531911.221250082] [robot_state_publisher]: got segment imu_link
[robot_state_publisher-3] [INFO] [1738531911.221254838] [robot_state_publisher]: got segment wheel_left_link
[robot_state_publisher-3] [INFO] [1738531911.221258555] [robot_state_publisher]: got segment wheel_right_link
[gzserver-1] MESA: error: ZINK: failed to choose pdev
[gzserver-1] glx: failed to create drisw screen
[gzclient-2] MESA: error: ZINK: failed to choose pdev
[gzclient-2] glx: failed to create drisw screen
[spawn_entity.py-4] [INFO] [1738531912.851636909] [spawn_entity]: Spawn Entity started
[spawn_entity.py-4] [INFO] [1738531912.853206484] [spawn_entity]: Loading entity XML from file /home/traversaro/pixiws/reproduce-turtlebot-spawn-issue/.pixi/envs/mutex06/share/turtlebot3_gazebo/models/turtlebot3_burger/model.sdf
[spawn_entity.py-4] [INFO] [1738531912.857453153] [spawn_entity]: Waiting for service /spawn_entity, timeout = 30
[spawn_entity.py-4] [INFO] [1738531912.858502812] [spawn_entity]: Waiting for service /spawn_entity
[spawn_entity.py-4] [INFO] [1738531915.399213585] [spawn_entity]: Calling service /spawn_entity
[gzclient-2] MESA: error: ZINK: failed to choose pdev
[gzclient-2] glx: failed to create drisw screen
[gzserver-1] [INFO] [1738531915.714028922] [turtlebot3_imu]: <initial_orientation_as_reference> is unset, using default value of false to comply with REP 145 (world as orientation reference)
[spawn_entity.py-4] [INFO] [1738531916.005121794] [spawn_entity]: Spawn status: SpawnEntity: Successfully spawned entity [burger]
[gzserver-1] Fatal glibc error: pthread_mutex_lock.c:94 (___pthread_mutex_lock): assertion failed: mutex->__data.__owner == 0
[INFO] [spawn_entity.py-4]: process has finished cleanly [pid 39244]
[ERROR] [gzserver-1]: process has died [pid 39238, exit code -6, cmd 'gzserver /home/traversaro/pixiws/reproduce-turtlebot-spawn-issue/.pixi/envs/mutex06/share/turtlebot3_gazebo/worlds/empty_world.world -slibgazebo_ros_init.so -slibgazebo_ros_factory.so -slibgazebo_ros_force_system.so'].
[ERROR] [gzclient-2]: process has died [pid 39240, exit code -11, cmd 'gzclient'].
~~~

The `[gzserver-1] Fatal glibc error: pthread_mutex_lock.c:94 (___pthread_mutex_lock): assertion failed: mutex->__data.__owner == 0` line suggest some kind of mutex error, similar to `[ERROR] [gzserver-1]: process has died [pid 43957, exit code -6, cmd 'gzserver /Users/bauti/miniforge3/envs/ros_env/share/turtlebot3_gazebo/worlds/empty_world.world -slibgazebo_ros_init.so -slibgazebo_ros_factory.so -slibgazebo_ros_force_system.so'].`. There is no similar error however if I just run `gzserver --verbose -s libgazebo_ros_init.so -s libgazebo_ros_factory.so` .

However, everything works fine with `ros2-distro-mutex==0.5.0`, can you try that as a temporary workaround?

bchax, 2025-02-02:
I am having the same problem as with all the downgrades I tried, it removes nearly all my packages, including Humble. Is there any workaround to this?

```
  (ros_env) bauti@arm64-apple-darwin20 envs % mamba install ros2-distro-mutex==0.5.0
  conda-forge/noarch                                          Using cache
  conda-forge/osx-arm64                                       Using cache
  DEPRECATION: --no-python-version-warning is deprecated. pip 25.1 will enforce this behaviour change. A possible replacement is to remove the flag as it's a no-op. Discussion can be found at https://github.com/pypa/pip/issues/13154
  
  Pinned packages:
  
    - python=3.11
  
  
  Transaction
  
    Prefix: /Users/bauti/miniforge3/envs/ros_env
  
    Updating specs:
  
     - ros2-distro-mutex==0.5.0
  
  
    Package                                                Version  Build                 Channel                 Size
  ──────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
    Remove:
  ──────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
  
    - gazebo                                               11.15.1  orignameh8bf80bb_200  conda-forge           Cached
    - libboost-devel                                        1.86.0  hf450f58_3            conda-forge           Cached
    - pcl                                                   1.14.1  hd09b3b3_7            conda-forge           Cached
    - ros-humble-action-msgs                                 1.2.1  np126py311h38cf310_6  robostack-staging     Cached
    - ros-humble-action-tutorials-cpp                       0.20.5  np126py311h38cf310_6  robostack-staging     Cached
    - ros-humble-action-tutorials-interfaces                0.20.5  np126py311h38cf310_6  robostack-staging     Cached
    - ros-humble-action-tutorials-py                        0.20.5  np126py311h38cf310_6  robostack-staging     Cached
    - ros-humble-actionlib-msgs                              4.2.4  np126py311h38cf310_6  robostack-staging     Cached
    - ros-humble-actuator-msgs                               0.0.1  np126py311h38cf310_6  robostack-staging     Cached
    - ros-humble-ament-cmake-auto                           1.3.11  np126py311h38cf310_6  robostack-staging     Cached
    - ros-humble-ament-cmake-gmock                          1.3.11  np126py311h38cf310_6  robostack-staging     Cached
    - ros-humble-ament-cmake-gtest                          1.3.11  np126py311h38cf310_6  robostack-staging     Cached
    - ros-humble-ament-cmake-ros                            0.10.0  np126py311h38cf310_6  robostack-staging     Cached
    - ros-humble-behaviortree-cpp-v3                         3.8.7  np126py311h38cf310_6  robostack-staging     Cached
    - ros-humble-bond                                        3.0.2  np126py311h38cf310_6  robostack-staging     Cached
    - ros-humble-bondcpp                                     3.0.2  np126py311h38cf310_6  robostack-staging     Cached
    - ros-humble-builtin-interfaces                          1.2.1  np126py311h38cf310_6  robostack-staging     Cached
    - ros-humble-camera-calibration-parsers                 3.1.10  np126py311h38cf310_6  robostack-staging     Cached
    - ros-humble-camera-info-manager                        3.1.10  np126py311h38cf310_6  robostack-staging     Cached
    - ros-humble-cartographer-ros                         2.0.9002  np126py311h38cf310_6  robostack-staging     Cached
    - ros-humble-cartographer-ros-msgs                    2.0.9002  np126py311h38cf310_6  robostack-staging     Cached
    - ros-humble-common-interfaces                           4.2.4  np126py311h38cf310_6  robostack-staging     Cached
    - ros-humble-composition                                0.20.5  np126py311h38cf310_6  robostack-staging     Cached
    - ros-humble-composition-interfaces                      1.2.1  np126py311h38cf310_6  robostack-staging     Cached
    - ros-humble-compressed-depth-image-transport            2.5.2  np126py311h38cf310_6  robostack-staging     Cached
    - ros-humble-compressed-image-transport                  2.5.2  np126py311h38cf310_6  robostack-staging     Cached
    - ros-humble-control-msgs                                4.7.0  np126py311h38cf310_6  robostack-staging     Cached
    - ros-humble-control-toolbox                             3.4.0  np126py311h38cf310_6  robostack-staging     Cached
    - ros-humble-controller-interface                       2.47.0  np126py311h38cf310_6  robostack-staging     Cached
    - ros-humble-controller-manager                         2.47.0  np126py311h38cf310_6  robostack-staging     Cached
    - ros-humble-controller-manager-msgs                    2.47.0  np126py311h38cf310_6  robostack-staging     Cached
    - ros-humble-costmap-queue                              1.1.17  np126py311h38cf310_6  robostack-staging     Cached
    - ros-humble-cv-bridge                                   3.2.1  np126py311h38cf310_6  robostack-staging     Cached
    - ros-humble-demo-nodes-cpp                             0.20.5  np126py311h38cf310_6  robostack-staging     Cached
    - ros-humble-demo-nodes-cpp-native                      0.20.5  np126py311h38cf310_6  robostack-staging     Cached
    - ros-humble-demo-nodes-py                              0.20.5  np126py311h38cf310_6  robostack-staging     Cached
[... cut ...]
    - ros2-distro-mutex                                      0.6.0  humble_0              robostack-staging     Cached
    + ros2-distro-mutex                                      0.5.0  humble                robostack-staging     Cached
  
    Summary:
  
    Remove: 262 packages
    Change: 48 packages
    Downgrade: 81 packages
  
    Total download: 0 B
  
  ──────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
  
  
  Confirm changes: [Y/n]  
```

traversaro, 2025-02-02:
I strongly suggest to create new environments specifying all the packages you want to install, thanks to conda's cache you do not need to download new packages, but it make the debugging of what is going wrong much easier, as you always do operations from a clean state. For example, in this case I suggest to do something similar to:

~~~
mamba create -n newenv -c conda-forge -c robostack-staging ros-humble-desktop ros-humble-turtlebot3-gazebo ros-humble-turtlebot3 ros2-distro-mutex==0.5.0
~~~

traversaro, 2025-02-02:
The situation is the following:

| Gazebo Classic version   | gazebo-ros-pks version | Working? |
|:----------------:|:------------------:|:------------------:|
| 11.15.1          | 3.7.0              | No  |
| 11.14.0          | 3.7.0              | Yes |

I wonder if it is Gazebo Classic 11.15 regression.

traversaro, 2025-02-02:
I wonder if https://github.com/ros-simulation/gazebo_ros_pkgs/pull/1453 is related. In that case we could look into forcing the use of 3.9.0 and check if that works.

bchax, 2025-02-02:
1- Excelent recommendation. Gazebo launched but the Turtlebot3 does not show up, the error changed:

```
(newenv) bauti@arm64-apple-darwin20 envs % ros2 launch turtlebot3_gazebo empty_world.launch.py
[INFO] [launch]: All log files can be found below /Users/bauti/.ros/log/2025-02-02-19-21-26-693669-Bautis-MacBook-Air.local-2305
[INFO] [launch]: Default logging verbosity is set to INFO
urdf_file_name : turtlebot3_burger.urdf
urdf_file_name : turtlebot3_burger.urdf
[INFO] [gzserver-1]: process started with pid [2424]
[INFO] [gzclient-2]: process started with pid [2425]
[INFO] [robot_state_publisher-3]: process started with pid [2426]
[INFO] [spawn_entity.py-4]: process started with pid [2427]
[spawn_entity.py-4] Traceback (most recent call last):
[spawn_entity.py-4]   File "/Users/bauti/miniforge3/envs/newenv/lib/gazebo_ros/spawn_entity.py", line 26, in <module>
[spawn_entity.py-4]     from gazebo_msgs.msg import ModelStates
[spawn_entity.py-4] ModuleNotFoundError: No module named 'gazebo_msgs'
[ERROR] [spawn_entity.py-4]: process has died [pid 2427, exit code 1, cmd '/Users/bauti/miniforge3/envs/newenv/lib/gazebo_ros/spawn_entity.py -entity burger -file /Users/bauti/miniforge3/envs/newenv/share/turtlebot3_gazebo/models/turtlebot3_burger/model.sdf -x 0.0 -y 0.0 -z 0.01 --ros-args'].
[robot_state_publisher-3] [INFO] [1738534901.274858082] [robot_state_publisher]: got segment base_footprint
[robot_state_publisher-3] [INFO] [1738534901.275001708] [robot_state_publisher]: got segment base_link
[robot_state_publisher-3] [INFO] [1738534901.275022125] [robot_state_publisher]: got segment base_scan
[robot_state_publisher-3] [INFO] [1738534901.275034792] [robot_state_publisher]: got segment caster_back_link
[robot_state_publisher-3] [INFO] [1738534901.275046708] [robot_state_publisher]: got segment imu_link
[robot_state_publisher-3] [INFO] [1738534901.275058000] [robot_state_publisher]: got segment wheel_left_link
[robot_state_publisher-3] [INFO] [1738534901.275069167] [robot_state_publisher]: got segment wheel_right_link
[gzserver-1] NodeShared::RecvSrvRequest() error sending response: Host unreachable
[gzclient-2] 2025-02-02 19:23:16.758 gzclient[2425:1297487] +[IMKClient subclass]: chose IMKClient_Legacy
[gzclient-2] 2025-02-02 19:23:16.759 gzclient[2425:1297487] +[IMKInputSession subclass]: chose IMKInputSession_Legacy
[ERROR] [gzclient-2]: process has died [pid 2425, exit code -11, cmd 'gzclient'].
```

2- It looks like Gazebo 11.14.0 to me:

```
(newenv) bauti@arm64-apple-darwin20 envs % conda list gazebo
# packages in environment at /Users/bauti/miniforge3/envs/newenv:
#
# Name                    Version                   Build  Channel
gazebo                    11.14.0              h816cae4_7    conda-forge
ros-humble-gazebo-dev     3.7.0           py311h84d614b_5    robostack-staging
ros-humble-gazebo-msgs    3.7.0           py311hbfa132e_5    robostack-staging
ros-humble-gazebo-plugins 3.7.0           py311hbfa132e_5    robostack-staging
ros-humble-gazebo-ros     3.7.0           py311hbfa132e_5    robostack-staging
ros-humble-gazebo-ros-pkgs 3.7.0           py311hbfa132e_5    robostack-staging
ros-humble-turtlebot3-gazebo 2.2.5           py311hbfa132e_5    robostack-staging
```

3 and 4-  This tips are more advanced than my skills on these matters. I assume that I have to download the gazebo_ros_pkgs and try to implement this Git branch. However since it is not a Conda package I do not know how to deal it with them. I also tried with ros-humble-gazebo-ros-pkgs but could not find the cpp file I see in this Git.

traversaro, 2025-02-02:
Sure, the 3 and 4 comments were most notes to self or other contributors to the project. I plan to try them tomorrow.

traversaro, 2025-02-04:
> 1- Excelent recommendation. Gazebo launched but the Turtlebot3 does not show up, the error changed:

Can you provide the `conda list` also in that case? It seems that somehow in that case the `gazebo-msgs` pkgs is not installed or somehow python can't find it, what happens when you run `python -c "import gazebo_msgs"`.

> 3 and 4- This tips are more advanced than my skills on these matters. I assume that I have to download the gazebo_ros_pkgs and try to implement this Git branch. However since it is not a Conda [...]

Unfortunately I tried and also the patch https://github.com/ros-simulation/gazebo_ros_pkgs/pull/1453 or the releases 3.9.0 and the problem is still present.

traversaro, 2025-02-04:
If nothing happens, it means that `gazebo_msgs` Python module can be found. If it can't be found, probably somewhere in the ros launch chain a non-conda python executable is used, and then that one can't find the `gazebo_msgs` python module.

bchax, 2025-02-05:
I have tried on a different Mac without Conda nor Python and as you very well predicted _it works_:

```
(newenv) lilianaressia@MacBook-Air-de-Liliana ~ % ros2 launch turtlebot3_gazebo empty_world.launch.py
[INFO] [launch]: All log files can be found below /Users/lilianaressia/.ros/log/2025-02-04-22-34-32-289136-MacBook-Air-de-Liliana.local-43457
[INFO] [launch]: Default logging verbosity is set to INFO
urdf_file_name : turtlebot3_burger.urdf
urdf_file_name : turtlebot3_burger.urdf
[INFO] [gzserver-1]: process started with pid [43458]
[INFO] [gzclient-2]: process started with pid [43459]
[INFO] [robot_state_publisher-3]: process started with pid [43460]
[INFO] [spawn_entity.py-4]: process started with pid [43461]
[robot_state_publisher-3] [INFO] [1738719273.018233721] [robot_state_publisher]: got segment base_footprint
[robot_state_publisher-3] [INFO] [1738719273.018479088] [robot_state_publisher]: got segment base_link
[robot_state_publisher-3] [INFO] [1738719273.018490004] [robot_state_publisher]: got segment base_scan
[robot_state_publisher-3] [INFO] [1738719273.018496962] [robot_state_publisher]: got segment caster_back_link
[robot_state_publisher-3] [INFO] [1738719273.018504920] [robot_state_publisher]: got segment imu_link
[robot_state_publisher-3] [INFO] [1738719273.018510837] [robot_state_publisher]: got segment wheel_left_link
[robot_state_publisher-3] [INFO] [1738719273.018516753] [robot_state_publisher]: got segment wheel_right_link
[spawn_entity.py-4] [INFO] [1738719273.224597908] [spawn_entity]: Spawn Entity started
[spawn_entity.py-4] [INFO] [1738719273.225013977] [spawn_entity]: Loading entity XML from file /Users/lilianaressia/miniforge3/envs/newenv/share/turtlebot3_gazebo/models/turtlebot3_burger/model.sdf
[spawn_entity.py-4] [INFO] [1738719273.227168406] [spawn_entity]: Waiting for service /spawn_entity, timeout = 30
[spawn_entity.py-4] [INFO] [1738719273.227433647] [spawn_entity]: Waiting for service /spawn_entity
[spawn_entity.py-4] [INFO] [1738719273.749058674] [spawn_entity]: Calling service /spawn_entity
[gzserver-1] [INFO] [1738719274.510930894] [turtlebot3_imu]: <initial_orientation_as_reference> is unset, using default value of false to comply with REP 145 (world as orientation reference)
[spawn_entity.py-4] [INFO] [1738719274.516740160] [spawn_entity]: Spawn status: SpawnEntity: Successfully spawned entity [burger]
[INFO] [spawn_entity.py-4]: process has finished cleanly [pid 43461]
[gzserver-1] [INFO] [1738719275.357696186] [turtlebot3_diff_drive]: Wheel pair 1 separation set to [0.160000m]
[gzserver-1] [INFO] [1738719275.357753392] [turtlebot3_diff_drive]: Wheel pair 1 diameter set to [0.066000m]
[gzserver-1] [INFO] [1738719275.379201809] [turtlebot3_diff_drive]: Subscribed to [/cmd_vel]
[gzserver-1] [INFO] [1738719275.601505844] [turtlebot3_diff_drive]: Advertise odometry on [/odom]
[gzserver-1] [INFO] [1738719275.604283711] [turtlebot3_diff_drive]: Publishing odom transforms between [odom] and [base_footprint]
[gzserver-1] [INFO] [1738719275.813123357] [turtlebot3_joint_state]: Going to publish joint [wheel_left_joint]
[gzserver-1] [INFO] [1738719275.813147106] [turtlebot3_joint_state]: Going to publish joint [wheel_right_joint]
[gzclient-2] 2025-02-04 22:34:36.004 gzclient[43459:2827301] +[IMKClient subclass]: chose IMKClient_Legacy
[gzclient-2] 2025-02-04 22:34:36.004 gzclient[43459:2827301] +[IMKInputSession subclass]: chose IMKInputSession_Legacy
```

[image]

[mambaList.txt](https://github.com/user-attachments/files/18665400/mambaList.txt)

I am going to reinstall Conda on my computer to solve this clashing python problem.

Thank you very much.

solonovamax, 2025-09-20:
Hi, I am having the exact same issue on linux.
What steps exactly were taken to solve this, other than just uninstalling python. uninstalling python isn't an option for me, as I have other things on my system which depend on python.

here is a stack trace from when it gets coredumped:
```
#0  0x00007ff30dc9894c n/a (libc.so.6 + 0x9894c)
#1  0x00007ff30dc3e410 raise (libc.so.6 + 0x3e410)
#2  0x00007ff30dc2557a abort (libc.so.6 + 0x2557a)
#3  0x00007ff30dc26613 n/a (libc.so.6 + 0x26613)
#4  0x00007ff30dc35ed6 n/a (libc.so.6 + 0x35ed6)
#5  0x00007ff30dc9a137 pthread_mutex_lock (libc.so.6 + 0x9a137)
#6  0x00007ff2441d7f89 _ZN6gazebo9transport4Node9SubscribeINS_4msgs16LaserScanStampedEN14gazebo_plugins25GazeboRosRaySensorPrivateEEEN5boost10shared_ptrINS0_10SubscriberEEERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEMT0_FvRKNS8_IKT_EEEPSJ_b (libgazebo_ros_ray_sensor.so + 0x5df89)
#7  0x00007ff2441d8643 _ZN14gazebo_plugins25GazeboRosRaySensorPrivate24SubscribeGazeboLaserScanEv (libgazebo_ros_ray_sensor.so + 0x5e643)
#8  0x00007ff30df8f1b2 _ZN6gazebo7sensors6Sensor10LoadPluginESt10shared_ptrIN3sdf2v97ElementEE (libgazebo_sensors.so.11 + 0x9b1b2)
#9  0x00007ff30df8f9b2 _ZN6gazebo7sensors6Sensor4InitEv (libgazebo_sensors.so.11 + 0x9b9b2)
#10 0x00007ff30df85c71 _ZN6gazebo7sensors9RaySensor4InitEv (libgazebo_sensors.so.11 + 0x91c71)
#11 0x00007ff30df9a1d2 _ZN6gazebo7sensors13SensorManager6UpdateEb (libgazebo_sensors.so.11 + 0xa61d2)
#12 0x00007ff30e9e7a18 _ZN6gazebo6Server3RunEv (libgazebo.so.11 + 0x4da18)
#13 0x0000561a21cf1ac4 n/a (/home/solonovamax/Programming/C++/robot-repo-ros2/.pixi/envs/default/bin/gzserver-11.15.1 + 0xbac4)
#14 0x00007ff30dc27675 n/a (libc.so.6 + 0x27675)
#15 0x00007ff30dc27729 __libc_start_main (libc.so.6 + 0x27729)
#16 0x0000561a21cf1b8c n/a (/home/solonovamax/Programming/C++/robot-repo-ros2/.pixi/envs/default/bin/gzserver-11.15.1 + 0xbb8c)
```

I believe it's caused by this here:
https://github.com/ros-simulation/gazebo_ros_pkgs/blob/ros2/gazebo_plugins/src/gazebo_ros_ray_sensor.cpp#L181-L182