RoboStack/ros-humble #266: Spawn problems with Turtlebot3 Gazebo Humble in Windows

bchax, 2025-02-09:
### Solution to issue cannot be found in the documentation.

- [x] I checked the documentation.

### Issue

Hi. I cannot launch Gazebo with Ros Humble. This is the error:

ros2 launch turtlebot3_gazebo```
 empty_world.launch.py
[INFO] [launch]: All log files can be found below C:\Users\tadeo\.ros\log\2025-02-07-16-44-48-376144-Tadro-196
[INFO] [launch]: Default logging verbosity is set to INFO
urdf_file_name : turtlebot3_burger.urdf
urdf_file_name : turtlebot3_burger.urdf
[ERROR] [launch]: Caught exception in launch (see debug for traceback): executable 'spawn_entity.py' not found on the libexec directory 'C:\Users\tadeo\miniforge3\envs\newenv\Library\lib\gazebo_ros'
```

I went for the minimal install `mamba create -n newenv -c conda-forge -c robostack-staging ros-humble-desktop ros-humble-turtlebot3-gazebo ros-humble-turtlebot3`

[mamba_list.txt](https://github.com/user-attachments/files/18726650/mamba_list.txt)

### Installed packages

```shell
See .txt uploaded
```

### Environment info

```shell
(newenv) C:\Users>mamba info

       libmamba version : 2.0.5
          mamba version : 2.0.5
           curl version : libcurl/8.11.1 Schannel zlib/1.3.1 libssh2/1.11.1
     libarchive version : libarchive 3.7.7 zlib/1.3.1 liblzma/5.6.3 bz2lib/1.0.8 liblz4/1.10.0 libzstd/1.5.6
       envs directories : C:\Users\tadeo\miniforge3\envs
          package cache : C:\Users\tadeo\miniforge3\pkgs
                          C:\Users\tadeo\.mamba\pkgs
                          C:\Users\tadeo\AppData\Roaming\.mamba\pkgs
            environment : newenv (active)
           env location : C:\Users\tadeo\miniforge3\envs\newenv
      user config files : C:\Users\tadeo\.mambarc
 populated config files : C:\Users\tadeo\miniforge3\.condarc
       virtual packages : __win=10.0.22631=0
                          __archspec=1=x86_64
                          __cuda=12.7=0
               channels : https://conda.anaconda.org/conda-forge/noarch
                          https://conda.anaconda.org/conda-forge/win-64
       base environment : C:\Users\tadeo\miniforge3
               platform : win-64
```

bchax, 2025-02-12:
No followup?

Tobias-Fischer, 2025-04-08:
This looks familiar. Can you please provide the list of installed packages? I cannot see a .txt.

Aetherbase, 2025-04-27:
@bchax please fix your syntax of description of the issue so its easier for us to review your attachments.